MarlinFirmware/Marlin #5225: LCD screen and steps per mm

bluffchuffer, 2016-11-15:
I am using a Ramps with lcd screen on a small milling machine - 3 axis and all are leadscrew driven,  but find the REPRAP LCD display for the X or Y steps per mm in the control/motion menu can only display under 999, ie does not show the 1000's digit. Repetier or Pronterface can set the eeprom value to 1600, which is what I require require.
The Z axis does not have this problem, so imagine that it must be in the LCD setting somewhere - Where can I alter it?

esenapaj, 2016-11-15:
There is two solutions.
You are able to test those.

A: Introduce a +1234.56 format, and use it for all the axes and e
https://github.com/esenapaj/Marlin/tree/Introduce-a-%2B1234.56-format

B: Use +1234.5 format also x and y
https://github.com/esenapaj/Marlin/tree/Use-%2B-1234.5-format-also-steps/mm-of-X-and-Y

bluffchuffer, 2016-11-16:
Thanks,
I used B below.

esenapaj, 2016-11-20:
@thinkyhead 
About patch for this, are you submit and merge it? or I submit?

bluffchuffer, 2016-11-20:
Hello,
If I am thinkyhead then please submit and merge it. I know little enough 
about all this stuff.
Thanks

thinkyhead, 2016-11-21:
@esenapaj I think your first solution (A) is better, because steps/mm should be as accurate as possible, down to 100ths of a mm. You may submit it, if you like!

esenapaj, 2016-11-21:
I've submitted a patch by solution (A) just now.

closed 2016-11-21 by thinkyhead