MarlinFirmware/Marlin #5227: Z dual stepper drivers issue

3RLikeNoOther, 2016-11-15:
Hi,
recently I update marlin, before I was using a version of the 2013 and it was working without problem.
I've a prusa i3 with two Z motors and I'm using **Z_DUAL_STEPPER_DRIVERS** and my board is BOARD_RAMPS_13_EFB

Most of times it works perfectly, but sometimes the second motor doesn't move, I've noticed that it happens when it's just turned on. 
For example when I switch on and then I send "home X", it should move up for 5mm and then move to the X endstop, but only the first Z motor moves up.
It's a very annoying problem and even dangerous for the machine.
I've also noticed that when it ends the auto leveling routine the motor makes a noise for about a second (I'm not sure that it's the same problem).

If I upload the old one i don't get any problem, so I guess I need to keep the older one.
Let me know if you know any trick to solve this issue.
Thanks

PS
This is my firmware if you need: https://drive.google.com/drive/folders/0B8_YOuMOnuqQeXYzODdmM3ljRWs?usp=sharing

thinkyhead, 2016-11-16:
Welcome to Github!

The acceleration algorithm has changed since 2013, so your Z max speed and acceleration may be too high. Try lowering them and see if the problem persists.

3RLikeNoOther, 2016-11-16:
You are right, thanks.
I just tried to lower those values and now it works.
Is there any way to estimate the values that I should use? Acceleration is not so easy to see...

thinkyhead, 2016-11-17:
@3RLikeNoOther I've never found a good way to do it. I just do trial-and-error until something works. I usually start with a value somewhere around 10x the max feedrate setting.

closed 2016-11-17 by thinkyhead