MarlinFirmware/Marlin #5231: CoreXY always the same direction

hexxter, 2016-11-16:
Hi @ all,

I use RC7 and my CoreXY Printer moves always in the same direction.
desc (tests with proterface):
+x moves in +x direction
+y moves in +y direction
but...
-x moves in +y dir
-y moves in +x dir

i have changed my motors (X<>Y) and changed the direction of each motor in all combinations but only the  direction will be changed.
The Printer dives only in 2 direction and not back.

the settings in config has no effects:
#define INVERT_X_DIR false
#define INVERT_Y_DIR true

I have all combinations tested with no effect. Does this settings not work if i use CoreXY?

FHeilmann, 2016-11-16:
check this tutorial made by the Voron CoreXY community, maybe it helps:
https://www.reddit.com/r/voroncorexy/comments/4fm163/debugging_corexy_motion_on_voron_without_losing/

hexxter, 2016-11-16:
i found the problem ;o) my rumba board is buggi. I used a second one and it works!

closed 2016-11-16 by hexxter