MarlinFirmware/Marlin #5232: THERMAL_PROTECTION_HOTENDS doesn't consider some time whilst cooling

psavva, 2016-11-16:
Dear Team, 

I think I have found a bug, which really has to do with the way Thermal Protection is implemented.

If my hotend is cold, and starts heating up, there is no issue at all.

If my hotend is hot, say 180 Deg Celsius, and the heaters are OFF.
We understand from this that its cooling down, and probably quite rapidly (considering external environment)

If I ask the heater to go to say 190 Deg, the heater will turn on, and attempt to heat up the hotend.
At this point, I will get a thermal runaway.

So, what's happening:
As the hotend is cooling down, it may drop 3 or 4 or even 5 degrees (depending on external environment)
The thermal runaway kicks in because it's most probably using the temperature at the point the heater has turned on, without considering the cooling down that's still in progress.

I suggest that the Thermal runaway protection identify and re-baseline the initial temperature, and understand using some curve that the heater is actually working, and it's just taking some time to heat up.

If you consider that it has dropped 3 degrees, and is taking since the time it recorded the initial temperature, it's not 2 degrees every 20 seconds, but instead becomes 5 degrees in the 20 seconds.

Blue-Marlin, 2016-11-16:
Or __you__ set THERMAL_PROTECTION_PERIOD to a time long enough for your heater.

psavva, 2016-11-16:
I actually did increase it.
I will have to increase it significantly to actually work.

Is that such a good idea safety wise?

thinkyhead, 2016-11-17:
It really comes down to configuration. If the heater is on and the hotend temperature is failing to rise, that's a reasonable condition to assume the heater or the thermistor is faulty. If your heater is particularly cold, and especially if it's an all-metal hot-end with an always-on fan, then the base condition (2 degree rise in 20 seconds) may be too tight. Simply follow the advice in the configuration file:
```cpp
   * If you get false positives for "Heating failed" increase WATCH_TEMP_PERIOD and/or decrease WATCH_TEMP_INCREASE
   * WATCH_TEMP_INCREASE should not be below 2.
```

closed 2016-11-17 by thinkyhead

psavva, 2016-11-17:
Thank you thinkyhead.
I'll follow that advice