MarlinFirmware/Marlin #5235: Save auto-calibration mesh easily

eried, 2016-11-17:
It would be so awesome to be able to save a complex 9 point calibration mesh using the menu functions and just do the homing once a while.

Any simple way to do this? maybe some special gcode files in the sd card? any hints?

Roxy-3D, 2016-11-17:
> It would be so awesome to be able to save a complex 9 point calibration mesh using the menu functions and just do the homing once a while.

Probably you mean doing the G29 once.   You always have to home the machine because it doesn't now where it is when it powers up.   But then...  I think you want it to just use the previously saved Mesh data.

UBL already does that.   It automatically loads the Mesh every time the board powers up (if you decide you want that behavior).

You don't even need to do a G29 in your startup GCode.  You power up the printer, and it knows how to correct.    And it can handle a 15x15 Mesh.   And it has built in tools to Validate the Mesh and easily edit the Mesh.    You can try it out here:    https://github.com/MarlinFirmware/Marlin/tree/devel-ubl

Ianmcmill, 2016-11-17:
Just to clarify (as I am ((again)) trying to understand and get z probe bed leveling running):

1.1.0 RC7 does not automatically load the last calibration mesh?
UBL does automatically load the last calibration mesh.

Do I really need a connected LCD to run the UBL branch?

thinkyhead, 2016-11-19:
> 1.1.0 RC7 does not automatically load the last calibration mesh?
> UBL does automatically load the last calibration mesh.

Correct.

> Do I really need a connected LCD to run the UBL branch?

Currently, yes.

mihadongle, 2016-11-21:
Hallo but i have a delta printer and i can not save them

danzca6, 2016-11-22:
UBL is great and I look forward to using it, but I think what @eried was referring to is the current mesh routines.  Recently I have run into the same frustrations with having to redo mesh probing.  I see that in M503 that the list of G29 S3 commands are reported for each of the current mesh probe points.  I was able to replay them, but M420 would not activate mesh without mesh being enabled.  I didn't see a simple command to do that.  So I created a G29 S6 parameter for MeshEnable that looks at each Z value in the current mesh grid for any zero values.  If there are non, then I enable mesh and call into the M420 routine to activate the mesh again.  Sending an M500 then saves everything and I am back up and running without probing.  I now have that as a gcode file on my SD card to execute whenever needed.  It would be great to have this as a menu item on an LCD to either save off the M50s G29 commands to a gcode file and to replay that gcode to reload the mesh and go.  I think this would help with adoption of Mesh making it easier on users to backup their probed points.

Roxy-3D, 2016-11-22:
>  I now have that as a gcode file on my SD card to execute whenever needed. It would be great to have this as a menu item on an LCD to either save off the M50s G29 commands to a gcode file and to replay that gcode to reload the mesh and go.

Yes.  But it is important to realize how expensive (time wise) firmware is to develop.   Having multiple ways to do the same thing takes away resources to accomplish other much needed things.   I know you are talking about the current Mesh Bed Leveling.   But I'm spending all my time trying to bring that to the next level with the UBL code base.   

If you bring up the UBL code base you will see that saving the mesh and having it automatically loaded and enabled when the printer powers up is very nice.   And you can save multiple mesh's if you print on several pieces of glass.   And there are very nice tools to validate and edit the mesh.

My suggestion would be to give the UBL code base a try and see if does what you need.  I think you will find it does.

danzca6, 2016-11-22:
Believe me I understand what goes into developing.  I also am very appreciative of all of the great features being added by the Marlin developers.  UBL sounds great and I may take you up on implementing the UBL code base.  Mesh has been amazing over ABL and I just wanted known that reloading your mesh points is possible.  Thank you for the reply.  I truly love the work being done here.

Roxy-3D, 2016-11-22:
Yes...  It will do those things very easily by design.  That is what it is meant to do.   But it is also supposed to combine the benefits of 3-Point and Grid Based leveling.   It does the 3-Point.  It does not do the Grid Based yet.   The Grid Based is really not very hard to do.   It hasn't been done because there are more important things to work.   The truth is, with a 10 x 10 Mesh that is finely tuned, there really isn't much need to do any further adjustments to it.

thinkyhead, 2016-11-26:
> @mihadongle Why on delta printer there is no mesh bed levelling?

Delta ONLY has mesh bed leveling. But it is not called that.

Marlin 1.1.0 has renamed the option to `AUTO_BED_LEVELING_BILINEAR` and it is only in `RCBugFix` which will soon be released as `1.1.0-RC8`.

closed 2017-10-31 by thinkyhead