MarlinFirmware/Marlin #5236: After changing MARLIN firmware getting some lines on model. Please help

Shrikant888, 2016-11-17:
[image]

emartinez167, 2016-11-17:
I think you have already been asked to post those questions in the reprap
forum. This is not a user support forum. All you are going to achieve is
piss everyone here.

Shrikant888, 2016-11-17:
I have closed the all issues now help me..

emartinez167, 2016-11-17:
I think you just don't understand. Go and SEARCH in the reprap forum. Put effort in actually reading and learning like everybody else here, instead of expecting someone else to do it for you.

Again, go to the reprap forum and read their documentation, and post your questions there. Google is a nice tool too.

@thinkyhead put it very simply. This is not an end user support forum, but a place to contribute to the Marlin firmware or post errors RELATED to the firmware, not user issues.

I am afraid you are going to make some people very upset if you insist.

closed 2016-11-17 by Shrikant888