MarlinFirmware/Marlin #5237: developer question: keep menu from exiting to info screen

stigjoergensen, 2016-11-17:
Hi Gang

is there a way to keep menu from exiting to info screen after timeout?

i have a loop like this:

      while(thermalManager.isHeatingHotend(extruder) && !lcd_clicked) {
        safe_delay(250);
      }

Thanks

Kaibob2, 2016-11-17:
You can just increase the timeout if this suits your needs
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.h#L72

or, read this post #5031

Blue-Marlin, 2016-11-17:
```
  while(thermalManager.isHeatingHotend(extruder) && !lcd_clicked) {
    safe_delay(250);
    return_to_status_ms = millis() + LCD_TIMEOUT_TO_STATUS;
  }
```
and making `return_to_status_ms` global could also work for you.

stigjoergensen, 2016-11-17:
Uhhh - lots of options...  thanks

will investigate what suits me the best...

thanks a bunch.

closed 2016-11-17 by stigjoergensen