MarlinFirmware/Marlin #5241: Keep Power supply ON on connect

orangeudav, 2016-11-17:
I have Rumba (with 1.1.0-RC7 of Marlin) board and ATX power supply. PS_ON rumba's pins connected to PS_ON power supply's pins. Also i have OctoPrint server (CubieBoard 2) powered from same ATX (from regular 5V line, i don't want use standby lines, it has limited max current). By default PS_ON on Rumba is active (PS_DEFAULT_OFF is disabled), so Rumba and OctoPrint working ok after power on. 
But after press Connect in OctoPrint's interface Rumba's PS_ON breaks, ATX turn off and OctoPrint turn off. How to prevent it? I think setup_powerhold() working incorrectly in my case. Am i right?

OctoPrint not send M80 nor M81 during connection.
```
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0xaff0dc70, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: External Reset
Recv: Marlin 1.1.0-RC7
Recv: echo: Last Updated: 2016-07-31 12:00 | Author: (none, default config)
Recv: Compiled: Sep 19 2016
Recv: echo: Free Memory: 5141  PlannerBufferBytes: 1232
Recv: echo:V24 stored settings retrieved (398 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X160.85 Y160.85 Z5120.00 E1105.08
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X160.00 Y160.00 Z5.00 E50.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X800 Y400 Z80 E5000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P800.00 R5000.00 T800.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X5.00 Z0.40 E5.00
Recv: echo:Home offset (mm)
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P23.81 I1.11 D127.20
Recv: echo:  M304 P114.45 I10.20 D321.13
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-7.50
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: M115
Recv: FIRMWARE_NAME:Marlin 1.1.0-RC7 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Rumba EXTRUDER_COUNT:1 UUID:8ee39833-9115-4961-a1e8-dbb1a5da35b6
Recv: ok
Send: M105
Recv: ok T:22.4 /0.0 B:22.2 /0.0 @:0 B@:0
```

thinkyhead, 2016-11-17:
The `setup_powerhold` function is also used by `setup` and it does seem to work there. And `setup` is called when connecting to a host also. So the power pin is being set the same way in both cases.

Try moving the `setup_powerhold` function to the very end of the `setup` function and see if it makes any difference.

orangeudav, 2016-11-17:
> Try moving the setup_powerhold function to the very end of the setup function and see if it makes any difference.
Nothing change

> called by setup and it does work for you on the initial power-up.
Can't find point of initial call of setup(). 

OUT_WRITE(PS_ON_PIN, PS_ON_AWAKE) always break pin state without testing current state, why?

i don't know C language well :(

thinkyhead, 2016-11-17:
No evidence that Octoprint is turning off the power?

thinkyhead, 2016-11-17:
> Can't find point of initial call

Can't find the `setup` function? In `RCBugFix` it's at line 9879.
The call to `setup_powerhold` is line 9893.

thinkyhead, 2016-11-17:
> always break pin state without testing current state

In code-world this is considered silly:
```cpp
if (!my_bool) my_bool = true;
```
It's just as efficient, and uses less code, to simply do:
```
my_bool = true;
```

orangeudav, 2016-11-17:
Ok, i tried something like

```
bool awake_already = false;

void setup() {
...
if (!awake_already) {
    setup_powerhold();
    awake_already = true;
  }
...
}

without any result, hmm

orangeudav, 2016-11-17:
> No evidence that Octoprint is turning off the power?
Pronterface and OctoPrint have same behaviour, Marlin resets while connecting.

I found https://github.com/MarlinFirmware/Marlin/issues/4041#issuecomment-226082542 :(

thinkyhead, 2016-11-18:
> Pronterface and OctoPrint have same behaviour, Marlin resets while connecting.

Ok, so you find that Pronterface also turns off the 12V power when connecting?

boelle, 2016-11-18:
i just discovered the same

did a flowrate test print and disconnected... nothing happened

but when i connected again marlin resets and power goes off.

its a bit anoying since i dont have a touch probe yet so will have to set height each time :-/

thinkyhead, 2016-11-19:
You're not using a RUMBA also, are you @boelle ?

boelle, 2016-11-19:
nope i use Ramps 1.4 

it did this when connecting from pronterface... ie it resets and 
1) everything is lost 
2) power goes off even if on

i have not tried serial monitor from the IDE but will check later today

boelle, 2016-11-19:
maybe this helps... maybe not

start
echo: External Reset
Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
Compiled: Nov 18 2016
echo: Free Memory: 3121  PlannerBufferBytes: 1200
echo:Hardcoded Default Settings Loaded

thinkyhead, 2016-11-19:
On reset or serial connect the same code is run:
```cpp
    #if ENABLED(PS_DEFAULT_OFF)
      OUT_WRITE(PS_ON_PIN, PS_ON_ASLEEP);
    #else
      OUT_WRITE(PS_ON_PIN, PS_ON_AWAKE);
    #endif
```

boelle, 2016-11-19:
So yes... if set to default off it will be off... nothing wrong there. could just change that setting

But why does it reset? should not do that just because the user disconnects and connect 5 sec after

out for a few hours but have the whole weekend for testing

thinkyhead, 2016-11-19:
My comment had nothing to do with `PS_DEFAULT_OFF`. Let me put it another way.

On reset or serial connect the same code is run:
```cpp
OUT_WRITE(PS_ON_PIN, PS_ON_AWAKE);
```

Blue-Marlin, 2016-11-19:
You'll never get it to work if you try it in software.
It's the same problem as the on heaters during reset with the RAMPS_FD - but in reveres.

During reset the PS_ON_PIN is an input - so neither HIGH nor LOW.
The power supply needs a HIGH/LOW to be on.
A 'external' pull up/down resistor likely could solve that.

boelle, 2016-11-19:
yeah but why does it also reset?

Blue-Marlin, 2016-11-19:
The Arduino Mega, and several other boards, are designed that way.
Otherwise you'd have to push a button to flash them.

Google for "Arduino Mega RESET-EN"

https://www.google.de/search?q=Arduino+Mega+RESET_EN&rlz=1C1CHBF_deDE717DE717&oq=Arduino+Mega+RESET_EN&aqs=chrome..69i57j69i64.18215j0j7&sourceid=chrome&ie=UTF-8#q=Arduino+Mega+RESET-EN

boelle, 2016-11-19:
ahh.... thanks. i was paranoid enough that it was a software error

orangeudav, 2016-11-19:
> The Arduino Mega, and several other boards, are designed that way.
Can i prevent this behaviour (for Rumba) by changing bootloader? Or removing it completly?

Blue-Marlin, 2016-11-20:
If you are really good in soldering try to open the circuit at this capacitor and put a switch in series.
[image]
[image]
[image]
__NO__ warranty for anything!

orangeudav, 2016-11-20:
@Blue-Marlin i can but i do not want :)
What if i disable RESET line in usb-serial firmware (http://reprap.org/wiki/File:RRD-RUMBA_SRC_16U2_FIRMWARE.zip) 
```
diff --git a/RUMBA_16U2_USB2Serial/ARDUINO-SERIAL/Arduino-usbserial.c b/RUMBA_16U2_USB2Serial/ARDUINO-SERIAL/Arduino-usbserial.c
index 8e6b0ae..74ca498 100755
--- a/RUMBA_16U2_USB2Serial/ARDUINO-SERIAL/Arduino-usbserial.c
+++ b/RUMBA_16U2_USB2Serial/ARDUINO-SERIAL/Arduino-usbserial.c
@@ -170,10 +170,10 @@ void SetupHardware(void)
 //DEL  TCCR0B = (1 << CS02); //CLOCK/256
 //     TCCR0B = (1 <<CS01) | (1 << CS00); //CLOCK/64
        TCCR0B = (1 << CS01); //CLOCK/8
-
+
        /* Pull target /RESET line high */
-       AVR_RESET_LINE_PORT |= AVR_RESET_LINE_MASK;
-       AVR_RESET_LINE_DDR  |= AVR_RESET_LINE_MASK;
+       // AVR_RESET_LINE_PORT |= AVR_RESET_LINE_MASK;
+       // AVR_RESET_LINE_DDR  |= AVR_RESET_LINE_MASK;
 }
@@ -272,10 +272,10 @@ void EVENT_CDC_Device_ControLineStateChanged(USB_ClassInfo_CDC_Device_t* const C

        if (CurrentDTRState)
        {
-         AVR_RESET_LINE_PORT &= ~AVR_RESET_LINE_MASK;
+         // AVR_RESET_LINE_PORT &= ~AVR_RESET_LINE_MASK;
        }
        else
        {
-         AVR_RESET_LINE_PORT |= AVR_RESET_LINE_MASK;
+         // AVR_RESET_LINE_PORT |= AVR_RESET_LINE_MASK;
        }
 }
```
am i on the right way?

orangeudav, 2016-11-20:
The one thing that i not understand is why it's named PORTD in the source code 
```
<Value>AVR_RESET_LINE_PORT=PORTD</Value>
<Value>AVR_RESET_LINE_DDR=DDRD</Value>
<Value>AVR_RESET_LINE_MASK=(128)</Value>
```
but connected to port B (PB0) on the schema

Blue-Marlin, 2016-11-20:
Sorry never messed with the software of the usb-serial-converters.

But while tracing the connection between 16U2 and the capacitor, i assumed to have counted wrong.
In the red picture i counted pin 13. That would be Port D Bit 7 - like in the software.
I assume the first picture is wrong or out dated.

However. Your board is what counts.

Blue-Marlin, 2016-11-20:
Looking at your software changes:
I'd let the initialisation in. Otherwise you have a undefined state again.

orangeudav, 2016-11-20:
@Blue-Marlin I found more actual schema https://raw.githubusercontent.com/Aus3D/RUMBA-Plus/master/Schematic.png with right reset pin

Blue-Marlin, 2016-11-20:
Oh - nice. So i can count. ;-)
My source was the RepRap Wiki.

orangeudav, 2016-11-20:
@Blue-Marlin i compiled firmware with disabled RESET, but dfu-programmer (and FLIP too) dont want flush it to atmega16u2 :(
[RUMBA_16U2_USB2Serial.hex.zip](https://github.com/MarlinFirmware/Marlin/files/602170/RUMBA_16U2_USB2Serial.hex.zip)

orangeudav, 2016-11-21:
Ok, now i can prevent External Reset on host reconnecting, but first connection still causes power down.

ntoff, 2016-11-21:
The easy way to disable the Mega's auto reset is to whack a capacitor between reset and ground

http://forum.arduino.cc/index.php?PHPSESSID=bv3tkg3en539724h7aedl4vij7&topic=28167.msg208351#msg208351

Worked for me

macst34, 2016-11-23:
I have a mega 2560 and ramps 1.4 setup. This issue that orangeudav pointed out bothered me to no end since the mega resets whenever I a) plug in usb or when I b) open cura or anything else connects to the 2560's com port.

Now this doesn't matter if the printer is idle, but if I am printing from the sdcard at that moment, it stops printing in its tracks resulting in a failed print.

I did this mod by adding a 10uF as ntoff suggested. I also added a tiny switch so I can change behavior on the fly in case i need to flash the firmware some more without having to time the reset button push. 

I wirewrapped some connections to RESET and GND on the underside of the ramps and put the switch on a little pcb that I fixed to ramps by soldering on longer pins for Aux3 so they extend to the underside of ramps (the connection is only physical, I just didn't want to have the switch separate from the ramps board and I definitely didn't want to drill the ramps or kludge and hotglue the pcb on)

here are some pix:

before:
[image]

after:
[image]
the little round pcb under the LCD connection has the 10uF cap and a switch to select behavior.

I'd show you the underside, but i'd have to take it apart again when I just put it back together.

Anyways, the wiring is so simple:

RST----10uF---switch---GND

Anyways, for the rumba, I'd try blue_marlins's advice and tombstone that cap and rig a switch inline. You could probably mount the switch by soldering it to the micro usb grounding pads.

For added fun: If you look closely, you'll see single pin headers for PS_ON, VCC and GND because I added a relay to my elcheapo 12v power supply so I can turn it on using marlin's PS_ON capability from the LCD or with M80.

happy hacking,
macst34

thinkyhead, 2016-11-26:
For OctoPrint, I note that you can supply commands to be sent to the printer [when connected](http://docs.octoprint.org/en/master/configuration/config_yaml.html#events). You can just add an `M80` there.

macst34, 2016-11-26:
@thinkyhead.... if octoprint connects, the rumba resets because of DTR being sent over USB. Issuing a m80 from the octoprint would not be possible because the reset causes the power supply to switch off which cuts power to the rumba and the Octo.

This is why this has devolved into soldering and switches for this issue.

Michael

thinkyhead, 2016-11-26:
@macst34 I see. I'm used to boards with a diode allowing separate 5V and 12V power. It's interesting that we now have setups where `M80` can never be used.

macst34, 2016-11-26:
What follows is a summary and what I beleive is the is the definitve answer(s) to close this issue.

@thinkyhead dual power supplies still does not prevent this issue totally. Default behavior of arduino and derivatives is to reset when DTR is encountered. The issue arises when you use a power supply that heeds the PS_ON pin and you are powering your logic off the same PS. When the arduino/Mega/Rumba is reset, PS_ON is momentarily pulled high which cuts off the board's own power supply.  This happens regardless if you define PS_DEFAULT_OFF or not in Marlin's configuration.h, because after the reset, nothing is parsed because there is no power to anything. 

The board may be still powered by the ATX PS's standby 5v. In this case, if PS_DEFAULT_OFF is defined, PS_ON stays high and the PS continues to be off. If PS_DEFAULT_OFF is not defined, PS_ON gets pulled low and the PS turns on. But this step never gets to execute since the power to everything is already off and this logic is never run.

Like @Blue-Marlin said, you can prevent the default reset behavior of the MEGA by cutting the RESET_EN trace, but this means you have to hit the reset button yourself when you are flashing a MEGA with RESET_EN cut. This is a pain in the ass, but doable. For @orangeudav, this does not apply because the rumba has no reset trace on that board (though you can unsolder the C16 resistor that @Blue-marlin mentioned which I believe is equivalent to cutting RESET_EN on the MEGA.)

Hey @orangeudav, what @thinkyhead said got me thinking, there is probably another solution for you, I think. Power your octoprint/cubieboard and the rumba (just for the logic) with a separate 5V PS since you don't want to use the standby 5V from the ATX PS.  That way you can connect to the octoprint and then issue a M80 to the rumba which should kick your ATX PS on. The diode protection that @thinkyhead mentioned should let you run both power supplies concurrently.

This comes with with one caveat, if you are printing something, it's gonna fail when you connect to the octoprint/cubieboard since the DTR is still sent and everything resets. At least in this case, you can turn things back on remotely since the octoprint can still talk to your rumba. For me, I didn't really care about the reset behavior, except(!) when it broke my in-process print, so I went the route of adding a switch and cap to prevent the default behavior of the board when encountering a DTR. So if you don't care about the rumba resetting, just get another 5v power supply for the cubieboard and rumba. If you don't want printing to stop like me, you still got to break out the soldering iron and do some

happy hacking,
Michael

ps. @orangeudav, you probably seen this before, but if you didn't, it's pretty close to what you want. 
Just remember to add the secondary always-on 5v PS and  :%s/Ramps/Rumba/ | %s/PI/cubieboard/.
https://github.com/foosel/OctoPrint/wiki/Control-your-printer's-ATX-PSU-through-a-RAMPS-board-using-OctoPrint

orangeudav, 2016-11-26:
Good summary, @macst34, thank you!

> Power your octoprint/cubieboard and the rumba (just for the logic) with a separate 5V PS since you don't want to use the standby 5V from the ATX PS.

Yep, this is my starting point for this topic ;) The problem is that i have bad quality of 220V/110V AC power, and sometimes OctoPrint randomly (after switching off the motor of the refrigerator :) ) hung and broke current printing, but ATX PS feels good in the same time.

As i understand, another option to prevent USB's resets is ... not using USB completely (exclude usb-serial 16u2, who generate resets signals)! I have hardware UART on the OctoPrint's host. And i can solder wires directly to atmega2560's tx/rx pins. And, perhaps, i need a levels converter.

closed 2016-11-26 by orangeudav

Blue-Marlin, 2016-11-26:
An other way could be to invert the PS_ON signal between Arduino output and ps-input and the changing the config to a X-Box-ps witch has the reverse logic.

thinkyhead, 2016-11-27:
> if you are printing something, it's gonna fail when you connect 

For me this has always been expected behavior. It is the norm with most RepRaps.

Andrei-Pozolotin, 2017-06-02:
try 
```
stty -F /dev/ttyACM0 -hupcl
```
to disable CTS -> RESET action