MarlinFirmware/Marlin #5242: Totally lost with endstops and z probe (never had endstops before)

Ianmcmill, 2016-11-17:
Hello,

beforehand I want to apologize for asking for help instead of submitting an issue but I don't know of any appropriate forum with users and coders acquainted with the marlin code base and especially the z probe stuff (in my case an inductive probe). The most expertise is bundled in this git.

I ask for someone be willing and generous to put together a comprehensible tutorial on how to setup a z probe.
I have found some tutorials on instructables and generally the net but most lack entirety or assume different configurations.
As Marlin's code grew over time so did the structure of the code and the descriptions of the options in configuration.h and _adv.h.
Maybe I am getting old and tired but to me the descriptions raise more questions or confuse me even more.

That said, maybe someone is already working on such a tutorial. The documentation on http://www.marlinfw.org/docs/development/configuration.html#axis-homing as the descriptions addressed above do confuse my little brain too much.
For several years I have printed without endstops altogether. I always parked the extruder after printing with G1 X0 Y0 Z10 and executed "G92 Z10" after each reboot.
See https://github.com/MarlinFirmware/Marlin/issues/4831#issuecomment-247872331 for a work-around when omitting endstops and NEVER forget to enter G92 Z10. Becomes a habit over time though

For the sake of easier z probe configuration I finally managed to hook up X and Y makerbot endstops but am confused how to setup them and get a z probe running.

Thank you.

Roxy-3D, 2016-11-18:
The current organization for probes and endstops makes perfect sense to most people.  But I have a hard time with it.   The way things are currently organized is if the controller board has some label silk screened on a it, that is how it is named in the software.      I know I'm in the minority, but that gets me confused and I always have to ask @thinkyhead to help me get my Z-probe set up right.    (I can do the end stops just fine!!!)

In fact...   I've had a problem with this long enough, when we started a 'Documentation' section, I thought it would make sense to make this topic the very first one!   https://github.com/MarlinFirmware/MarlinDocumentation/issues/20

Take your best shot at getting the endstops and Z-Probe configured.   If it does the wrong thing, post an issue with a copy of your Configuration.h file attached.   Somebody will be able to help you.

thinkyhead, 2016-11-18:
Marlin now (1.1.x) has individual flags for which endstops you're using. Just enable those you need. The Z probe has two base options that work for most setups:
- For a deltabot (or any machine with a Z max endstop) the probe should be connected to the Z-min connector, and both the `USE_ZMIN_PLUG` and `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` options should be enabled.
- For machines that have a Z-min endstop, the probe should be connected to the `Z_MAX_PIN`, and both `USE_ZMAX_PLUG` and `Z_MIN_PROBE_ENDSTOP` should be enabled.

Then you need to specify what kind of probe you have. The options are:
- `FIX_MOUNTED_PROBE`
- Z Servo (e.g., `#define Z_ENDSTOP_SERVO_NR 0`)
- `BLTOUCH`
- `Z_PROBE_ALLEN_KEY`
- `Z_PROBE_SLED`

Then set your probe offsets, indicating how far the probe "trigger-point" is from the nozzle.
```cpp
#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
```

For bed leveling, set one of these:
```cpp
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
```

All this is pretty-well documented in `Configuration.h`. I've tried to translate the geek-speak into English in most places.

mtowara, 2016-11-18:
What I was able to do in an older version of Marlin but what I can't seem to configure anymore is to have both a z_min stop active as well as the (fixed inductive) z-probe connected e.g. to z_max.
What I then want to do is to use the probe for both probing and homing and use the minstop as a fallback if for whatever reason the probe fails (e.g. it happened to me the other day because there was a lot of ooze under the nozzle and the whole carriage bent upwards instead of the probe triggering, the min endstop would have caught this but I cant configure it to do anything).
As far as I can tell I can only configure it to have the probe for probing and the endstop for homing.

Would the before mentioned behavior be something you would like to see implemented (or is it possible and I just don't get how)?

thinkyhead, 2016-11-19:
Min-stop as a fallback would be a new feature. It sounds like a reasonable idea. I'm surprised to hear of a probe that fails in any manner other than triggering early.

closed 2017-10-31 by thinkyhead