MarlinFirmware/Marlin #5249: [Feature Request]

boelle, 2016-11-18:
Got my printer work and printing from SD

It could be nice if we could delete files from the Display

just a thought

thinkyhead, 2016-11-19:
I doubt I'll be implementing such a thing, as it's kind of a strange paradigm, and the interface would be rather odd, unlike anything else.

boelle, 2016-11-19:
could be simple...

sd card 
   |   ------>> print
   |   ------>> delete

thinkyhead, 2016-11-19:
Sure, sure. Or rather, `SD Card` -> `Select File` -> `Print`/`Delete` -> `Are you sure?` because we don't want accidental deletion.

Kaibob2, 2016-11-19:
I also wouldn't want a file deletion option. Because of the slow refresh it sometimes happens to me, that i select a file and press the encoder too fast. Then the highlighted file is in fact already one file above or below and the print starts.
At this point i could imagine a message something like:
`XYZ.gcode selected. Start Print?`

boelle, 2016-11-19:
forcing is never a good option

but oh well... it was just a thought so i did not have to take out the card just to delete at file and put it back again

Roxy-3D, 2016-11-19:
> I also wouldn't want a file deletion option. 

The LCD Screen has so much latency.   If we do decide to support a delete option, we need to also have a confirmation screen for it.   My personal belief is this is better managed on a computer and not the printer.

Blue-Marlin, 2016-11-19:
How do you put the files onto your card?
The same interface provides a way to delete them.

boelle, 2016-11-19:
"so i did not have to take out the card just to delete at file and put it back again"

ie its stupid to take the card out go and fire up the pc, delete the file, turn off pc and go back and insert card

Blue-Marlin, 2016-11-19:
For what do you need to delete a file, if you don't need the space for new data?

boelle, 2016-11-19:
when i know i will not print that file again

boelle, 2016-11-19:
putting this one to sleep. Who wants to ride a dead horse anyway

closed 2016-11-19 by boelle