MarlinFirmware/Marlin #5260: [ISSUE] Filament unloads after print

boelle, 2016-11-20:
So i got my Z max endstop working so i can home 

Trying to get the perfect height for the first layer

after the print it unloads the filament for no good reason

configuration.h: http://pastebin.com/ErqPAzhx
configuration_adv.h: http://pastebin.com/YDxuBb9w
pins_RAMPS.h: http://pastebin.com/y2ZH70Pb

and dont tell me to do my homework.. i have paced through the files allready and dont have a clue.

Blue-Marlin, 2016-11-20:
Most likely a problem with the slicers/hosts end-script, a retract and a messed up relative/absolute mode.

boelle, 2016-11-20:
oki, will try and work with that. it worked under repetier so just assumed it was ok

boelle, 2016-11-20:
```
G91 ;switch to relative positioning
G1 Z10 ;Move up Z 10mm
G90 ; back to absolute mode
G1 E-1 F1200 ; extra retract
G92 E0 ; zero the extruded length
```
i did it this way as i wanted to make sure it lifted 10mm after the print is done. hence the G91 and then G90. i then retract just 1mm to prevent the worst ozze while the nozzle cools off

will see if google is my friend or if i get porn pics when i do a search :-D

boelle, 2016-11-20:
some forums say to do the retract while in relative mode and others do the same as i do...

will try to retract while in relative mode

boelle, 2016-11-20:
that did the trick.. just thought that repetier and marlin handled it the same

oh well

@Blue-Marlin thanks for the pointer

closed 2016-11-20 by boelle

Blue-Marlin, 2016-11-20:
Maybe in the other examples a M82/M83 was involved?

thinkyhead, 2016-11-21:
@boelle It may depend on the slicer too. Some of them reset E to zero between layers, and even between moves, so E ends up being a small number. But some slicers may just keep adding onto E, so then `G1 E-1` ends up going backward for a very long time. If you truly want to reverse only 1mm then definitely go to relative mode first.