MarlinFirmware/Marlin #5263: [FR]: Allow users to create different configuration.h files and swap between them as profiles in the menu system.

onebaldman, 2016-11-21:
I'm in the process of creating a modular system for my printer to allow users to quickly swap out hot ends for different hot ends. Doing this, the users would need to go in and change out settings in the configuration file each time, then reflash the firmware before proceeding with the new extruder. A good example of this would be swapping from a dual bowden style extruder to a direct feed single extruder to print flexible material. 

I would like to label the different configuration files, then be able to switch between them from the menu system. "Load Profile X", where X is the name of a configuration.

thinkyhead, 2016-11-21:
Configurations aren't amenable to being changed on the fly. However, you can change the parameters that differ using GCode, and store the scripts on an SD card. I do this for my two extruders. They save the new values to EEPROM so they will survive connect/reboot.

Extruder 1:
```gcode
; Set E-Steps for Wade with Hobbed Bolt
M92 E694.49429712
M500
```

Extruder 2:
```gcode
; Set E-Steps for MK7 with 50:1 motor
M92 E1258.1418426
M500
```

landodragon141, 2016-11-21:
@thinkyhead Wow 50:1 that's some serious torque ;)

SparkyCola, 2016-12-04:
This will probably never be realised (on this fork) because it would use a lot of RAM since all definitions in the configuration files are constants and not variables. So they are hard coded into the firmware which is very useful if you want to keep the RAM-usage low. The RAM usage needs to be low since 8-bit MCUs like the ATMega 2560 which is used in most 3D-Printers don't have that much RAM (8KB in case of the ATMega 2560).

thinkyhead, 2016-12-05:
> This will probably never be realised (on this fork) because it would use a lot of RAM since all definitions in the configuration files are constants and not variables

True, and that is to make the code more optimal, and the runtime more tailored to the machine — only including the required code. Only some of the values represented in the configuration can be changed at runtime. To realize this feature, all the values that someone would want to change would need to be made variable. Functions would need to be made more generalized. All this would require a major re-working of the codebase.

lrpirlet, 2016-12-05:
@onebaldman... Well, even if a complete configuration change is not thinkable, Marlin may already have what you need.... 

If we talk only about the **variables** tied with the hot-end then what else would you need that is not already there to swap a hot-end? Try M503 to get both the variable, the associated function AND the code to access it...

You 'only' need to write a short g-code program to adapt the variables...

Are you missing a variable to swap your hot-end?

```
10:50:00.634 : N20 M503*23
10:50:00.634 : echo:Steps per unit:
10:50:00.634 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
10:50:00.634 : echo:Maximum feedrates (mm/s):
10:50:00.649 : echo:  M203 X300.00 Y300.00 Z2.00 E25.00
10:50:00.649 : echo:Maximum Acceleration (mm/s2):
10:50:00.649 : echo:  M201 X1000 Y1000 Z50 E5000
10:50:00.649 : echo:Accelerations: P=printing, R=retract and T=travel
10:50:00.649 : echo:  M204 P1500.00 R2000.00 T1500.00
10:50:00.671 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
10:50:00.671 : echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
10:50:00.687 : echo:Home offset (mm)
10:50:00.687 : echo:  M206 X0.00 Y0.00 Z0.00
10:50:00.687 : echo:Material heatup parameters:
10:50:00.687 : echo:  M145 S0 H180 B70 F0
10:50:00.687 : M145 S1 H240 B110 F0
10:50:00.687 : echo:PID settings:
10:50:00.687 : echo:  M301 P22.41 I1.56 D80.41
10:50:00.703 : echo:Filament settings: Disabled
10:50:00.703 : echo:  M200 D3.00
10:50:00.703 : echo:  M200 D0
10:50:00.703 : echo:Z-Probe Offset (mm):
10:50:00.703 : echo:  M851 Z-4.79
10:50:00.703 : ok
```

closed 2017-10-31 by thinkyhead