MarlinFirmware/Marlin #5269: Configuration for ultimaker 1.5.7 board

OverlordBZ, 2016-11-21:
Hello. I'm new to 3d printing so please excuse any stupid mistakes I may make. 
Long story short. I wanted a 3d printer and decided to build one. I made a kossel mini and bought a ultimaker controller from a friend. All well until I made all the connections and turn it on. It gave the err: mintemp( figured it out eventually), and then mintemp for the bed ( it had his old settings from his printer). Since I didn't have a heated bed I decided to download the updated marlin(RC 7 I belive) and make the necessary configurations myself. After 20 or so tabs on chrome and 5 configuration files later I thought I finally cracked it. It verified correctly and uploaded fine but now the LCD is blank. I tried other setting in the LCD part of the config but nothing seems to help. So in short i need help. Maybe someone has a working one and can give me a config file that works, or help me figure out what i did wrong.Thank you.

Barrocas, 2016-11-21:
What LCD?

-- 
Cumprimentos,
D.er Hugo Barrocas

OverlordBZ, 2016-11-21:
12864 lcd. Is there anything more specific that i should give you?

Blue-Marlin, 2016-11-21:
Marlin is aware of about 12 different 12864 displays.

OverlordBZ, 2016-11-21:
It's soldered so i can't read the back side but on the board it says  Full Graphic Smart Controller. This one to be exact [http://www.bigtree-tech.com/index.php/en/production/detail/100](url)

Blue-Marlin, 2016-11-21:
Please activate 
```
// RepRapDiscount FULL GRAPHIC Smart Controller
// http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

//
```
and NO other display.

OverlordBZ, 2016-11-21:
OMG it works.Thank you so much.  Do i need to activate the sd card part or does that line take care of that to?

OverlordBZ, 2016-11-21:
A;so i don't get the mintemp err and nothing is attached to the controller .

Blue-Marlin, 2016-11-21:
If you want to use the sd-card you have to activate it.

Barrocas, 2016-11-21:
You get the 'mintemp err' if you define the firmware to expect a
thermistor (in the hotend and also in the heatbed), but If you have
nothing attached to the controller, you get the error.

You must connect the hotend thermistor to the controller board, set
the firmware to the desired thermistor number (table), and if all is
good, there will be no error.

-- 
Cumprimentos,
D.er Hugo Barrocas

OverlordBZ, 2016-11-21:
Ok. New predicament . On the screen the xyz flash with a "?" and i can't seem to make any axis move. Also i looked online for the menu tree and mine seems to not have all the items. Can the axis be moved trough the controller ?

Barrocas, 2016-11-21:
When powered on, the printer doesn't know were the carriages are, that's
why the XYZ flashes.

You need to 'home' it first, either with a G28 on the host, or in the LCD
menu - prepare - home all axis

Some menu options are enabled in the configuration files.

-- 
Cumprimentos,
D.er Hugo Barrocas

Barrocas, 2016-11-21:
Also, if it's the first time making the 'Kossel mini' move its axis, be
sure to have a way to quick disconect the power, in case the endstops are
not well setup (doesn't stop at endstops), or motors spin in the wrong
direction, making the carriages go down instead of up... :)

-- 
Cumprimentos,
D.er Hugo Barrocas

OverlordBZ, 2016-12-22:
Ok. Took a break while i ordered some new stepper as the ones i had didn't quite agree with the controller . They all move now but when i home 1 axis moves and jams into the endstop, one works fine(as in triggers then goes down like 2cm. I think this is how it's supposed to work)and the last one triggers then goes up and jams into the switch. Is there a certain way to configure the limit switches for delta?

Barrocas, 2016-12-22:
Double check your endstop connections.

I had the same problem, then I found out I had an X and Y endstops cables
switched over.

With all endstops free, make a M119 in Pronterface (or Repetier), and see
if they are all open.
Then, trigger one with a finger (if they are mechanical switch) and while
pressed, do M119 again, check if it is listed as being the intended endstop
triggered.
Do the same for the other endstops.

-- 
Cumprimentos,
D.er Hugo Barrocas

OverlordBZ, 2016-12-23:
All endstops work . They all appear open and trigger when I press them but it still has the same behavior when I home.

OverlordBZ, 2016-12-23:
Ok. New development. X and Z axis seem to work as intended but the Y axis seem to move only up. I think the homing works but when the y axis is supposed to go down like the others instead it goes up and jams into the endstop. Any ideas why?

OverlordBZ, 2016-12-23:
Yay. Found it. Bad driver apparently. Luckily i had another one spare and i swapped them. Here goes the calibration.

Barrocas, 2016-12-23:
Sure, but you have to see if M119 reports the endstop you are triggering.
So if you trigger endstop X, M119 must report X triggered, and the same for
the others.

Also check motor wires order, it's quite common to cross wires and cause
those sorts of issues.

With the machine in front of you, X is the left one, Y is the right, Z is
the centered in the opposite side.

http://chibidibidiwah.wdfiles.com/local--files/delta/delta.png

-- 
Cumprimentos,
D.er Hugo Barrocas

OverlordBZ, 2016-12-26:
Hello again. Having some issues calibrating.First: from what I've seen online when they home the carriages stay there next to the endstop, but on mine they come down ~3.5 cm. Second: when I set the z_home_pos on the lcd when i start it it's nowhere near that value. Any ideas ?

OverlordBZ, 2016-12-27:
Hello again. I managed to calibrate it in the end. Still travels that distance after homing but i just added that to the z height and it seems to have worked. If anyone know why it moves after homing I would like to know.

Barrocas, 2016-12-27:
The distance it lowers after homing is a feature, to move the effector to a
safe zone, from where it can be moved in any direccion without hitting the
endstops.

That feature can be disabled, in the config.h, comment out to disable:

# DELTA_HOME_TO_SAFE_ZONE

-- 
Cumprimentos,
D.er Hugo Barrocas

closed 2017-04-02 by thinkyhead