MarlinFirmware/Marlin #5270: U8glib MOD for RepRapDiscount G LCD JHD12864F screen / Ramps 1.4

GrevanGER, 2016-11-21:
Greetings. 

As some of you might know, there is a problem with certain versions of the _RepRapDiscount Full Graphic Display_ (RRD FGD for short from here on), since they don't seem to be supported by the default RRD FGD settings.
I have had this exact same issue with 1.0.2 about a year ago, but fergot to file a report about it here since I found something on youtube to solve it and, shame on me, you guys know how it is, once something is running rarely anybody makes the effort to go back here to report how they fixed it. So I figured I#d want to write this up. Even if it just for the case I have to workaround the support problem again in case I should loose my backup files again. 

Right now I am running the current RCBugFix version. And since this worked with 1.0.2 as well, I guess that you should be able to use this workaround with most Marlin versions. 

So this is what I get when running the default RRD FGD  settings by uncommenting 

> #define DOGLCD

at line 1065 in the Configuration.h
No beeps, no graphic glitches, just a blank screen. 
(Beeps and flickering displays indicate, that you might want to check this topic here: https://github.com/MarlinFirmware/Marlin/issues/4689)

[image]

I am not a huge fan of cutting away pin sockets and soldering right away, since the software solution might just be around the corner. So after some research I came across a youtube video with someone discribing the exact same problem that I ran in to with my hardware. 
[u8glib MOD for G LCD JHD12864F screen + RAMPS 1 4 + Arduino Mega 2560](https://youtu.be/4CD8ERaylmY)
With that said, all the praise goes to zeeproduction of course, since he was the one who did all the work.

But to sum up his workaround: 
All you have to do is to go in to the u8glib library and navigate to: 

> C:\Users\...\Documents\Arduino\libraries\U8glib\examples\GraphicsTest

There you will find an arduino file named "GraphicsTest".
Open it with your favorite text editor and add the following constructor call to the list which begins in line 43 and ends in line 140. 

`U8GLIB_ST7920_128X64_1X u8g(23, 17, 16); // SPI Com: SCK = en = 23, MOSI = rw = 17, CS = di = 16`

Done. 

[image]

Kaibob2, 2016-11-21:
Only trying to understand:

Isn't DOGLCD [automatically defined](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Conditionals_LCD.h#L101-L105) when uncommenting [REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L1231) as display to be used

And how does adding this line to an example file affect Marlin?

I have three RRD FGD from different suppliers and they all worked out of the box.

GrevanGER, 2016-11-21:
Well, my understanding of electronics is not big enough to explain it in english, but the Video explains it pretty Well.

Kaibob2, 2016-11-21:
Hmm, for my understanding this line might be necessary to get the example running, but after flashing Marlin to the Arduino this example sketch is gone. So whatever line was changed in it, it doesn't survive reflashing another sketch to the Arduino. Maybe someone else can clarify this. I really really wonder if this worked for you, but am still glad it did :)

Blue-Marlin, 2016-11-21:
If you really want, you can replace
```
#elif ENABLED(U8GLIB_ST7920)
  //U8GLIB_ST7920_128X64_RRD u8g(0,0,0);
  U8GLIB_ST7920_128X64_RRD u8g(0);
```
with this line.
But i swear you don't want. 
The original u8glib device uses a 8-stripe update and can't use fastio. It's at lease 50% slower than ours.

thinkyhead, 2016-11-26:
@GrevanGER It's not clear from the thread, but let me see if I understand the issue.

You're finding that the display fails to work when using the `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` option, but you can get it to work if you instead specify the generic `DOGLCD` with no other options, then explicitly set the `SCK`, `MOSI`, and `CS` in a custom constructor?

I guess that makes is Hardware SPI, which might also improve performance.

I note that we don't have any such option in Marlin…
```cpp
U8GLIB_ST7920_128X64_1X u8g(23, 17, 16);
```
…but we do have these:
```cpp
U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_RS);
U8GLIB_ST7920_128X64_RRD u8g(0);
U8GLIB_DOGM128 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);
U8GLIB_DOGM128 u8g(DOGLCD_CS, DOGLCD_A0);
U8GLIB_LM6059 u8g(DOGLCD_CS, DOGLCD_A0);
U8GLIB_NHD_C12864 u8g(DOGLCD_CS, DOGLCD_A0);
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE | U8G_I2C_OPT_FAST);
U8GLIB_SH1106_128X64 u8g(U8G_I2C_OPT_NONE | U8G_I2C_OPT_FAST);
U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0);
U8GLIB_DOGM128 u8g(DOGLCD_CS, DOGLCD_A0);
```
You can take a look at `ultralcd_impl_DOGM.h` to see how each one is invoked.

@Blue-Marlin Does it make sense to add a custom option for `JHD12864F` display? Is that distinct from the usual `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`?

thinkyhead, 2016-11-26:
Another question, @GrevanGER — What is your `MOTHERBOARD` setting? I want to check and see if perhaps your display is simply mis-wired. Perhaps the chip-select, enable, and MOSI pins could simply be changed in your board's pins file and it will fix the issue.

thinkyhead, 2016-11-26:
Ah. I see that your table indicates the pins are probably correct, at least with RAMPS.

[image]

So, which `MOTHERBOARD` are you using?

thinkyhead, 2016-11-26:
And just for kicks, maybe try this  in `ultralcd_impl_DOGM.h`…
```cpp
#elif ENABLED(U8GLIB_ST7920)
  U8GLIB_ST7920_128X64_RRD u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);
  //U8GLIB_ST7920_128X64_RRD u8g(0);
```

OlevP, 2017-02-28:
Hi,
I had the same issue with my Chinese Ramps + RRD Full Graphics LCD (Marlin 1.1.0-RC8) and was banging my head against the wall. It worked perfectly with Repetier so I knew the HW was working and had to be something in Marlin. Found this thread while searching for solution, tried the original suggestion (editing "GraphicsTest"), this did change anything. After I uncommented this in "ultralcd_impl_DOGM.h", the screen started working, although it's ssllooww:
<code>U8GLIB_ST7920_128X64_1X u8g(ST7920_CLK_PIN, ST7920_DAT_PIN, ST7920_CS_PIN);</code>

@thinkyhead's suggestion in previous post causes a compile error. As much as I understand <code>U8GLIB_ST7920_128X64_RRD</code> expects a single argument, but this provides 3.

Slow screen is much better than a blank one in my books, but I wouldn't mind getting it working with the faster option and I would be willing to spend some time on this. Unfortunately my programming skills are on elementary level and I don't know much about how LCDs work...

thinkyhead, 2017-03-01:
> U8GLIB_ST7920_128X64_RRD expects a single argument

Ah. Then your `U8GLIB_ST7920_128X64_1X` is closer to the mark. Perhaps we need to supply an advanced flag in `Configuration_adv.h` for this crop of RRDFGSC units. We might take a look at Repetier-Firmware also to see what their special mojo is.

OlevP, 2017-03-04:
After living with this solution now for a few days, it's not unbearably slow and has worked without hiccups, so I'm happy. Probably about the same as in Repetier, I guess Repetier uses the same U8glib standard WoW , and not the U8GLIB_ST7920_128X64_RRD that is created in Marlin (which seems not to work with my display and some others).

closed 2018-02-06 by thinkyhead