MarlinFirmware/Marlin #5272: [Question] Temp. disable heating monitoring

boelle, 2016-11-21:
Hi,

I want to torture test my printers heat bed circuit.

Practical i do this by setting temp to 110 and have a big desktop fan going at full blast

It will of course never reach 110 and will eat full current the whole time, and that is the point of my test. I want to make sure that all the wiring and connectors are up to the job and not melting and fail on me

I googled for a way to temporarily disable but came up empty.. is there a G or M code that allows this or will i have to disable in firmware?

FHeilmann, 2016-11-21:
Are you heating your bed from your board or by switching a relay? 

If its the former, you could upload a simple sketch that simply turns on the appropriate pin after bootup.

If its the latter, you can simply apply some input voltage (within the specs of course) to the relay input to switch it on.

boelle, 2016-11-21:
i do it via the ramps board

and yes your idea would be the simple one... i just did not think of it

instead i changed #define WATCH_BED_TEMP_PERIOD 60               // Seconds
to #define WATCH_BED_TEMP_PERIOD 14400                // Seconds

if nothing melts in 4 hours or discolors the connectors it should hold up

closed 2016-11-21 by boelle