MarlinFirmware/Marlin #5273: Compilation warnings in RCBugFix

Fixer2k, 2016-11-21:
Getting these warnings while compiling RCBugFix:

`C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2325:17: warning: 'menu_action_setting_edit_float43' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2325:17: warning: 'menu_action_setting_edit_float51' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2325:17: warning: 'menu_action_setting_edit_float62' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2325:17: warning: 'menu_action_setting_edit_long5' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2329:17: warning: 'menu_action_setting_edit_callback_float3' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, screenFunc_t callback) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2329:17: warning: 'menu_action_setting_edit_callback_float32' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, screenFunc_t callback) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2329:17: warning: 'menu_action_setting_edit_callback_float5' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, screenFunc_t callback) { \

                 ^

C:\Users\Fixer\AppData\Local\Temp\arduino_build_817334\sketch\ultralcd.cpp:2329:17: warning: 'menu_action_setting_edit_callback_float51' defined but not used [-Wunused-function]

     static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, screenFunc_t callback) { \

                 ^`

Blue-Marlin, 2016-11-21:
This are all just warnings. Everything is fine.

Fixer2k, 2016-11-21:
Yes, I know. The fw works great. I just like to compile without warnings. ;-)

Blue-Marlin, 2016-11-21:
In that case you can comment that lines.
But don't forget to activate them later, if you want to compile the other type of display.

thinkyhead, 2016-11-24:
Because it's a hassle to check all the compile-time conditions that determine whether a particular menu function is utilized, I just ignore these particular warnings.

thinkyhead, 2016-11-24:
Turns out that if we simply drop the `static` keyword on the function declarations within `ultralcd.cpp`, there's no effect on the code output, but it does eliminate these warnings.

closed 2016-11-24 by thinkyhead