MarlinFirmware/Marlin #5276: UBL Overview

BillStruve, 2016-11-22:
Missing detailed overview of what UBL does.

UBL seems to fix bed adhesion of the first layer by moving Z up and down to match the bed. During the next few layers, does UBL modify the E rate, and reduce the correction to Z, so that the print will eventually become flat, regardless of the bed topology?

If the top of the completed print matches the the shape of the bed, UBL is propagating the topology of the bed to the structure, so a warped bed will produce a warped print using UBL. Maybe I'm missing something...

Blue-Marlin, 2016-11-22:
As far as i know UBL is currently fading out the correction to a definable height. @Roxy-3D will know that for sure.

A new idea would be to fade with a fixed maximum difference. If the max hight difference on the bed is a mm the fade would take longer than if the max difference is only 0.2mm. One could define a max difference/layer - forget it - Marlin does not know about layers.

BillStruve, 2016-11-22:
Thanks that's great news!

Roxy-3D, 2016-11-22:
> UBL seems to fix bed adhesion of the first layer by moving Z up and down to match the bed. During the next few layers, does UBL modify the E rate, and reduce the correction to Z, so that the print will eventually become flat, regardless of the bed topology?

Yes, UBL attempts to track the surface of the bed by moving the nozzle up and down as it moves the nozzle across the bed.

> If the top of the completed print matches the the shape of the bed, UBL is propagating the topology of the bed to the structure, so a warped bed will produce a warped print using UBL. Maybe I'm missing something...

UBL slowly turns off the correction being applied as the nozzle moves up.   You still have the 'errors' caused be the bed not being flat on the bottom side of the print.   But the error correction is gone at the 'Fade Height'.   

I have often wondered what the 'right' thing to do is with regard to the extrusion amount.  Right now...   No correction is being applied.  And everything seems to turn out OK.   But that really doesn't make any sense.   If there is a big dip in the bed, UBL can track that dip and provide good adhesion.  But more filament is needed to fill that volume.  

I think part of the answer is the filament flows more freely when there is less resistance.  So if the nozzle is a little too high, more filament flows and the nozzle is less 'too high' on the next layer.   I really don't know the answer.  But it seems to work out OK.   

If you want to play around and put in extra (or less) filament to handle the dips (or bulges) in the bed...  I can tell you where to do it.  It all happens in Mesh_Buffer_Line.cpp   You would just look at the location of the nozzle with see if the mesh points are above or below the mean.   And then add (or subtract) some amount of filament as the nozzle passes over that mesh cell.   It actually would be a very easy (just a few lines of code) experiment to do.    But I think before you do that...  It would be useful to find a print where you can see the 'Fade Height' correction not producing a good, even density of filament.    

If you can find a model to print that is very sensitive to the filament density, then it would easier to take a scientific approach and figure out what it takes to fix this omission.

> As far as i know UBL is currently fading out the correction to a definable height. 
Yes.   That is exactly what it does.
> At least MBL used to do that. 
Oh?   I wasn't aware the original Mesh Bed Leveling did that!  I was pretty sure I was adding a new feature!

BillStruve, 2016-11-22:
@Roxy-3D Thanks for the details. Regarding 
" I think part of the answer is the filament flows more freely when there is less resistance. So if the nozzle is a little too high, more filament flows and the nozzle is less 'too high' on the next layer. I really don't know the answer. But it seems to work out  OK. "

Flow = Pressure / Resistance to flow (like I=V/R)
Resistance is the series combination of orifice resistance and after orifice resistance. The latter depends on layer height and width. Pressure comes from the filament driver (for printers using filament, the Mini Metal Maker, MMM, below, does not). The orifice resistance is much larger than the resistance after the orifice, so the actual difference in flow rates will be small during Z fading.

What is probably happening when the Z fading is applied (increase/decrease) is that the layer width changes (thinner/fatter) - probably not noticeable. Thanks for including F(ade) in the G29, since that would be my first thing to try.

The pressure in the MMM comes from a piston (E axis) pressing on a cylinder filled with paste. The paste is a mixture of fine metal particles (bronze or silver), water, and a binder. After printing, the print is dried to remove the water and put into a kiln. In the kiln, the binder goes away, and the metal particles fuse into a solid bronze or sterling silver article. It would be nice to run with a constant extrusion rate. Currently, I'm using spiral vase hollow forms in an attempt to keep the extrusion rate constant.

One of the problems with the dense paste (5 gm/cc) is that gravity tends to collapse a print with any height. To solve this, I made an attachment that surrounds the tip of the extruder with a vortex of warm air to partially dry the paste as it is being printed. In the photo below, the left print was done with no vortex, the middle one is the same print with vortex, and the right is a quad cone printed with the vortex (no support during printing).
[image]

Roxy-3D, 2016-11-22:
Very cool!   The way the Mesh Bed Leveling (and UBL uses Mesh Bed Leveling as its foundation) works is the bed gets divided into Mesh Cells.  You know the height of the bed on the four corners of the cell and based on that you can approximate the bed height at any location within the cell using bi-linear interpolation.   That sounds complicated...   But what is happening is the height of the bed is being calculated partway across the Mesh Cell on two opposite edges of the Cell.   Then a final interpolation is done using each of those heights to the location inside the Cell.   

It would seem to me you could use this information on a cell by cell basis to calculate how much extra (or less) filament you need in that location of the bed to compensate for the extra volume that needs to be filled in.   

The current Mesh_Buffer_Line() has a number of different but very similar Mesh Cell Calculation routines.   For example, has the nozzle crosses X Mesh Lines, you don't need to do the full bi-linear interpolation.  You already know the corresponding heights (as Mesh Points) for the bounding +/- Y Mesh Point.  So it becomes a simple interpolation (with much less floating point math).   

If you ignore the fact there are multiple (optimized for different situations) routines in that piece of code, it should be very easy to tell it to extrude a little more or a little less on a per cell basis.  And these routines know exactly how far off the bed is from perfect in the current location.   You should have all the information you need to tweak things in different ways.

BillStruve, 2016-11-22:
Yup, just use the difference in volume, aka area at a fixed width, and calculate the needed +- delta E. May take just a few mm to get to perfect. Bottom is always going to reflect the bed shape, so the quicker it gets perfect, the less to sand off. Good thoughts!

 Currently, Z min on the MMM is detected with 3 micro-switches under the bed and I need to resolve the differences in Z min's reported, so it will be a while before I can implement/test UBL.

Roxy-3D, 2016-11-22:
> Currently, Z min on the MMM is detected with 3 micro-switches under the bed and I need to resolve the differences in Z min's reported, so it will be a while before I can implement/test UBL.

Yes.  And in fact...  That type of issue is why I'm working to integrate the Z-BabyStepping into the UBL code base to be much quicker and easier to use.   It is possible to set up a very finely tuned mesh without even using a Z-Probe.  (Really...  You should use a Z-Probe to get as much preliminary information as possible.  But you can use UBL without a Z-Probe!!!)   The problem is, any error in the Z-Homing (and use of the Z-Probe) can still leave you off a little bit for your first layer.   So having quick and easy access to the Z-BabyStepping helps mitigate that issue.     

The point I'm trying to get to is this:   If you just ignore the reported Z and not count on it being super accurate...  It may not matter because you can just use Z-BabyStepping to dial in the homed height and get the adhesion how you need it.

thinkyhead, 2016-11-26:
> Oh? I wasn't aware the original Mesh Bed Leveling did that! I was pretty sure I was adding a new feature!

It never did. But now it does. #5299

closed 2016-12-11 by BillStruve