MarlinFirmware/Marlin #5277: Donation

DjDemonD, 2016-11-22:
A recent thread on reprap forum http://forums.reprap.org/read.php?1,724175,724998#msg-724998 has asked if we would donate to the Marlin project, with a view to moving to 32 bit. I am happy to as are several others. How do I go about it?

MagoKimbra, 2016-11-22:
I know that Scott Lahteine is working, and soon we will have an official version of Marlin on 32Bit .. Just wait ...

DjDemonD, 2016-11-22:
So donations not necessary or wanted?

boelle, 2016-11-22:
Donations are possible

@thinkyhead what are the page where we can donate?

boelle, 2016-11-22:
@DjDemonD 

https://www.gofundme.com/zmh9h9vc

boelle, 2016-11-22:
@DjDemonD and if you are able to do code then you can also help by solving bugs... those issues marked with red and orange in the issue list

or by testing if those bugs are still there

DjDemonD, 2016-11-23:
I'm not a programmer but I am happy to test things, and have done a little in the past.

thinkyhead, 2016-11-24:
Also http://patreon.com/thinkyhead 😸

thinkyhead, 2016-11-24:
P.S. I have also set up a Twitter for MarlinFirmware, which you can find at —of course— http://www.twitter.com/MarlinFirmware Follow us!

thinkyhead, 2016-11-24:
Aaaaand… I finally got it together to understand our documentation project, set up Jekyll on my computer, etc., and have updated the README to help others to get going. So if anyone wants to contribute some help please visit https://github.com/MarlinFirmware/MarlinDocumentation

@MarlinFirmware/documentation-maintainers

DjDemonD, 2016-11-24:
Thanks I'll have a think about that. Probably make a donation for now, but if you need anything tested on corexy let me know.

closed 2016-11-24 by DjDemonD