MarlinFirmware/Marlin #5278: Bowden Filament change

tibuck, 2016-11-22:
Hi team, First off Thank you for all the hard and fantastic work you been doing for use all to use a great open 3d printer firmware, with out you guys I wouldn't have a 3d printer.  on one of my printers I use a dual Bowden extruders, Doing filament change is more then a PITA, i was thinking if there was a filament change right in the menu where I could set X length  of filament to retract at a faster speed rate example extruder 1 filament change 500mm and so on, and something like that for the second extruder as-well. would that be a hard feature to  add?

Barrocas, 2016-11-22:
New feature - Change filament from the menu when not printing #5262

-- 
Cumprimentos,
D.er Hugo Barrocas

tibuck, 2016-11-22:
Great ill give that a try thanks a bunch.

tibuck, 2016-11-22:
hahaha looks like that was already add to configuration_adv.h Whooo hooooo i'm a happy man . only re-enforces  that the Marlin team is hard at work!

tibuck, 2016-11-22:
I tried adding the codes to marlin but when I compile it gives me errors maybe in not putting everything exactly where they should be??

thinkyhead, 2017-10-31:
Sorry this issue got lost. I was indisposed during this period. I'm just going through old issues now to resolve them. Please re-open if this is still something needing help!

closed 2017-10-31 by thinkyhead