MarlinFirmware/Marlin #5280: Vote for feature requests??

boelle, 2016-11-23:
I looked through the issues a few days ago and faint remember that someone asked if it would mane any sense to have a vote thing going on..

Since we see less bugs i thought it would make sense to bring this up again

Ie vote for the features and do those with most votes first

or

do the old ones first?

Roxy-3D, 2016-11-23:
Or...   The people writing code will try to accommodate people requesting features based on three things:
- How hard the feature is to implement. 
- How important the feature is to those requesting it.
- How interesting the feature is to the developer actually implementing it.

boelle, 2016-11-26:
i could go with that...

maybe rewrite this one: How important the feature is to those requesting it.

to

How important the feature is to those requesting it. And how much it would benefit others?

closed 2016-12-06 by boelle