MarlinFirmware/Marlin #5281: Advantage of TX_BUFFER_SIZE (Doku question)

Sebastianv650, 2016-11-23:
For me it's not clear whats the advantage of TX_BUFFER_SIZE. After reading the comment regarding it, there is the feeling it would only eat up memory.

But I guess that's not true ;-)
Is it right that without a TX buffer Marlin will block the main loop somewhere to wait until it has sent something like the "OK" messages. This will take time depending on the selected baud rate.
On the other side, investing some bytes, Marlin would add the string like the "OK" only to the buffer, not blocking the main loop. The string would then be sent byte per byte by an ISR from the buffer.

If that's true, maybe someone knowing the function should update the comments to make it clear for everybody that we can save some cycles using this option.

landodragon141, 2016-11-25:
I would also like to know what this does so I can improve the documentation for it (and for personal use). Can we get clarifications on these lines:
`//===========================================================================
//================================= Buffers =================================
//===========================================================================

// @section hidden

// The number of linear motions that can be in the plan at any give time.
// THE BLOCK_BUFFER_SIZE NEEDS TO BE A POWER OF 2, i.g. 8,16,32 because shifts and ors are used to do the ring-buffering.
#if ENABLED(SDSUPPORT)
  #define BLOCK_BUFFER_SIZE 16 // SD,LCD,Buttons take more memory, block buffer needs to be smaller
#else
  #define BLOCK_BUFFER_SIZE 16 // maximize block buffer
#endif

// @section serial

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 96
#define BUFSIZE 4

// Transfer Buffer Size
// To save 386 bytes of PROGMEM (and TX_BUFFER_SIZE+3 bytes of RAM) set to 0.
// To buffer a simple "ok" you need 4 bytes.
// For ADVANCED_OK (M105) you need 32 bytes.
// For debug-echo: 128 bytes for the optimal speed.
// Other output doesn't need to be that speedy.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256]
#define TX_BUFFER_SIZE 32

// Enable an emergency-command parser to intercept certain commands as they
// enter the serial receive buffer, so they cannot be blocked.
// Currently handles M108, M112, M410
// Does not work on boards using AT90USB (USBCON) processors!
#define EMERGENCY_PARSER

// Bad Serial-connections can miss a received command by sending an 'ok'
// Therefore some clients abort after 30 seconds in a timeout.
// Some other clients start sending commands while receiving a 'wait'.
// This "wait" is only sent when the buffer is empty. 1 second is a good value here.
#define NO_TIMEOUTS 1000 // Milliseconds

// Some clients will have this feature soon. This could make the NO_TIMEOUTS unnecessary.
#define ADVANCED_OK
`

Blue-Marlin, 2016-11-25:
The original implementation that is still used when TX_BUFFER_SIZE 0 0 or undefined is in MarlinSerial.cpp line 275++
```
#else
  void MarlinSerial::write(uint8_t c) {
    while (!TEST(M_UCSRxA, M_UDREx))
      ;
    M_UDRx = c;
  }
#endif
```
"Wait until the previous char is sent. If so, send the next."
When sending more than one char, v.e. a string, this results in busy waiting.

The new implementation uses a buffer and an interrupt.
If the buffer is too small for the string to send, this will again cause busy waiting until the complete string is fed into the buffer.

landodragon141, 2016-11-25:
What is the real world effect of different settings? I assume that a larger buffer is faster. What are the pros and cons to different sizes?

Blue-Marlin, 2016-11-25:
Assume there is no buffer. Now we send "ok\n". One byte is pushed into the __register__ imminently, Then we have to wait until this byte is sent. We push the 'k', wait until sent and finally can push the '\n'.
Now lets assume we have 4 bytes of buffer. If we send "ok\n" this will fit entirely into the __buffer__ and we can return immediately. But if we try to send "This is a long echo" only the "This" fits into the buffer. When the 'T' is transfered we can push the next ' ' into the buffer and when 'h' is transferred we can push the 'f' - and so on. So we are waiting for strlen("This is a long echo") - TX_BUFFER_SIZE transfer times. 
So if you want to send the bible without waiting gfor it you'd need a some MB large buffer. Ok - you don't want to send the bible, but maybe something a bit smaller. 
If you don't want to wait, the buffer must have the size of the data you want to send at once. (And if sending an other big chunk of data directly after the firs, the added sizes of the chunks.)

Having a big buffer will enable you to send larger amounts of data without waiting.
Having a big buffer is a waste of memory if you use it for sending "ok\n"'s only.

landodragon141, 2016-11-25:
I like the seasonal metaphor (do you know this from experience?). That's exactly what I was looking for thank you! I also have a question about the command buffer. For one why 96? For two what is BUFSIZE for? Are these values that can/should be changed?

Blue-Marlin, 2016-11-25:
Yesterdays host planed for Bud Spencer, Terence Hill and Obelix - but they did not show up.

If you could not alter the values they wouldn't be configurable.
The largest possible commandline  has to fit into MAX_CMD_SIZE (with stripped comments).
The input buffer can hold BUFSIZE commandlines to smooth the input stream. If we receive, and acknowledge that by sending "ok\n" it will last some time until the hos registers that and sends the next line. We have to wait until this line arrives. If the host knows about the buffers, he can send BUFSIZE lines in advance. Not all hosts can use this. (Disclaimer: This is a light weight explanation - the reality is much more complex)

landodragon141, 2016-11-26:
That's fine I'm just trying to get a better understanding of how these work so I can write some good documentation for it.

closed 2017-10-31 by thinkyhead