MarlinFirmware/Marlin #5282: Q: Difference LIN_ADVANCE_K and  EXTRUDER_ADVANCE_K?

Sungod3000, 2016-11-23:
Hi,

I just configured the pressure advance feature and I see two variables. The comment block explains how to get the K value but to which variable does it apply?

Cheers

Kaibob2, 2016-11-23:
If you are on RCBugFix then this formula 
```
 * Filament diameter           |   1.75mm  |    3.0mm   |
 * ----------------------------|-----------|------------|
 * Stiff filament (PLA)        | K=47*L/10 | K=139*L/10 |
 * Softer filament (ABS, nGen) | K=88*L/10 | K=260*L/10 |
```
is for estimating the K value for `#define LIN_ADVANCE_K 0`

Better use `#define LIN_ADVANCE` as //#define` ADVANCE` isn't working as well as Lin_advance.
Maybe take a look at this, quite recent, thread #5222

Sebastianv650, 2016-11-23:
There are two functions inside Marlin that tries to compensate nozzle pressure effects. The older, and as far as I know not longer maintained ADVANCE and my newer LIN_ADVANCE.

I don't recommend using ADVANCE, even if the main bug why it was not working for a long time is fixed will not always work as expected.
Regarding LIN_ADVANCE, please read [the documentation](http://marlinfw.org/docs/features/lin_advance.html). The page is under construction, that's why you might get a warning about a not signed certificate. There you will especialy learn how to calibrate K, please see the given formula only as a reference. Up to now there is not enough feedback to be sure if the formular is working or completely wrong.

The main difference between the older ADVANCE and LIN_ADVANCE is the way how the extra extruder steps are distributed during acceleration and deceleration. ADVANCE (as also JKN does) assumes a linear distribution along each axis step which means equal distance steps. LIN_ADVANCE distributes them equaly over time, which leads to a better result in the printed part. This also has some drawbacks, for example bowden systems may not always be capable doing the fast extruder movements necessary. The longer the free filament length, the more the extruder has to compensate.

I will try to answer further questions when they rises ;)

Sungod3000, 2016-11-24:
Great, thanks for the fast answers, Ill test it soon.

closed 2017-10-31 by thinkyhead