MarlinFirmware/Marlin #5287: Mesh Bed Leveling:  Mesh pts saving correctly?

scott1326, 2016-11-23:
Just flashed the ubl marlin version, and it looks to be a great update.  One issue I think I'm having:  I'm probing 5x5 pts on a 200x200mm bed.  Origin is 0,0 in lower, left corner, however, y-limit switch is at the top end (y=200).  Probe is in a neg y direction (X_PROBE_OFFSET... 0, Y_PROBE_OFFSET... 40).  So I can't probe pts where y<~40 because probe will be off the bed.  g29 p1 probes 20 pts but not where y=0, which is fine, I can fill those manually.  However, a g29 o command shows mesh pts in the first four rows, and the last row (y=190, actually) unfilled.  Problem?

scott1326, 2016-11-23:
Also, is there a way to change the z-axis when performing a g29 p4 command?  I can change the value with the lcd knob, but I can't actually change the extruder height that way and test the actual height.

Roxy-3D, 2016-11-23:
It maybe the code is deciding a 0.0 offset for X makes that column unprobable.  I'll look at that.   But if you just fill those values in...  And do a Mesh Validation Print with G26 P C O 2.0   You can then edit the mesh with a G29 P4 R

I originally had the Z-Axis going up and down as you edit the mesh.  But it is problematic for a bunch of reasons.  You can pretty much look at how the lines are squished or not squished and say "That would be better if it was .1mm higher."   And that you can do initially with the LCD Display.  But after you have done a few Mesh Points, you almost don't even have to look at the LCD Panel.  You just see where the nozzle is, and decide which way and how much to turn the dial.

scott1326, 2016-11-23:
Hi Roxy, thank you for your prompt reply.  My concern is that the whole mesh is getting shifted down a row, and thus everything is off.  The test prints are not very close.

Roxy-3D, 2016-11-23:
Can you post a picture of your G29 M  after doing the initial G29 P1 ?    And then fill the unprobed areas of the Mesh and show me the Mesh Validation Print on the bed after you do a G26 P C O2.0 ?  That would help me understand what you are seeing.   (You can save an extra copy of your current mesh in an unused mesh slot so you don't loose any progress.  But it would be helpful to see those things as you create and validate the mesh.)

scott1326, 2016-11-23:
No problem:

>>>g29 p1
SENDING:G29 O
Bed Topography Report:
(0,4)                                                                        (4,4)
(0,200)                                                                    (200,200)
       .                 .                 .                 .                 .

 -1.07725    -0.43700   [-0.04450]   0.18400    -0.00925

 -0.97875    -0.36525    -0.01300     0.16300    -0.10300

 -0.90875    -0.27300     0.20075     0.38750     0.12725

 -0.77875    -0.21400     0.28175     0.33425     0.20000
(0,0)                                                                        (200,0)
(0,0)                                                                           (4,0)

>>>g29 p3 r c.5
SENDING:G29 P3 R C.5
>>>g29 o
SENDING:G29 O
Bed Topography Report:
(0,4)                                                                        (4,4)
(0,200)                                                                    (200,200)
  0.50000     0.50000     0.50000     0.50000     0.50000

 -1.07725    -0.43700   [-0.04450]   0.18400    -0.00925

 -0.97875    -0.36525    -0.01300     0.16300    -0.10300

 -0.90875    -0.27300     0.20075     0.38750     0.12725

 -0.77875    -0.21400     0.28175     0.33425     0.20000
(0,0)                                                                        (200,0)
(0,0)                                                                           (4,0)

Picture of test print is attached.

Incidentally, after both g29 p1 and g26, pronterface showed TEST, and lcd screen showed M105(?), and I couldn't issue any more commands until I pressed lcd button.

[image]

Roxy-3D, 2016-11-24:
> Incidentally, after both g29 p1 and g26, pronterface showed TEST, and lcd screen showed M105(?), and I couldn't issue any more commands until I pressed lcd button.

That doesn't sound right.  We will need to figure out what you are seeing.   

Can you tell me you nozzle diameter?  It looks huge.  But if it is not huge, that means the nozzle is too close to the bed and you will want to edit those Mesh Points and raise them up a little bit.   I can't really see why you are not getting good adhesion.  Maybe it is because the nozzle is too far from the bed and the filament isn't sticking.  In that case you would want to edit the mesh points and give them a more negative number.   Maybe it is because the nozzle is too close to the bed and no filament is being extruded.  In that case you would want to edit the mesh points and raise up the number more positive.

My suggestion is you do a G29 P4 R and just raise or lower (as appropriate) each mesh point by .1mm and then do a G29 A and then a G29 S1 to save the mesh in Slot 1. 

If you do another G26 P C O2.0 you will probably see improvement, and get a better feel for how much you have to raise or lower trouble points.   If the nozzle is not putting out the right amount of filament, you can adjust that.   That might be part of the problem.  It is hard to tell from the pictures.

But doing some corrections on the Mesh and doing another G26 is probably going to show progress.

scott1326, 2016-11-24:
Thanks for your help, Roxy.  Nozzle diameter is .35mm.  I was aware the mesh was off, but before I started to edit it, I wanted to make sure the points were lined up with what I was seeing.  Incidentally, I think I read a post of yours before that said more negative numbers will lower the mesh (make the nozzle farther from the bed), and more positive numbers will raise the mesh.  Is this correct, or do I have it backwards?  Going out of town for the holidays, will play with it more when I return.  Happy Thanksgiving!

Roxy-3D, 2016-11-24:
No...  A more negative number will lower the nozzle as it travels over that point.  A more positive number will raise the nozzle.    Admitably...  You are going to be able to find a few posts where I said the exact opposite.  This stuff gets confusing!!!!

Have a good Turkey Day!  Eat lots of Turkey with piles of gravy dripping off of it!!!!   :)

closed 2017-10-31 by thinkyhead