MarlinFirmware/Marlin #5293: RCBugFix: Bootloop on splashscreen when SD Card Support is enabled

Careless, 2016-11-25:
Hi

having an issue where I got my firmware that i've set up to run great, except the only feature I can't seem to utilize at this point is the SD Card support for the RepRap Smart LCD Controller with the EXP1 and EXP2 ribbon cables.

It's connected to an MKS Gen 1.4.

I tried enabling both SD Card Support, as well as SD Card CRC Checking... and it just shows the Marlin splash screen with the logo square, and then reboots, over and over again- continuing to show the same splash screen every time before rebooting.

disabling both features enables the LCD controller and printer to boot and function properly.

I noticed in the RC6 firmware that was supplied with the printer, it had the DOGLCD enabled as well, and I assume it has the U8glib file to allow that to work- but is that option necessary or related?

the LCD section looks nearly the same as RC6, so I just mimic'd the settings by going over them and making them match where they options are defined and whatnot.

has anyone had this issue?

landodragon141, 2016-11-25:
Do you have a SD card inserted?

Careless, 2016-11-25:
Tried with and without.

Same ordeal.

Careless, 2016-11-26:
Updating this issue as SOLVED due to my foolishness.

Landon, your question made me wonder; I use a MicroSD Card adapter. And on
my PC and Mac, it does not detect the adapter as "Card Inserted" when it is
not populated with a MicroSD in the adapter card slot.

I removed the adapter from the slot, reflashed my compiled version of
RCBugFix with SD support enabled, And it seems to work fine now.

Error on my part. I didn't think the adapter was detected as an erroneous
card format of somesort when not accompanied by a MicroSD card. I was under
the impression that the SD Card Adapter was just an array of contact vias
reduced the MicroSD size... Though I'm not sure if SD Card slots are
manufactured with a detent switch of some-sort to detect whether something
is inserted and it just can't read it.

Anyways... problem solved.

Not a firmware issue.

Thanks, Landon.

closed 2016-11-26 by thinkyhead