MarlinFirmware/Marlin #5296: G28 w/ABL diagonal movement too fast

NoNaym, 2016-11-25:
I've been struggling trying to find the bit of code inside Marlin that effects the speed of the X/Y movement from X0 Y0 to bed center when running G28. I have an ABL sensor installed and working fairly well, but the diagonal movement to the center of the bed is occurring way too fast and causes my Y-steppers to skip every time (maybe X too, but tough to tell). I have tried changing just about every part of the code that I could find that seemed like it might make a difference, but no dice! While it may not be the best illustration, I have a video of the issue here: https://goo.gl/S2vmDU

This is more or less the copy of Marlin I am using: https://bitbucket.org/NoNaym/marlin-tam-series-1-abl/src

Does anybody have any definitive answers to how I might go about slowing that one movement down?

Thanks in advance!!

Roxy-3D, 2016-11-25:
If you are missing steps...  You probably have something set too high.  But it could be any one of a few things.  Try setting the Y axis (the 2nd number) lower in each of these lines.   Probably, just change one at a time and restore the previous version each time to see if you can find the item that is too high.
```cpp
#define HOMING_FEEDRATE {40*60, 50*60, 50, 0}  // set the homing speeds (mm/min)
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,52.2}      // steps per unit for FolgerTech i3-2020
#define DEFAULT_MAX_FEEDRATE          {250, 250, 2, 17}    // (mm/sec
#define DEFAULT_MAX_ACCELERATION      {1000,1000,8,750}    // X, Y, Z, E maximum start speed for accelerated moves. 
```

NoNaym, 2016-11-25:
Thanks @Roxy-3D,

So after dropping the DEFAULT_MAX_ACCELERATION for Y, I have stopped the skipping from occurring during that G28 diagonal movement.

My question to you now is this: If the only time missed steps are occurring is during the diagonal movement of the G28 command, why can't this one movement be slowed down instead of dropping the acceleration for all Y movements? What I'm getting at is the fact that my machine has not experienced any skipped steps during any print moves... only during the one diagonal movement to the center of the print bed during G28 does it skip.

Roxy-3D, 2016-11-25:
I understand the question.   I'm not sure I can answer it.  Somebody else will probably jump in with a more complete answer.    Here is the the thing:  The Planner code has to look at the move being requested and make sure no axis involved in the move has its parameters exceeded.    In the move you are talking about, both the X & Y axis have components contributing to the move.

The one thing that is unique about this move is it has a very large delta on both X & Y.  That is something you never see on most print moves.    I'm wondering if it is a Max Feed Rate issue instead.  By bumping down the acceleration, are you keeping the Max Feed Rate from being exceeded on this very long move?

I'm wonder if you should try going back to your original settings that failed, and bump the Max Feed Rate down but leave the acceleration alone.    My guess is you might be able to eliminate the lost steps that way.  And if so...  This may not be an acceleration problem.

NoNaym, 2016-11-25:
Okay, yes... I just tried restoring my original acceleration settings, and dropping the Max Feed Rate for Y down. This also appears to stop the skipping from occurring. So this being the case, and if "this may not be an acceleration problem," then is dropping the Max Feed Rate for Y going to be my best bet? Or is there something else going on here that might need to be adjusted to get just that movement speed to change?

Sorry I only have a partial understanding of how everything about Marlin works.... thus my reaching out :)

Thanks again!

thinkyhead, 2016-11-25:
GCode produced by the slicer for printing is almost always constrained, first by the slicer settings themselves, and then by the slowest axis. In most cases, while extruding, this will be the E axis.

Homing moves are not constrained, but use the `HOMING_FEEDRATE_XY` speed. If that speed is too high for the steppers to manage the torque, then the max feedrate (by default `DEFAULT_MAX_FEEDRATE`) is possibly set too high. You want to make sure it's constrained so that a move like `G1 X100 Y100 F9999` will work.

thinkyhead, 2016-11-25:
Another thing to check is your stepper driver current. For most NEMA17 steppers, the current should be set to around 0.9V. See [this video](https://www.pololu.com/blog/484/video-setting-the-current-limit-on-pololu-stepper-motor-driver-carriers) for the basics.

NoNaym, 2016-11-26:
Thanks for the info @thinkyhead! Not sure if you saw my previous responses, but there were definitely some improvements noticed when adjusting those two settings independently.

My stepper driver currents are correct, so I'm not really concerned about them. The main issue I'm experiencing is definitely being caused by firmware settings and/or whatever calculates that one (and only) movement that causes some skipping.

I don't think this makes a difference, but I'm using dual Y steppers in this configuration... which is the axis that skips during that fast diagonal movement.

thinkyhead, 2016-11-26:
I'm glad to know you got it resolved. If using dual Y steppers, make sure everything is aligned well or you may experience binding, and that can also limit your speed.

closed 2016-11-26 by thinkyhead