MarlinFirmware/Marlin #5302: CoreXY Z Axis won't move during print

MH15, 2016-11-26:
Hello everyone! I've been building a custom CoreXY printer and just finished assembling it. Through the software I can move the Z axis perfectly fine; the endstops work as well (I have the Full Graphic Smart Controller). The problem arises when I try to print something. The bed won't move so the printer keeps piling more and more layers on top of each other.

Anybody know why the printer might be doing this? 

Thanks!

landodragon141, 2016-11-26:
Is your Z axis inverted? The g-code might be trying to push it up when it's already at the endstop.

MH15, 2016-11-26:
It might be... when I was assembling the printer I reversed the z motor plug on the RAMPS unit because the display read 200mm when the bed was at the extruder (the position I wanted it to be 0mm, if that makes any sense). Let me go check that.

MH15, 2016-11-26:
How would you suggest fixing that in the firmware

landodragon141, 2016-11-26:
Invert the value of this line under motor movement:
`#define INVERT_Z_DIR true`

landodragon141, 2016-11-26:
i.e. if it's currently true change it to false and vice versa.

MH15, 2016-11-26:
I changed that and the z axis is now reading the proper position (0 when bed is at extruder, 200 when it is at the bottom) but during printing the screen says the z axis is at 0.3mm and not moving.

MH15, 2016-11-26:
Well I fixed this issue, your code and some other things helped. On to the next!

Thanks

closed 2016-11-27 by thinkyhead