MarlinFirmware/Marlin #5305: Feature requests: 1)option to define PARK_AFTER_HOMING and CLEAN_PARK_AFTER_HOMING 2) add logic to turn on PS if off for gcodes requiring PS.

macst34, 2016-11-26:
1)  PARK_AFTER_HOMING and CLEAN_PARK_AFTER_HOMING

Use case:
I'm running a self built prusa-derivative.  My X,Y endstops are fixed past the range of the build plate. After homing, my printhead is off the build plate in both the X and Y axes and just hangs out in midair. If I heat the hotend to prep for a print, it'll start oozing and then I start cursing and move it over to the corner of the buildplate.

To compensate for being off the build, I define:
```
#define X_MIN_POS -1
#define Y_MIN_POS -16
#define Z_MIN_POS 0
```
which corresponds to the endstop positions and puts 0,0,0 to be the front left corner of my build plate. Since we already have NOZZLE_PARK coded, can we add a define to allow the option of automatic parking after homing? That way I don't have to curse if I heat the hotend before moving it to 
` #define NOZZLE_PARK_POINT {0,0,0}`
 
While we are at it, can we tack on the ability to do a nozzle clean before parking since NOZZLE_CLEAN_FEATURE is already coded?

I think the the logic would go something like this:

> M80           ;turn on PS
> G28            ;home
> G1 Z8         ;raise printhead by 8mm
> G27            ;park 
> M109 S100 ;heat to 100C (or whatever temp needed for cleaning)
> G12            ;nozzle clean
> M104 S0    ;cooldown.
> G27            ;park

-----------------------------------------------

2) Check if PS is on before doing any commands that require the PS then turn on the PS before the command if it is not on.

Usecase:
I am using a 5v PS that is on all the time to power my mega2560. I have another 12v PS that I turn on via `M80 ;PS ON` when I want to print something. I also added a `M81; PS OFF` to my end script to turn it off after the print. I usually keep the 12v PS off when idle since it has a fan and it's noisy. Sometimes I forget and issue commands while the PS is still off. This results in unexpected behaviors from the printer and additional cursing from me. For example, axis moves get updated but the printhead does not move since the steppers are not powered.  Worse yet is when I home Z without power and it freezes since it is waiting for the Z endstop to change state,

```
//#define TURN_ON_PS_BEFORE_COMMANDS ; If not on, turns on the PS before executing commands
requireing it.
```

Thanks,
Michael

thinkyhead, 2016-11-27:
These are all things which can be done by adding GCode to your slicer profile. We don't want to bloat Marlin with things that can already be accomplished with the available options.

macst34, 2016-11-27:
Hi @thinkyhead, since these are defines, they would only be compiled into the code of the people who would like these conveniences. I realize Marlin is very much still in hackerspace, but little changes like this may make it more idiotproof and increases usability for less experienced 3dprinter users. Perhaps other people may also encounter the use cases that I mentioned. The convenience would be worth the tradeoff for us.

thanks,

Michael

thinkyhead, 2016-11-28:
I can only reiterate what I already said. This can be accomplished without any change to the firmware. Ideally we want Marlin to be as "dumb" as possible, and merely carry out the commands as-given. I disagree with features that extend the behavior of well-defined procedures like `G28` to do more than just obtain the current position.

> If I heat the hotend to prep for a print, it'll start oozing and then I start cursing and move it over to the corner of the buildplate.

You can stop cursing and use `Z_SAFE_HOMING`. Just set the `Z_SAFE_HOMING_X_POINT` and `Z_SAFE_HOMING_Y_POINT` to where you want the nozzle to be after homing.

macst34, 2016-11-28:
@thinkyhead, cool, thanks. I'll go try out `Z_SAFE_HOMING' when i get a chance.

macst34, 2016-11-28:
@thinkyhead 
Z_safe_homing solved the dangling hotend issue. Thanks. I also saw that the PS_on issue is referenced in 
https://github.com/MarlinFirmware/Marlin/issues/4142

so i'm happy to close this FR out.

Thanks!

Michael

closed 2016-11-28 by macst34

macst34, 2016-11-29:
as an addendum, using Z_safe_homing solved the dangling hotend issue, but prevented me from doing manual z homing at other points besides x0, y0. To reallow this feature, I ended up editing Marlin_main.cpp line 3077

          // Let's see if X and Y are homed
          //if (axis_unhomed_error(true, true, false)) return;

thinkyhead, 2016-12-02:
Thanks for the update @macst34. Of course, there's a safety reason why `Z_SAFE_HOMING` prevents homing Z by itself when you haven't already homed X and Y. At that point the machine has no way to determine if it's safe.

Cheers!