MarlinFirmware/Marlin #5306: Marlin always throws MINTEMP error

MH15, 2016-11-26:
My thermistor is plugged in to T0 on the RAMPS and reads about 100 ohms with a multi meter ( the correct value), so I'm not sure why it does this.

Roxy-3D, 2016-11-26:
Marlin isn't seeing the thermistor.    Do you have a heated bed and a Thermistor for it to?   Has the printer ever worked correctly with the current configuration?

There are 3 thermistor pin pairs on the RAMPS board.  You should try moving the connector to each of the other pin pairs and see if anything changes.

MH15, 2016-11-26:
I don't have a heated bed (only thermistor is connected to an E3D V6) and I will try to swap the pins.

MH15, 2016-11-26:
What code would that be?

Roxy-3D, 2016-11-26:
No...  Not code...   Try moving the thermistor connector to each of the other two sockets on the RAMPS board.   And before you move it off of T0 (which is where it is supposed to be), take a good look and make sure you have it plugged on there correctly.

MH15, 2016-11-26:
It gives a mintemp on T2 and a maxtemp on T1

Grogyan, 2016-11-26:
Do you have the correct thermistor declared in the configuration.h?
100 Ohms is awfully low, typically thermistors are 100K ohms

Blue-Marlin, 2016-11-26:
http://wiki.e3d-online.com/wiki/E3D_PT100_Amplifier_Documentation ?

MH15, 2016-11-26:
Would you suggest doing that, Blue Marlin

MH15, 2016-11-27:
Oh, and they are actually 100k ohms not 100 ohms

Fulg, 2016-11-29:
> Would you suggest doing that, Blue Marlin?

Because you said you were reading 100 ohms, in which case this isn't a thermistor but a PT100 sensor. E3D's PT100 and Thermistor sensors in cartridge form both look the same (I believe the difference is the wire color and/or length) so you can easily mistake one for the other.

Since you then corrected yourself and confirmed you were reading **100k** ohms, then it is indeed a thermistor. MINTEMP usually means a disconnected thermistor, so either you have a wiring problem, you are plugging into the wrong pin on the RAMPS (so Marlin isn't seeing it), or you have configured the firmware to read the wrong pin (wrong board selected?).

You should attach your Configuration.h file.

closed 2017-10-31 by thinkyhead