MarlinFirmware/Marlin #5308: Configuring for a MAX 6675 - Error Message

pezzer2003, 2016-11-27:
I updated to the new firmware as I purchased a 12V Single-mouth Nozzle Extruder Heating Print Head With a K-type thermocouple (PT100 reference temperature of 0 degrees) and a Type K SPI Interface Thermocouple Temperature Sensor Module for Arduino MAX6675 from EBAY.
I changed the Configuration.h to: 
#define TEMP_SENSOR_0 -2
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1
And it returned the Error Message: 'DIODIO53_DDR' was not declared in this scope

I change it to -3 and the same Error Message.
My setup is : Ramps 1.4, RepRap Discount Smart Controller, And the above extruder.

I'm not shaw what else I need to set to make this all work?
Here is the full Error Message. It's Really long.

In file included from sketch\MarlinConfig.h:26:0,
                 from sketch\Marlin.h:36,
                 from sketch\temperature.cpp:27:
sketch\temperature.cpp: In member function 'void Temperature::init()':
fastio.h:74: error: 'DIODIO53_DDR' was not declared in this scope
 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)
                            ^
sketch\fastio.h:97:25: note: in expansion of macro '_SET_OUTPUT'
 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)
                         ^
sketch\fastio.h:108:28: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
                            ^
sketch\temperature.cpp:1026:7: note: in expansion of macro 'OUT_WRITE'
       OUT_WRITE(SS_PIN, HIGH);
       ^
In file included from sketch\MarlinConfig.h:26:0,
                 from sketch\Marlin.h:36,
                 from sketch\temperature.cpp:27:
fastio.h:74: error: 'DIODIO53_PIN' was not declared in this scope
 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)
                                                         ^
sketch\fastio.h:38:28: note: in definition of macro 'MASK'
   #define MASK(PIN)  (1 << PIN)
                            ^
sketch\fastio.h:97:25: note: in expansion of macro '_SET_OUTPUT'
 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)
                         ^
sketch\fastio.h:108:28: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
                            ^
sketch\temperature.cpp:1026:7: note: in expansion of macro 'OUT_WRITE'
       OUT_WRITE(SS_PIN, HIGH);
       ^
In file included from sketch\MarlinConfig.h:26:0,
                 from sketch\Marlin.h:36,
                 from sketch\temperature.cpp:27:
fastio.h:66: error: 'DIODIO53_RPORT' was not declared in this scope
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                    ^
sketch\fastio.h:89:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
sketch\fastio.h:108:44: note: in expansion of macro 'WRITE'
 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
                                            ^
sketch\temperature.cpp:1026:7: note: in expansion of macro 'OUT_WRITE'
       OUT_WRITE(SS_PIN, HIGH);
       ^
fastio.h:55: error: 'DIODIO53_WPORT' was not declared in this scope
                                          {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } \
                                           ^
sketch\fastio.h:66:80: note: in expansion of macro '_WRITE_C'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                ^
[... 112 lines cut ...]
 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
                                            ^
sketch\temperature.cpp:1026:7: note: in expansion of macro 'OUT_WRITE'
       OUT_WRITE(SS_PIN, HIGH);
       ^
fastio.h:51: error: 'DIODIO53_PIN' was not declared in this scope
 #define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                                                                               ^
sketch\fastio.h:38:28: note: in definition of macro 'MASK'
   #define MASK(PIN)  (1 << PIN)
                            ^
sketch\fastio.h:66:105: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                                         ^
sketch\fastio.h:89:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
sketch\fastio.h:108:44: note: in expansion of macro 'WRITE'
 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
                                            ^
sketch\temperature.cpp:1026:7: note: in expansion of macro 'OUT_WRITE'
       OUT_WRITE(SS_PIN, HIGH);
       ^
exit status 1
'DIODIO53_DDR' was not declared in this scope

Grogyan, 2016-11-27:
Please use the latest RCBugFix

thinkyhead, 2016-11-28:
That kind of error message indicates a needed pin has not been defined. As suggested, if you're not using `RCBugFix` then give that a try. We might have already fixed the issue causing the error.

pezzer2003, 2016-11-28:
Thanks  Grogyan and thinkhead, that fixed it. 

Happy printing,

PEZ

closed 2016-11-28 by pezzer2003