MarlinFirmware/Marlin #5311: auto level grid wrong

kaliboe, 2016-11-27:
sorry for my English but let me explane
I finally configure my printer from scratch, all working until today, my prints are in mirror?
So I surf the internet looking for answers, there a find that I need to reverse the Y steppenmotor.
Now I need to change also the Y home position, after some time the extruder is working fine with the g28 command. Now I want to start g29 command for auto bed leveling ... here is goes wrong, the last 3 point are almost of the bed.
Before I did this change the auto bed was working fine only thing was that the object were in mirroring.
who can help me out here?

thinkyhead, 2016-11-28:
If your prints are mirrored it means that one of your axes is backward. Your bed configuration should be one of the following when viewed from the top:

With endstops mounted on the left and rear…
```
{ X0 Y200 }     { X200 Y200 }
    ^
    |
    |
    |
{ X0 Y0 } ----> { X200 Y0 }
```
With endstops mounted on the right and front…
```
{ X200 Y0 } <---- { X0 Y0 }
                      |
                      |
                      |
                      v
{ X200 Y200 }     { X0 Y200 }
```

kaliboe, 2016-11-28:
What I find strange is that the printer was working fine with pronterface, when I press +y the bed moves to the back, -x the nozzle moves to the left, +z the nozzle goes up. how is it possible then to have mirror prints if the printer follow pronterface correct?

So my setup is endstop x on the right side, endstop y in the back and auto bed leveling with servo on z
I am using RCbugfix version of marlin

thinkyhead, 2016-11-28:
+Y should move the bed forward, not back.

kaliboe, 2016-11-28:
ok, done, it's configure like this now,.
g28 homeing is working correct
now the auto bed leveling is wrong, normal it starts left in the back and will measure 3 points, then come to the middle left and measure 3 points and then front and measure 3 times. What happens now, it start almost in the middle left measure 3 times then goes to the back measure 3 times and then it goes almost of the bed! how can I correct this, before I change the settings the auto bed was working fine.

kaliboe, 2016-11-28:
anyone who can help me out here?

christianh17, 2016-11-29:
Perhaps you should attach your configuration.h.
Without it is hard to understand.
Regards
Christian

kaliboe, 2016-11-29:
thx, attacht you find my configuration.h file, just remove .txt at the end.
I have played with this settings:
#define X_PROBE_OFFSET_FROM_EXTRUDER 26 // X offset: -left +right [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 16 // Y offset: -front +behind [the nozzle]

Now it's better but the inly thing is that the nozzle need to go 1,5cm to the front to have it at the center of the bed.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/618420/Configuration.h.txt)

christianh17, 2016-11-29:
Hi kaliboe!

In your configuration I find:

#define X_PROBE_OFFSET_FROM_EXTRUDER -26  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -16  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -10.20   // Z offset: -below +above  [the nozzle]

This means:

#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0

#define X_MAX_POS 200
#define Y_MAX_POS 180 
#define Z_MAX_POS 200

#define LEFT_PROBE_BED_POSITION 50
#define RIGHT_PROBE_BED_POSITION 150
#define FRONT_PROBE_BED_POSITION 50
#define BACK_PROBE_BED_POSITION 150

That is an interesting configuration - I do not really understand how your . Do you have a display? What does it display for x and y on the first 6 testing points?

Other option: I personally did the testing for the settings with 
G29 P3 T V4
Using this the positions and measurements are logged via the serial port. There you can see the positions during testing. With the logged positions I hope I will have an idea.

Regards
Christian

kaliboe, 2016-11-30:
I solved the problem with the nozzle 1,5cm behind the middle point
When you change the size of your bed in Marlin you need to enter the same in your slicer program, that was my mistake! 
Christian thanks for your help!

closed 2016-11-30 by kaliboe