MarlinFirmware/Marlin #5312: PLA and ABS Preheat Settings maxtemp error

Kaibob2, 2016-11-27:
When i edit the PLA and ABS Preheat temperatures via the LCD the Maximum temperature for `HEATER_0_MAXTEMP` is used for both Materials. In my case Heater 0 is PLA and Heater 1 is ABS.
HEATER_0_MAXTEMP = 250°
HEATER_1_MAXTEMP = 300°

In the preheat settings i can set PLA to maximum 235° (250-15) which is correct, but ABS too.

Jinxycob, 2016-11-27:
have a peek at  line 260 in the config.h  whats those at, by default its 275C ?

Kaibob2, 2016-11-27:
@Jinxycob This is correct, the default is 275. But what is your point. 
I suspect the issue somewhere here in ultralcd.cpp 
```

  void _lcd_control_temperature_preheat_settings_menu(uint8_t material) {
    START_MENU();
    MENU_BACK(MSG_TEMPERATURE);
    MENU_ITEM_EDIT(int3, MSG_FAN_SPEED, &lcd_preheat_fan_speed[material], 0, 255);
    #if TEMP_SENSOR_0 != 0
      MENU_ITEM_EDIT(int3, MSG_NOZZLE, &lcd_preheat_hotend_temp[material], HEATER_0_MINTEMP, HEATER_0_MAXTEMP - 15);
    #endif
    #if TEMP_SENSOR_BED != 0
      MENU_ITEM_EDIT(int3, MSG_BED, &lcd_preheat_bed_temp[material], BED_MINTEMP, BED_MAXTEMP - 15);
    #endif
    #if ENABLED(EEPROM_SETTINGS)
      MENU_ITEM(function, MSG_STORE_EPROM, Config_StoreSettings);
    #endif
    END_MENU();
  }
```

Jinxycob, 2016-11-27:
sry bud just deleted that comment  it was confusing the issue

thinkyhead, 2016-11-28:
The limit is based on the first hotend's limit. I'll make it so that the limit is based on all hotends, but that you still can't preheat the nozzles above their set maxima.

Added to #5313

Kaibob2, 2016-11-29:
#5315 fixed it. Thanks Scott!

closed 2016-11-29 by Kaibob2