MarlinFirmware/Marlin #5316: #define DELTA_CALIBRATION_MENU

Barrocas, 2016-11-28:
Hi, 

I have a custom built delta bot, using spectra fishing line, a ball-socket effector (like the CherryPi 3 deltabot), Arduino Mega and Ramps 1.4 with RRD Full graphics lcd.

Yesterday I downloaded and updated to latest Marlin RCbugfix.

All moves are working properly as expected.

At line 453 in Configuration.h, I tried uncommenting the:
#define DELTA_CALIBRATION_MENU

A new menu option is then available, to calibrate X, Y, Z, or Center.
I expected it to do moves to put the nozzle tip near the bed, at each tower base.

Tried the option to calibrate X, the effector started moving down, then went towards Z tower until crashing and disassembling all rods :p

Blue-Marlin, 2016-11-28:
Did you 'home' before?
Is your DELTA_PRINTABLE_RADIUS set to reasonable value?

```
  #if ENABLED(DELTA_CALIBRATION_MENU)

    void _goto_tower_pos(const float &a) {
      do_blocking_move_to(
        a < 0 ? X_HOME_POS : sin(a) * -(DELTA_PRINTABLE_RADIUS),
        a < 0 ? Y_HOME_POS : cos(a) *  (DELTA_PRINTABLE_RADIUS),
        4
      );
    }

    void _goto_tower_x() { _goto_tower_pos(RADIANS(120)); }
    void _goto_tower_y() { _goto_tower_pos(RADIANS(240)); }
    void _goto_tower_z() { _goto_tower_pos(0); }
    void _goto_center()  { _goto_tower_pos(-1); }

    void lcd_delta_calibrate_menu() {
      START_MENU();
      MENU_BACK(MSG_MAIN);
      MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
      MENU_ITEM(function, MSG_DELTA_CALIBRATE_X, _goto_tower_x);
      MENU_ITEM(function, MSG_DELTA_CALIBRATE_Y, _goto_tower_y);
      MENU_ITEM(function, MSG_DELTA_CALIBRATE_Z, _goto_tower_z);
      MENU_ITEM(function, MSG_DELTA_CALIBRATE_CENTER, _goto_center);
      END_MENU();
    }

  #endif // DELTA_CALIBRATION_MENU
```

Barrocas, 2016-11-28:
Hi,

Yes, first I made a 'home'.

Delta printable radius is set to 150.0 , I have a 330mm diameter round bed,
and the hotend nozzle can reach the bed limits.

I can configure and edit the basics, but I don't understand programming,
i'm just a user.

Today I removed the delta rods for safety and re-tested.

Home, then calibrate X

All carriages go down some 15cm, they stop, and continue lowering some
micro-steps, until it goes crazy, Z carriage goes all way up and rams past
the top endstop, while the others go down.

On a second atempt, Marlin rebooted itself...

Third atempt, made the same as the first.

Also tried making calibrate X without home, it makes a move simulating like
if it had reached the endstops (but it only was midway the towers height),
then moved the nozzle to X calibration position (near tower X), but midway
the tower height.

-- 
Cumprimentos,
D.er Hugo Barrocas

Blue-Marlin, 2016-11-28:
There are some other mathematical limits for `DELTA_PRINTABLE_RADIUS`.
The 'rods' or better, their mathematical representations, shall never be perpendicular or horizontal. Else we get divisions by zero, flipping signs or infinite results, or something like that.

maukcc, 2016-11-29:
This code also expects a feedrate.
So change to:

    void _goto_tower_pos(const float &a) {
      do_blocking_move_to(
        a < 0 ? X_HOME_POS : sin(a) * -(DELTA_PRINTABLE_RADIUS),
        a < 0 ? Y_HOME_POS : cos(a) *  (DELTA_PRINTABLE_RADIUS),
        4,
        (HOMING_FEEDRATE_Z)

But this do_blocking_move only moves 2 towers at a time resulting in a large dip of your nozzle into your bed :)
      );
    }

maukcc, 2016-11-29:
Also this funtion is only usefull if:
- you have a static bed 
- your dynamic bed is supported at 3 points at the radians position of the towers

If you have your dynamic bed supported at 4 points (which most beds are) you want to measure near those 4 support screws.

Barrocas, 2016-11-29:
My rods can get vertical, but that is when the nozzle is already outside
the bed area, so it's a position not usable for printing or calibration.

I usually do my manual calibration carefully, using the Pronterface host,
and will continue doing it.

I tried the option available in the config.h just to see what it would do,
but as it is now, warning, it can be a self-detruction mode :p

The only way I see this menu entry being usefull, would be to place the
nozzle tip at about 5 or 10mm above the bed (at near X,Y,Z towers and
Center), and then the rotary encoder would be used to raise or lower the
nozzle until 0 could be reached, so that the user could verify if the bed
adjustments (or endstop adjustments) were correct or needed to be altered.

If it just moves the nozzle to Z=0, it can represent either being above the
bed, spot on, or crashing.

And any Calibrate X, Y, Z or Center could start with an automatic 'home'
G28 move first, then move to the desired calibration spot.

This is just my point of view, and I do appreciate all the great work done
so far ;)

-- 
Cumprimentos,
D.er Hugo Barrocas

Barrocas, 2016-11-29:
I do have an adjustable bed that is supported at 3 points at the radians
position of the towers.

-- 
Cumprimentos,
D.er Hugo Barrocas

thinkyhead, 2016-11-30:
> This code also expects a feedrate.

@maukcc I note that our current `do_blocking_move_to` function (for `DELTA`) will set the feedrate to `XY_PROBE_SPEED` when no other rate is supplied. But I gather that for `DELTA` kinematics, feedrates are applied at the stepper level as opposed to the "cartesian" level. Therefore, on a deltabot this value should be set to a similar limit as that used for `Z_PROBE_SPEED_FAST`. Maybe in the future we need to refine that, and deal better with feedrates in kinematic situations.

Looking further into `do_blocking_move_to` I see that if Z needs to raise (in this case, to 4mm) then it will do that first. Then it will move to XY, and finally, if Z was moving to a lower point, then it will do that. And, all these moves are being done without interpolation.

Given how much a deltabot can dip in an uninterpolated move, perhaps this menu item needs to be refined further, so it does interpolated moves instead, or makes sure to move to a greater height before moving to the towers.

----

@Barrocas As to the specific behaviors you're seeing, you can provide more detail for us by enabling the `DEBUG_LEVELING_FEATURE` option in `Configuration.h`. Re-flash the firmware, and then before you experiment with these menu items, enable all extra logging with `M111 S247`. (Of course, disconnect the effector for safety too.) Try each of the tower moves that you can, and they will be logged in detail to the host console. Paste the results here, and we should be able to see exactly what the routine is up to.

Barrocas, 2016-12-01:
After enabling the DEBUG_LEVELING_FEATURE, and re-flashing,
the comands were issued in the LCD menu only:

Home all axes, Calibrate X

but this time all axes stayed still, it didn't ram the top Z endstop
as it did the other day.

>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:enqueueing "G28"
echo:G28
>>> gcode_G28
Machine Type: Delta
Probe: NONE
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
  current_position=(0.00, 0.00, 0.00) : >>> home_delta
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, 450.00, 0.00)
  current_position=(0.00, 383.00, 383.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, -5.00, 0.00)
  current_position=(0.00, 383.00, 383.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, 10.00, 6.67)
  current_position=(0.00, 383.00, 383.00) : sync_plan_position
<<< do_homing_move(X)
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, 450.00, 0.00)
  current_position=(10.00, 0.00, 383.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, -5.00, 0.00)
  current_position=(10.00, 0.00, 383.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, 10.00, 6.67)
  current_position=(10.00, 0.00, 383.00) : sync_plan_position
<<< do_homing_move(Y)
<<< homeaxis(Y)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, 559.50, 0.00)
  current_position=(10.00, 10.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, -5.00, 0.00)
  current_position=(10.00, 10.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, 10.00, 6.67)
  current_position=(10.00, 10.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
<<< homeaxis(Z)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
[... 300 lines cut ...]
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
echo: too long extrusion prevented
  current_position=(-129.90, -75.00, 289.66) : xy move
  destination=(-129.90, -75.00, 4.00) :
prepare_uninterpolated_move_to_destination
start
echo:Marlin 1.1.0-RCBugFix
...

-- 
Cumprimentos,
D.er Hugo Barrocas

thinkyhead, 2016-12-01:
```
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:enqueueing "G28"
echo:G28
>>> gcode_G28
Machine Type: Delta
Probe: NONE
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
  current_position=(0.00, 0.00, 0.00) : >>> home_delta
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, 450.00, 0.00)
  current_position=(0.00, 383.00, 383.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, -5.00, 0.00)
  current_position=(0.00, 383.00, 383.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, 10.00, 6.67)
  current_position=(0.00, 383.00, 383.00) : sync_plan_position
<<< do_homing_move(X)
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, 450.00, 0.00)
  current_position=(10.00, 0.00, 383.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, -5.00, 0.00)
  current_position=(10.00, 0.00, 383.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, 10.00, 6.67)
  current_position=(10.00, 0.00, 383.00) : sync_plan_position
<<< do_homing_move(Y)
<<< homeaxis(Y)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, 559.50, 0.00)
  current_position=(10.00, 10.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, -5.00, 0.00)
  current_position=(10.00, 10.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, 10.00, 6.67)
  current_position=(10.00, 10.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
<<< homeaxis(Z)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -150.00
 soft_endstop_max = 150.00
> home_offset[X] = 0.00
  current_position=(0.00, 10.00, 10.00) :
<<< set_axis_is_at_home(X)
>>> set_axis_is_at_home(Y)
[... 290 lines cut ...]
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
>>> do_blocking_move_to(-129.90, -75.00, 4.00)
  destination=(0.00, 0.00, 289.66) : set_destination_to_current
echo: too long extrusion prevented
  current_position=(-129.90, -75.00, 289.66) : xy move
  destination=(-129.90, -75.00, 4.00) :
prepare_uninterpolated_move_to_destination
start
echo:Marlin 1.1.0-RCBugFix
...
```

thinkyhead, 2016-12-01:
The crash is interesting. What version of Arduino IDE did you use to build?

Barrocas, 2016-12-02:
Updated recently, it's now version 1.6.13

-- 
Cumprimentos,
D.er Hugo Barrocas

thinkyhead, 2016-12-03:
Well, I will be back in the company of my Micromake D1 in less than a week. I will debug the menu then. In the meantime, I suggest just don't bother using it. Just stick to `G1` for now.

Barrocas, 2016-12-03:
Will do ;)

-- 
Cumprimentos,
D.er Hugo Barrocas

esenapaj, 2016-12-03:
Perhaps It's related to this?
Delta Homing Stuck · Issue #4509 · MarlinFirmware/Marlin

Because It seems that @Barrocas 's Delta is large.
It has 330mm bed (bed of standard Kossel is about 200mm) and when rods can get vertical then nozzle is already outside the bed area,
It suggests that the printer itself is quite large.

thinkyhead, 2016-12-03:
Well, from what I can see the calibration menu is supposed to use `DELTA_PRINTABLE_RADIUS` and move to the points near the towers at that radius. If the `DELTA_PRINTABLE_RADIUS` is set to a reasonable value, I would expect the menu items to work properly. There shouldn't be any problem with numerical overflow in the case of r^2 since floats are being used.

closed 2017-10-31 by thinkyhead