MarlinFirmware/Marlin #5317: XYZ movement speeds on delta

maukcc, 2016-11-28:
When I tell my delta printer to move the Z-axis 100mm at 10mm/sec it takes 17 seconds
When I tell my delta printer to move the X or Y-axis 100mm at 10mm/sec it takes 6 seconds
(It does actually move the 100mm)
What is going on here ?

landodragon141, 2016-11-28:
What is your max Z-axis feedrate set to?

maukcc, 2016-11-28:
max X, Y and Z-axis feedrate is 200mm/sec.

landodragon141, 2016-11-28:
What are you using to control the printer? Have you tried sending raw gcode?

maukcc, 2016-11-28:
Just standard gcode through pronterface terminal
G1 X100 F600
G1 Y100 F600
G1 Z100 F600

Blue-Marlin, 2016-11-28:
In z move on a DELTA all 3 axes are involved and all of them have the same way.
It could be something like:
Somewhere in Marling the distance between start and end of the move is calculated based on a Cartesian coordinate system - resulting in a factor of sqrt(x²+y²+z²) = sqrt(3) ~ 1.73

Sebastianv650, 2016-11-28:
If you double the speed, especialy for the z only move, does it need 8-9s or is it still around 17?

maukcc, 2016-11-29:
Moving Z 100mm at F1200 takes 9sec
Moving X and Y 100mm at F1200 takes 3sec
So it goes twice as fast, but just not the correct speed

landodragon141, 2016-11-29:
@thinkyhead Scott where is the code for delta's located? I tried to find it yesterday so I could look in this but I ended up throwing in the towel.
@maukcc Is this issue isolated to manual movements or does this effect prints too?

maukcc, 2016-11-29:
It happens in printing as well. That is how I found out. As I always print at 50mm/sec. When I printed with the delta I thought: this is faster then 50mm/sec and started testing.

Blue-Marlin, 2016-11-29:
You can find the 'main' delta code in https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L8597-L8795. To find all specific code grep for DELTA.

What i was talking about is https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/planner.cpp#L944-L959. There is no specific DELTA code

landodragon141, 2016-11-30:
I'm going to go out on a limb and guess that you didn't put in correct values for the delta parameters. There are other using marlin with a delta just fine.

thinkyhead, 2016-11-30:
As I mentioned in another thread, feed-rates for kinematic robots apply to the carriages (or arm angles) and not to the effector / tool coordinates. Thus you will find that movements on a DELTA (or SCARA) accelerate smoothly along the towers (or in the arm angles), but change speed in the cartesian space. This is due to the fact that all acceleration is handled at the planner level after the kinematics have been applied. That's not just a design limitation, but a practical one, because the steppers are the articulation point where torque is most strongly manifested.

closed 2017-10-31 by thinkyhead