MarlinFirmware/Marlin #5319: G30

cabbagecreek, 2016-11-28:
Im you do multiple G30 the value of Z_CLEARANCE_BETWEEN_PROBES is added to the reported value

SENT: G28 Z
READ: X:145.00 Y:85.00 Z:3.01 E:0.00 Count X: 11600 Y:15300 Z:32000
READ: ok
SENT: G30
READ: echo:busy: processing
busy: processing
READ: Bed X: 145.00 Y: 120.00 Z: 3.70 **<---- First G30**
READ: X:145.00 Y:85.00 Z:3.70 E:0.00 Count X: 11600 Y:15300 Z:11838
READ: ok
SENT: G30
READ: Bed X: 145.00 Y: 120.00 Z: 4.70 **<---- Secund G30 has added Z_CLEARANCE_BETWEEN_PROBES**
READ: X:145.00 Y:85.00 Z:4.70 E:0.00 Count X: 11600 Y:15300 Z:15036
READ: ok

Trying to use G30 S0
SENT: G28 Z
READ: X:145.00 Y:85.00 Z:3.01 E:0.00 Count X: 11600 Y:15300 Z:32000
READ: ok
SENT: G30 S0
READ: echo:busy: processing
busy: processing
READ: Bed X: 145.00 Y: 120.00 Z: 3.70 **<------ First G30**
READ: X:145.00 Y:85.00 Z:3.70 E:0.00 Count X: 11600 Y:15300 Z:11838
READ: ok
SENT: G30 S0
READ: Bed X: 145.00 Y: 120.00 Z: 4.70 **<---- Secund G30 has added Z_CLEARANCE_BETWEEN_PROBES**
READ: X:145.00 Y:85.00 Z:4.70 E:0.00 Count X: 11600 Y:15300 Z:15036
READ: ok

Trying to use G30 S1
SENT: G28 Z
READ: X:145.00 Y:85.00 Z:3.01 E:0.00 Count X: 11600 Y:15300 Z:32000
READ: ok
SENT: G30 S1
READ: echo:busy: processing
busy: processing
READ: Bed X: 145.00 Y: 120.00 Z: 3.70 **<------ First G30**
READ: X:145.00 Y:85.00 Z:3.70 E:0.00 Count X: 11600 Y:15300 Z:11838
READ: ok
SENT: G30 S1
READ: Bed X: 145.00 Y: 120.00 Z: 4.70 **<---- Secund G30 has added Z_CLEARANCE_BETWEEN_PROBES**
READ: X:145.00 Y:85.00 Z:4.70 E:0.00 Count X: 11600 Y:15300 Z:15036
READ: ok

Conclusion The S0/S1 do the same result.. 

The module that do the probing is   probe_pt(X_probe_location, Y_probe_location, stow, 1); 
 (row 4275 Marlin_main.cpp)

Testing the probe M48: Measure Z-Probe repeatability
SENT: G28
READ: X:145.00 Y:85.00 Z:3.01 E:0.00 Count X: 11600 Y:15300 Z:32000

SENT: M48
READ: M48 Z-Probe Repeatability Test
READ: Mean: 2.018239 Min: 2.011 Max: 2.024 Range: 0.013
READ: Standard Deviation: 0.004585
READ: 
READ: X:145.00 Y:85.00 Z:2.70 E:0.00 Count X: 11600 Y:15300 Z:30993
READ: ok

Z_CLEARANCE_BETWEEN_PROBES is 1mm
M851 is -1.70mm 
The M48 reported 1mm + 1.70 = 2.70mm

nzinov, 2016-11-29:
@cabbagecreek, have you tried changing Z_CLEARANCE?  What commit did you build?

cabbagecreek, 2016-11-29:
I will Increase the Z_CLERANCE to 5mm and do the same test.

Still I cant see the differance of the S parameter (no S/S0S1 creates the same responce)

cabbagecreek, 2016-11-29:
Changed the Z_CLEARANCE to 10mm
Increasing the Z_CLEARANCE removed the problem with accumulated Z_CLEARANCE

I still think there is something wrong with The S parameter. In dokumentation it should give Z with or without offset.  

This is the same test after reprogramming
I tested all of the combinations I could find of G30 (no S/S0/S1/S-1 with and without space)

SENT: G28 Z

SENT: G30 S1
READ: Bed X: 215.00 Y: 120.00 Z: -0.25
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G30 S1
READ: Bed X: 215.00 Y: 120.00 Z: -0.32
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G30 S1
READ: Bed X: 215.00 Y: 120.00 Z: -0.46
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G30 S1
READ: Bed X: 215.00 Y: 120.00 Z: -0.43
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G30 S1
READ: Bed X: 215.00 Y: 120.00 Z: -0.43
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G30 S0
READ: Bed X: 215.00 Y: 120.00 Z: -0.44
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G30 S0
READ: Bed X: 215.00 Y: 120.00 Z: -0.37
READ: X:215.00 Y:85.00 Z:11.70 E:0.00 Count X: 17200 Y:15300 Z:37440

SENT: G28 Z
READ: X:145.00 Y:85.00 Z:3.01 E:0.00 Count X: 11600 Y:15300 Z:32000

SENT: G30
READ: Bed X: 145.00 Y: 120.00 Z: -0.06
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30
READ: Bed X: 145.00 Y: 120.00 Z: -0.03
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30
READ: Bed X: 145.00 Y: 120.00 Z: -0.06
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30
READ: Bed X: 145.00 Y: 120.00 Z: -0.06
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30
READ: Bed X: 145.00 Y: 120.00 Z: -0.09
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30 S1
READ: Bed X: 145.00 Y: 120.00 Z: -0.07
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30 S-1
READ: Bed X: 145.00 Y: 120.00 Z: -0.03
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30S1
READ: Bed X: 145.00 Y: 120.00 Z: -0.06
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30S0
READ: Bed X: 145.00 Y: 120.00 Z: 0.02
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

SENT: G30S0
READ: Bed X: 145.00 Y: 120.00 Z: -0.07
READ: X:145.00 Y:85.00 Z:11.70 E:0.00 Count X: 11600 Y:15300 Z:37440

nzinov, 2016-11-29:
You still haven't specified what version of Marlin and what type of probe you use. At least in current `RCBugFix` branch `S` parameter controls stowing the probe after probing.  I.e. `G30 S0` command does nothing after probing while `G30 S1` moves probe to stowed position after probing. This option is not supposed to affect output in **any** way.

cabbagecreek, 2016-11-29:
I have RC7Bugfix and E3D bigbox with IR sensor.  nzinov has described how the S parameter works. The test shows that it works this way.

My suggestion is to close this issue

closed 2016-11-29 by cabbagecreek

thinkyhead, 2016-11-30:
Apparently we need to update http://reprap.org/wiki/GCode#G30:_Single_Z-Probe to reflect how it works in Marlin. Sorry for the confusion!

cabbagecreek, 2016-11-30:
WHen I have totaly figured out how the G30 is working I will try to update the Marlin Documentation. I will try to do the documentation from the comments inside the code