MarlinFirmware/Marlin #5320: problem with BILINEAR Auto Bed Leveling/Z_CLEARANCE_BETWEEN_PROBES

solo220, 2016-11-28:
Dear Sirs
I am using Marlin-RCBugFix downloaded on 9/11/16 in corexy configuration with ardumega 2560,   Z_CLEARANCE_BETWEEN_PROBES and AUTO_BED_LEVELING_BILINEAR is enabled while using a proximity 
sensor for aluminium printing Bed detection.
Everything worked perfectly except the safe distance/clearance it is activated only if G29 is issued after some print job .
If at startup a G28 is issued then followed by G29 the movement between probing points is done with the nozzle very close to bed(at z=Z_PROBE_OFFSET_FROM_EXTRUDER), after any print is done G29 just work  as intended with Z_CLEARANCE_BETWEEN_PROBES  as in setting.
Any work around or something i should enable ?
Thank you.

solo220, 2016-11-29:
Hi.
After experimenting with the issue and using the special gcodes found inside the printing files the issue can be reproduced by the following steps:
1- G28
2-G29 (there is no Z clearance between probing points)
3- G28 X0  or  G28 Y0
4-G29 (clearance between probing points occur)

after step 4 he Z_CLEARANCE_BETWEEN_PROBES is always active whenever G29 is used
but whenever G28 or G28 Z is sent to the printer the issue occurs and you had to repeat steps 2 and 3.

solo220, 2016-11-30:
Issue Solved.
If the physical distance between Z_MAX_END__STOP and the nozzle is less than Z_MAX_POS  the Z_CLEARANCE_BETWEEN_PROBES is omitted.
Example: 
 if  z homing is at max z and  Z_MAX_POS =110  and if physical distance between limit switch and nozzle is 100mm at z home position Z_CLEARANCE_BETWEEN_PROBES will be omitted and z probing is sometimes fuzzy.
It looks that some equations will result in unhandled values.

thinkyhead, 2016-11-30:
Please post your `Configuration.h`. (You may need to rename it to `Configuration.txt` to be able to drop it onto the edit box here.)

solo220, 2016-11-30:
This is my Configuration file
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/621614/Configuration.h.txt)

thinkyhead, 2016-12-01:
What does the output of `M503 S0` say?

thinkyhead, 2016-12-01:
If you home with `G28` and then do `G0 Z0` do you find that the nozzle sits right on the bed?

Also, to help debug, please enable `DEBUG_LEVELING_FEATURE`, re-flash, then before `G28` and `G29` do `M111 S247` to enable maximum logging. Then post the log output here for examination.

solo220, 2016-12-02:
Log for  >>> M503 S0
SENDING:M503 S0
  M92 X200.00 Y200.00 Z800.00 E1002.60
  M203 X200.00 Y200.00 Z5.00 E20.00
  M201 X150 Y150 Z5 E10
  M204 P50.00 R20.00 T200.00
  M205 S0.00 T0.00 B20000 X5.00 Y5.00 Z5.00 E2.00
  M206 X0.00 Y0.00 Z0.00
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 P22.32 I1.66 D75.15
  M200 D3.00
  M200 D0
  M851 Z-2.50

Log for G28 G29
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
echo:G28
>>> gcode_G28
Machine Type: Core
Probe: FIX_MOUNTED_PROBE
Probe Offset X:-50 Y:0 Z:-2.50 (Left-Center & Below Nozzle)
Auto Bed Leveling: BILINEAR
reset_bed_level
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, 450.00, 0.00)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, -2.00, 0.00)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, 4.00, 2.00)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 300.00
> home_offset[Z] = 0.00
  current_position=(0.00, 0.00, 300.00) :
<<< set_axis_is_at_home(Z)
  current_position=(0.00, 0.00, 300.00) : sync_plan_position
  current_position=(0.00, 0.00, 300.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Z)
  current_position=(0.00, 0.00, 300.00) : > HOMEAXIS(Z)
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -450.00, 0.00)
  current_position=(0.00, 0.00, 300.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, 0.00)
  current_position=(0.00, 0.00, 300.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 3.50)
  current_position=(0.00, 0.00, 300.00) : sync_plan_position
<<< do_homing_move(X)
[... 302 lines cut ...]
Extrapolate [ 1-  0+] (done)
Extrapolate [ 1+  2-] (done)
Extrapolate [ 1-  2-] (done)
Extrapolate [ 0+  1+] (done)
Extrapolate [ 2-  1+] (done)
Extrapolate [ 0+  1-] (done)
Extrapolate [ 2-  1-] (done)
Extrapolate [ 0+  0+] (done)
Extrapolate [ 2-  0+] (done)
Extrapolate [ 0+  2-] (done)
Extrapolate [ 2-  2-] (done)
Bilinear Leveling Grid:
      0     1     2
 0 +191.12 +191.06 +191.05
 1 +190.38 +190.65 +190.71
 2 +190.15 +190.27 +190.27
G29 uncorrected Z:192.77
 corrected Z:2.50
<<< gcode_G29
  current_position=(170.00, 120.00, 2.50) : sync_plan_position
echo:M105
echo:M105
echo:M105
echo:M105
Disconnected.

thinkyhead, 2016-12-03:
If you home with G28 and then do G0 Z0 do you find that the nozzle sits right on the bed?

thinkyhead, 2016-12-03:
How can we account for the fact that your Z probe is triggering with the Z axis at 190mm above the bed? It should trigger when Z at around 2.5.

thinkyhead, 2016-12-03:
Something is getting confused. Your configuration indicates these limits:
```cpp
#define X_MAX_POS 300
#define Y_MAX_POS 200
#define Z_MAX_POS 110
```
But when you home with `G28` the Z position is set to 300. Subsequently, the `G29` corrects the position. But until you do the `G29` the Z position is incorrect — is is 190mm too large.

thinkyhead, 2016-12-03:
You may want to try compiling Marlin with an earlier version of Arduino IDE to see if it exhibits the same problem. It appears that the Z home position is being written into memory incorrectly, which would indicate a compiler issue.

solo220, 2016-12-04:
Dear thinkyhead 
Thank you for your help , the homing issue of difference between the log file and the configuration file is my own fault (i am sorry) because before i send the file i made changed it to solve the main issue with probe clearance by making Z_MAX_POS larger than the distance between the limit switch and the nozzle tip.

In my machine i can slide the limit switch up and down  to make the z distance smaller while printing tiny models so i am not obliged to wait for the 350mm z homing time with slow z stepper speed and this is where my problem started.

I understand that in marlin this is not acceptable (please correct me if i am wrong), i mean that the distance between limit switch and nozzle tip should be larger or equal to Z_MAX_POS, because i assume that absolute values are used to move the z axis in the case of probing clearance .
I made this assumption because if let say Z_MAX_POS=150  ,  Z_CLEARANCE_BETWEEN_PROBES=5
and distance between bed and nozzle at z home is 155(150+5) , the the nozlle wil rise for 10mm  (5+5) above the build plate.

thinkyhead, 2016-12-08:
> the distance between limit switch and nozzle tip should be larger or equal to Z_MAX_POS

`Z_MAX_POS` can be >= `Z_HOME_POS`, but it should not be smaller.

> absolute values are used to move the z axis in the case of probing clearance

There's no relationship between `Z_CLEARANCE_BETWEEN_PROBES` and `Z_MAX_POS` or `Z_HOME_POS`.

In-between probes, the nozzle will first raise up by `Z_PROBE_OFFSET_FROM_EXTRUDER` (if it's negative), and then by `Z_CLEARANCE_BETWEEN_PROBES`.

closed 2017-10-31 by thinkyhead