MarlinFirmware/Marlin #5322: no Fan on kossel xl after uploading RCbugfix 7

mihadongle, 2016-11-29:
Hallo on my kossel after update the new file and i staurt printing the hotend goes up and up  and no fan starts!! When i downgrade to the original Marlin of my kossel the Fan works fin

thinkyhead, 2016-11-30:
What is your setting for `E0_AUTO_FAN_PIN`?

thinkyhead, 2016-11-30:
> when i write in pronterface m119,m117,m28,m29

Weird. `M119` should print out your endstop status. `M117 Hello` should say "Hello" on your LCD controller. `M28` should home. `M29` should do bed leveling, if configured.

mihadongle, 2016-11-30:
i didn´t find the E0_AUTO_FAN_PIN in which file is that? 
yesterday i have testet again 
M119 is OK
M28 he do nothing
M29 he do nothing
M114 he do nothing
When i write M106 S255 the FAN Starts to run!!
When i turn manuel FAN on, on the LCD  then he start and the PWM is also OK
When i start to print or in Pronterface Set the Temp to 185C° the FAN DIDn´T start!!!

mihadongle, 2016-11-30:
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/621446/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/621447/Configuration.txt)

lrpirlet, 2016-11-30:
@thinkyhead I guess you mean G28 and G29 instead of M28 and M29...
@mihadongle Mind to tell us what realy you want to do (not that I have the answer, note, but obviously something is unclear...

ref from Marlin_main.cpp
``` C++
/**
 * -----------------
 * G-Codes in Marlin
 * -----------------
 *
 * Helpful G-code references:
 *  - http://linuxcnc.org/handbook/gcode/g-code.html
 *  - http://objects.reprap.org/wiki/Mendel_User_Manual:_RepRapGCodes
 *
 * Help to document Marlin's G-codes online:
 *  - http://reprap.org/wiki/G-code
 *  - https://github.com/MarlinFirmware/MarlinDocumentation
 *
 * -----------------
 *
 * "G" Codes
 *
 ... some lines removed ...
 * G28 - Home one or more axes
 * G29 - Detailed Z probe, probes the bed at 3 or more points.  Will fail if you haven't homed yet.
 ... some lines removed ...
 * "M" Codes
 *
 * M28  - Start SD write: "M28 /path/file.gco". (Requires SDSUPPORT)
 * M29  - Stop SD write. (Requires SDSUPPORT)
... some lines removed ...
 * M114 - Report current position.
... some lines removed ... * M119 - Report endstops status.
 * M120 - Enable endstops detection.
... some lines removed ... *
 */
```

thinkyhead, 2016-12-01:
> i didn't find the E0_AUTO_FAN_PIN in which file is that?

It replaces `EXTRUDER_0_AUTO_FAN_PIN`, which I see you are not using anyway.

Are you using `1.1.0-RC7` (`RC` branch) or are you using the `RCBugFix` branch?

In the main branch of Marlin, the fan does not come on automatically when you start heating. You have to use `M106` for that. Perhaps the Marlin you used before was from a different fork?

thinkyhead, 2016-12-01:
Depending on your host software, you might have to type `G28` in uppercase. A lowercase `g28` will be ignored.

Blue-Marlin, 2016-12-01:
Auto fan and part cooling fans are different.
The auto fan is ought to cool the hotend and should be switched on at a defined temperature (if configured).
The part cooling fan is that what you can switch on and of by the LCD and M106/M107.

mihadongle, 2016-12-06:
Thanks thinkyhead that was my problem

closed 2016-12-06 by mihadongle