MarlinFirmware/Marlin #5325: Advice needed to support Mini Metal Maker

BillStruve, 2016-11-29:
I'm very new to Marlin but not to software control of hardware (early '70's) nor to programming (professional since the 80's, lost count at a dozen languages and 0.5 M lines of code).

I've got a couple of Mini Metal Makers (50-100 exist) that extrude a metal loaded paste via a stepper motor driven piston in a cartridge of paste. The paste is dried, then fired in a kiln to produce a solid bronze or sterling silver print. 3D printing metal for under $2K (printer + kiln) is cool, n'est pas?

Anyhow, I wanted to upgrade the firmware supplied by the manufacturer, a highly modified Marlin 1.02, to RC7. After seeing the changes to 11 of the Marlin files, I decided to start (almost) over. So far, only 4 files need modification in addition to Configuration.h, and adding my own extrude_multiplier.h with a couple of variables and a class in it. Looking forward to RC8 with the RCBugFix (and hopefully, UBL) in it! The fewer mods to future versions of Marlin I need to make, the better.

I need advice on two things:
   How do I get the Configuration.h and my own extrude_multiplier.h into the release examples?
   What editor or IDE is good? Editing in Adruino IDE is a big step back for me, even UltraEdit is better.
   (I've gotten spoiled with the C# IDE with its real time bug check and information being just a click away.)

Any advice would be most welcome! Anything I write will be freely shared.

emartinez167, 2016-11-29:
Johnny Linden has developed a Windows-based tool to help migrating
Configuration.h. Look him up in Google+ in the RigidBot or 3D printing
groups.

BillStruve, 2016-11-29:
@emartinez167 Looks like a cool tool for migrating the Configuration.h file to a new version of Marlin. I was looking to have my Configruation.h incorporated into the examples section of a new release of Marlin.

emartinez167, 2016-11-29:
You will need to use git to submit a pull request with the changes.

Regards,

Ernesto

thinkyhead, 2016-11-30:
Hi @BillStruve — Personally, I use [Sublime Text 3](https://www.sublimetext.com/3) and I love it. I only use the Arduino IDE for building. You can check the "Use External Editor" option in Arduino IDE and it prevents being able to edit in the IDE. Other people like TextMate, Eclipse, and Atom. Any good external text editor with plugin capability should do.

shacal, 2016-11-30:
PlatformIO support?
I currently build with that and output is more optimized and less bugs (latest AVR compiler)
Visual Studio Code or PlatformIO IDE works well together.

BillStruve, 2016-12-01:
Loving Sublime Text 3 so far, Thanks @thinkyhead

petrzjunior, 2016-12-01:
For me, I started using Visual Studio Code or Atom (they are based on the same core). both work well, have some basic git integration and accept all line nedings ;)

BillStruve, 2016-12-01:
@petrzjunior great to know since I'm familiar with Visual Studio - I'll give it a whirl when I get a round tuit.

closed 2016-12-01 by BillStruve

thinkyhead, 2016-12-04:
I'm suddenly very fond of Atom. I installed the PlatformIO plugin that turns it into an IDE. Building Marlin within Atom using PlatformIO takes only 13-17 seconds (!), and I don't have to leave the editor. It's pretty sweet.

Then I installed the git-history plugin and that also blew my mind. I'm having way too much fun playing with this editor and trying different plugins and themes.

Of course it can open my BASH in a pane (with a full-color TERM). Running JOE and EMACS inside of a pane inside of Atom made me ROTFL.

I'm going to try using it exclusively for a while and see what else I can make it do.

petrzjunior, 2016-12-04:
@thinkyhead I'm happy with Atom as well, just instaled all plugins recommended by you and especially PlatformIO just blew my mind! Coding speed doubled :)

jbrazio, 2016-12-04:
I'm a license owner of Sublime but in the last month been having spinning Atom and I would say 100% replaces Sublime with no cost, there are some plugins you must install like minimap, highlight-line and highlight-selected.

thinkyhead, 2016-12-05:
I'm kind of annoyed with Atom because multi-line find-and-replace is broken and has been for a long time with no resolution and no other option. I need to use that feature pretty frequently. And the editor is a little unstable (perhaps owing to the plugins). It's also rather sluggish. So I've gone back to Sublime for the time-being.

shacal, 2016-12-05:
Visual Studio Code + PlatformIO command line build (task from VS Code) is best for cross platform develop. :)
Atom with plugin hell, is well... hell :D

thinkyhead, 2016-12-05:
> Visual Studio Code + PlatformIO command line build 

I'll try that out next. Otherwise I'll end up having to learn CoffeeScript to fix Atom.