MarlinFirmware/Marlin #5326: Marlin For A Self Constructed 3D Printer?

LuisSchulte, 2016-11-29:
Hey guys

Okay, I want to build a 3D printer which is self constructed. Why? The main reason is that I want to have a printer constructed by myself. And the printer should be a little bit bigger than printers you can build by yourself like a printer from RepRap. I hope you know what I mean. My question: Where can I get software for my project. I thought perhaps Marlin could solve my problem. I only want to know if it could be possible if I follow some rules.

Cheers

landodragon141, 2016-11-29:
Marlin supports the majority of available controller boards as well as popular kinematic styles (cartesian, core xy, delta, scara). It should work for pretty much any printer you try to build.

Blue-Marlin, 2016-11-29:
You can find the 'main' delta code in https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L8597-L8795. To find all specific code grep for `DELTA`.

What i was talking about is https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/planner.cpp#L944-L959. There is no specific DELTA code

LuisSchulte, 2016-11-29:
Thanks for the fast responses. I will build my 3d printer and then we'll see if it works. Additionally I know now that the printer which I want to build a Cartesian is without a movable print bed. Thanks ;)

thinkyhead, 2016-11-30:
Good luck! If you decide to use multiple X, Y, or Z motors, there are options. Keep us posted and don't hesitate to ask for help if you get stuck or need some special feature to support your concept.

closed 2016-11-30 by thinkyhead