MarlinFirmware/Marlin #5329: LCD configuration request

RogerRepRap, 2016-11-29:
I had a problem with a machine with an LCD and had to remove it from config.h in order to get the machine to work. It took some time and several compiles as there are several places to comment in order to remove it.

Would it be possible to to add a #define LCD in config.h and place all configuration.h LCD related stuff under a #if LCD. Also in configuration_adv.h put babystepping under #if LCD. This would simplify things a bit. Set the default to define LCD. It won't affect people who aren't using one and will not affect those who are using one.

Also... The kill for minimum temperature tells you where to look for problem. The LCD caused kill() message was unhelpful as it didn't identify where/why it was called. I had to search the source in order to figure out it had something to do with the LCD.

Thanks,
Roger

thinkyhead, 2016-11-30:
What version of Marlin were you using?

thinkyhead, 2016-11-30:
> Would it be possible to to add a #define LCD in config.h and place all configuration.h LCD related stuff under a #if LCD.

Not for the 1.1.0 release. Marlin 1.2 will be splitting things up a bit more.

thinkyhead, 2016-11-30:
> The LCD caused kill() message was unhelpful as it didn't identify where/why it was called.

You mean, due to the configured `KILL_PIN`?

RogerRepRap, 2016-12-01:
Was using candidate 7 with a graphics LCD. Message just appeared, did not press kill, just appeared in log. LCD had died... maybe Marlin thought it saw a kill do to hardware glitch.

I was not considering refactoring Marlin. Was just suggesting adding a #if  LCD (and define) to the config.h file which should(?) kill all the defines for LCD's.

Roger

Blue-Marlin, 2016-12-01:
You seem to misunderstand the concept of kill().
Kill is called when we see a serious problem and/or if the kill button is pressed.
Even if no LCD is connected the kill is performed - you just cant see it.
The complete error message, with the reason for the kill, is always send via the serial (as far as possible).

thinkyhead, 2016-12-01:
> Was just suggesting adding a `#if LCD` (and define) to the config.h file

@RogerRepRap I see. That would be redundant given our current scheme. You are only allowed to have a single LCD option selected, or we throw an error. When you want to compile without the LCD all you need do is comment-out the one LCD option that you have enabled. All other LCD-dependent defines are now in `Conditionals.h`, hidden from the casual user.

Blue-Marlin, 2016-12-01:
If you don't explicitly configure a LCD neither DOGM nor ULTRA_LCD is defined. All LCD-code depends ether on the one or the other.

RogerRepRap, 2016-12-01:
OK. No problem. Just in order to shut down failing LCD (which killed Marlin) I to make 3 changes to the configs. I was thinking a single switch to turn off/on LCD support would be simpler and not affect existing code. You can consider this item closed.

To shut down failing LCD I had to make 2 changes in config.h:
#define DOGLCD
 #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

And 1 in  Configuration_adv.h:
#define BABYSTEPPING 

Roger

closed 2016-12-01 by RogerRepRap

thinkyhead, 2016-12-01:
You can always leave `DOGLCD` commented-out. It will be enabled by those displays that need it. The `BABYSTEPPING` option in `Configuration_adv.h` will not enable any LCDs, so you can also leave that untouched.