MarlinFirmware/Marlin #5332: Ramps 1.4 Plus pins does not match 1.4 pins, option for RAMPS_PLUS?

shacal, 2016-11-30:


Blue-Marlin, 2016-11-30:
What is a RAMPS_PLUS? Not a skateboard ramp i suppose.

shacal, 2016-11-30:
http://reprap.org/wiki/Ramps1.4plus

shacal, 2016-11-30:
Quite popular in Scandinavian and baltic

Blue-Marlin, 2016-11-30:
The wiki page is quite useless. It's an ancient copy of the RAMPS page + a photo of the board + a comment by Grogyans. Where is the real info about features, layout, pinning, availability?

shacal, 2016-11-30:
http://www.yoycart.com/Product/521797326074/
http://www.iographix.com/products/3d-printer-main-control-board-automatic-leveling-double-shaft-reprap-ramps1-4-plus-bt7272a/

So this is considered as not RAMPS board?

thinkyhead, 2016-12-01:
It's a RAMPS-like board. All the boards in Marlin that are RAMPS-like have their own pins file, which includes `pins_RAMPS.h` then overrides any differences.

thinkyhead, 2016-12-01:
Differences include:
- Swap the E0 and E1 pins (like "Rigidboard")
- E0 Heater on D8 (instead of D10)
- Bed Heater on D10 (instead of D8)
- Kill pin 27 default? (Later overridden to 41.) Should be LCD-dependent (41, 31, or 64)
- LCD pins assume basic character-based controllers, but are custom so would need to be overridden at the connector level…
  - `LCD_PINS_RS` 41
  - `LCD_PINS_ENABLE` 31
  - `LCD_PINS_D4` 33
  - `LCD_PINS_D5` 23
  - `LCD_PINS_D6` 42
  - `LCD_PINS_D7` 44
- `SD_DETECT_PIN` 49, assumes a particular controller

boelle, 2016-12-03:
i bet @shacal has one

shacal, 2016-12-04:
I have 6, and have build around 20 with these and I love that board.
I think that best way to "fix" this is that I create pins_RAMPS_PLUS.h and have PR for that.

Cool?

shacal, 2016-12-04:
And I have both Full and slim reprap graphic LCD controllers to test.

thinkyhead, 2016-12-05:
> I think that best way to "fix" this is that I create pins_RAMPS_PLUS.h and have PR for that.

That would be great. If this board is gaining traction, we should have it in our boards list.

thinkyhead, 2016-12-05:
> And I have both Full and slim reprap graphic LCD controllers to test.

I'm trying to get `1.1.0-RC8` out tonight, if possible. But we should be able to include it into the final 1.1.0 release, as it is non-disruptive to the rest of the code.

shacal, 2016-12-05:
Final target sounds ok.
I have new printer on table that gets this same setup with latest rcfixes branch, so instead using own copy of ramps pins from private git, I create something more suitable for public usage.
:)

furiati, 2017-01-13:
Hi, Any update on this subject?

Thank you very much!

:)

shacal, 2017-01-14:
Yes and no...
I m still working on this, as I need to clean up my pin configs and test with different ramps configuration.

I try to make pull request soon.

shacal, 2017-01-14:
https://github.com/MarlinFirmware/Marlin/pull/5695

furiati, 2017-01-15:
Thanks for the update!
I have the last RCBugFix working on my printer, auto leveling working too, if you think it can help you, I can upload configuration.h and pins...h

heimerdinger91, 2017-02-07:
Hey guys I got an RRD fan extender board for autocooling (a workaround as I blew my d9 mosfet).

 I have now run into the problem of ramps 1.4 plus pins being different than 99% of ramps boards out there. Would someone here be so kind as to help me figure out the correct pins to install my RRD extender on? 

Google seems to be falling short on this one.

My best guess at this point is the AUX-2 set next to the MOSFETs but it's really just a guess and I don't want to fry my whole board.

Many thanks for any input anyone can provide.

shacal, 2017-02-07:
Check my links post2, there is full diagram in sellers details.

heimerdinger91, 2017-02-07:
Sorry for my ignorance here. I can see which pins my RAMPS 1.4plus has, but I do not know how to connect the RRD extender properly. Guides online all involve ramps boards but not the ramps 1.4plus, and some suggest AUX1 while others suggest servo pins. 

If there is a better place to be asking, please link to it.

Thanks

TilogyAspects, 2017-02-10:
Shacal, have you enabled auto leveling on this board? i can not seem to find anything on the web on it. Could you help?

shacal, 2017-02-10:
Yes, with probe in Z-minstop pin.

thinkyhead, 2017-04-02:
Let's continue this on #5820.

closed 2017-04-02 by thinkyhead