MarlinFirmware/Marlin #5333: TMC Driver support

DeathmannX, 2016-11-30:
Hi, is the support also avaible for TMC2100?
If yes how does this work?

cfelicio, 2016-11-30:
Since TMC2100 just interpolates from 1/16 to 1/256 microsteps, wouldn't it work by itself, without needing any firmware changes? Did you try? I haven't received mine yet, so can't test.

DeathmannX, 2016-11-30:
This is what i got currently: 
https://drive.google.com/open?id=0BxLYV9G8oHT2dkl6dllBX0hVVjQ

I don't did something in Marlin to define that this is a TCM 21XX.
My X, Y and Z are up to 12V and my E is up to 4V~

My Z dosen't work correctly currently.

Kaibob2, 2016-11-30:
Tmc2100 don't have SPI support, so there is no need for the firmware to support them explicitly. Just jumper your Mainboard to 16 microsteps for the axis you have installed the TMC2100 driver at and that's all. The driver will interpolate to 256 microsteps. See #4966

Kaibob2, 2016-11-30:
> My X, Y and Z are up to 12V and my E is up to 4V~

What are these voltages? Measured where?

DeathmannX, 2016-11-30:
X, Y, Z = https://de.aliexpress.com/item/JK42HS48-1684-01-1-8-degree-2-Phase-NEMA17-hybrid-stepper-motor-42mm-stepping-motor-for/2036904482.html

E = http://e3d-online.com/Electrical/Stepper-Motors?product_id=357

Kaibob2, 2016-11-30:
I see. The voltages are not very relevant. You have to adjust the current for each stepper with the potentiometer on the driver. 
Check this site http://reprap.org/wiki/TMC2100
Especially
`
Setting the reference voltage / motor current
`

Kaibob2, 2016-11-30:
Your X, Y, Z steppers are rated 1,6A and your E steppers 1,5A. Use the formula to calculate Vref, measure it and adjust it to a value as low as possible without loosing torque. The lower the current the lower the driver temperature.

Kaibob2, 2016-11-30:
The potentiometers react extremely sensitive. Be careful not to overpower the steppers. So, adjust them when the steppers are disabled. I would suggest you start with a Vref of 2.1V

Kaibob2, 2016-11-30:
Forget the previous post. Better start with 1.5V and then increase to a maximum of 2.1V.
Sorry for that.

DeathmannX, 2016-11-30:
Da ich dir hier keine Privaten Nachrichten schicken kann ganz kurz mal auf Deutsch.

Ich habe mein Multimeter angeschlossen und auf 20V Gleichstrom gestellt und da z.B. an meinem Potientiometer 0,77 gemessen.
Damit kann/muss ich jetzt hoch auf 1,5 - 2,1V liege ich da richtig?
Das sind mir gerade zu vielen Zahlen sry. :D

Kaibob2, 2016-11-30:
Aber nur ausnahmsweise in deutsch.
0,77V Vref sind ca 550mA am Stepper. Das ist reichlich wenig.

[image]
An dieser Stelle müssen gemessen gegen Masse ca. 1,5V anliegen. Dann bekommt der entsprechende Stepper 1,065A.
Mit diesem Wert würde ich starten. Dann gucken ob die Stepper genug Drehmoment haben. (Achse festhalten, sind später erfahrungswerte)
Wenn die Stepper kaum Kraft haben dann Vref bis maximal 2,0V hochdrehen. Wenn es den Treibern zu warm wird wirst du es im Betrieb an Schrittverlusten merken bzw. kompletter Ausfall einer oder mehrerer Achsen. Passieren kann aber eigentlich nichts, die schalten halt ab 150°C ab.

For the future: We should really try to always communicate in english, as this might become useful for another participant here.

Kaibob2, 2016-12-01:
Looking at your video it might also be that your Z steps/unit aren't set correctly. The Z steppers are trying to turn but then only make a humming sound. You can calculate them here http://prusaprinters.org/calculator/#stepspermmlead

Fulg, 2016-12-01:
Check that the coil pairs are correct with a multimeter. If the pairs are crossed then the motors will just vibrate instead of turning. There is no standard on the 4-pin connector or on the wire colors.

http://42bots.com/tutorials/stepper-motor-wiring-how-to/

DeathmannX, 2016-12-01:
Z Steps are correct callculated and the coil pairs are correct.

thinkyhead, 2016-12-01:
Try using lower speeds with the Z axis. If it works at lower speeds then you may only need to adjust the max speed and max acceleration for Z.

DeathmannX, 2016-12-01:
```
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 3200, 160 } //3703

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 */
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 20, 25 } //Feedratze Z before 200

/**
 * Default Max Acceleration (change/s) change = mm/s
 * Override with M201
 *
 * Maximum start speed for accelerated moves: { X, Y, Z, E }
 */
#define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 50, 1000 } //Z before 100

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          100    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   100    // X, Y, Z acceleration for travel (non printing) moves

/**
 * Default Jerk (mm/s)
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.2
#define DEFAULT_EJERK                  2.4
```

Is this slow enough?
I realy don't know how slow or fast i have to go to get this Z working. :/

justmyopinion, 2016-12-01:
You do not give any information which HW you are using so how should anyone guess what would be suitable for you  :-(
This is however a widely used setting for a standard RepRap based printer:
**#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }**

DeathmannX, 2016-12-01:
HW what do you mean?
My Printer is (if that is what you mean) a Prusa i3 with a Ramps 1.4 Board and TMC 2100 Stepper Driver.

Blue-Marlin, 2016-12-01:
The typical max feedrate for a i3 with M5 lead screws and 16 micro steps is usually about 3.5 to 4 - for me a bit more when using only 8 micro steps. Using 265 micro steps here will cause enormous step rates, a Arduino Mega can't handle.  My guess is a max fededrate of about 0.25mm/s for a M5 screw 265 microsteps and a 16MHz mega.

DeathmannX, 2016-12-01:
I got all config pins on the drivers so it is not 265 micro steps. Sry i forget that.

thinkyhead, 2016-12-01:
@Blue-Marlin I believe the 256 micro-steps is internal to the TMC2100 but from the firmware side it's just 16.

Kaibob2, 2016-12-01:
> I believe the 256 micro-steps is internal to the TMC2100 but from the firmware perspective it's just 16.

This is exactly what it's like. 

There is 3200 Z steps in his config which would be for a M6 threaded rod. Right? 

With M5 it should be 4000 and then a maximum speed of 4mm/s should be no problem.

Kaibob2, 2016-12-01:
Does this endless `echo:busy: processing` come up when you send a G28 Z?

Kaibob2, 2016-12-01:
One more thing. In my opinion this is not a Marlin related issue, maybe you better discuss this in the reprap forum at http://forums.reprap.org/

DeathmannX, 2016-12-01:
Yep 3200 steps and yes the endless Messages are still there.
I'am look for help for this Problem at 2 Forums and here.

thinkyhead, 2016-12-01:
If it says "busy:processing" it probably means it never returns from `G28`. But it also means that it's in a loop calling `idle()` somewhere.

Is this a delta or a cartesian?

Fulg, 2016-12-01:
If you have all the pins set (MS1-2-3 shorted) on an unmodified Waterott driver then you are in 1/16 mode, interpolated to 1/256 to the stepper. If you modified the board in some way (like some online sources suggest) or if you use the cheaper MKS clones instead of Waterott, then the drivers may behave differently.

If you use real Waterott boards, make sure you soldered all the pins (some guides suggest leaving some pins out) and have not soldered extra jumpers on the driver board. Then set all three MS jumpers and configure Marlin as if you had 1/16 microsteps, and you're golden.

If you are stuck in StealthChop mode (i.e. the fully quiet mode of the TMC2100) then the drivers have almost no useful torque, possibly failing to raise the Z axis on a cartesian-type machine like yours.

DeathmannX, 2016-12-01:
Prusa i3 https://drive.google.com/file/d/0BxLYV9G8oHT2dkl6dllBX0hVVjQ/view

thinkyhead, 2016-12-01:
Did you lower your max feedrate for Z to 5 or 4? After changing it, don't forget to use `M502` followed by `M500` to use the new value set in your config.

DeathmannX, 2016-12-01:
Did it and nothing happend.

thinkyhead, 2016-12-01:
From your video it looks like the Z starts to move, but then gets stuck. Is the current set properly on the Z stepper drivers?

DeathmannX, 2016-12-01:
Yes, i updated my Comment before with the configuration file. ;)

thinkyhead, 2016-12-01:
The feedrate of 5 may still be too high. Try going back to 2.

DeathmannX, 2016-12-01:
same thing with 2 :(

thinkyhead, 2016-12-01:
What does the output of `M503 S0` look like?

DeathmannX, 2016-12-01:
SENDING:M503 S0
  M92 X80.00 Y80.00 Z3200.00 E160.00
  M203 X300.00 Y300.00 Z2.00 E25.00
  M201 X2000 Y2000 Z50 E1000
  M204 P100.00 R1000.00 T100.00
  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.40
  M206 X0.00 Y0.00 Z0.00
  M301 P21.11 I1.49 D74.69
  M200 D3.00
  M200 D0
  M851 Z-0.82

Fulg, 2016-12-02:
Maybe your Z driver is defective.  If you swap the driver board from X (which we see working in your video) with Z, does the problem move to that axis? Do you have a spare TMC2100 driver to try?

I believe the constant `echo:busy: processing` is normal, since the Z is not moving the `G28` command never ends, because the Z endstop is never triggered. You should be able to regain control of the printer once you trigger the Z endstop manually.

DeathmannX, 2016-12-02:
Like i already postet a hour ago i tryed that already.
https://github.com/MarlinFirmware/Marlin/issues/5333#issuecomment-264325912

I've got a backup Ramps 1.4 Board here and i will try that later.

thinkyhead, 2016-12-02:
> Like i already postet a hour ago i tryed that already.

Let's list all the things you've tried so far:

- Used a different stepper driver board
- Adjusted the current on the driver board
- Slowed down the Z max speed
- Tested `G1 Z100` and `G1 Z0` but the axis was stuck
- Tightened the Z couplers
- Realigned the Z axis to prevent binding
- Reduced Z max acceleration to 20 or less

Anyone have other ideas?

justmyopinion, 2016-12-02:
Disconnect motor from threaded rod mechanically and see if motors are running when set free
Be sure motors are on same direction.
If motors are on same driver, set  current to drive two motors
if not working, change/swap motors.

Kaibob2, 2016-12-02:
> Drivers are real from Watterot with Pins pre assembled.

If we take a look at these tables from TMC2100 datasheet, then we want the driver board to have the following potentials on CFG1-3:

- CFG1: GND
- CFG2: OPEN
- CFG3: OPEN

[image]
[image]

Like this:
[image]

When all pins are assembled and all jumpers are installed ion the RAMPS 1.4 board the potentials will be
- CFG1: VCC
- CFG2: VCC
- CFG3: VCC

This results in 16 microsteps without interpolation to 256 microsteps. VCC on CFG3 might be problematic as the current sensing works different.

Long story short: If on the TMC driver board **ALL pins** are assembled then **NO jumper** on the Ramps 1.4 board have to be installed. Because Jumper 1 on the Ramps pulled down via R8 the potentials will then be correct. Inside Marlin you have to configure for 16 microsteps. 

[image]

Marlin doesn't know anything about the jumper settings, they only configure the hardware (stepper driver).

DeathmannX, 2016-12-02:
Ok i'am happy to say that i got it.
I changed my Ramps 1.4 Board and this dosen't change something after that i put out 1 Motor and 1 of to worked after that small test.
After that i thougt maybe the wire is damaged from the motor that dosen't worked so i changed on Board the Wires and it seems to be work.
So i decided to put the other Motor on and now both worked.

Thats it :)
Thx Guys for all you're help without you i didn't tryed it. 👍

Kaibob2, 2016-12-02:
Great, i don't understand exactly what it was, but i guess it was a wiring problem. 
Nevertheless you should consider my last post as this is the correct way your Ramps should be jumpered for use with the TMC2100.

cfelicio, 2016-12-02:
I got my TMC2100 today and will try on ramps. So, if I read correctly, since it has all the pins, I should remove all the jumpers on ramps to get 16 microsteps interpolated to 256? Thanks! :)

Kaibob2, 2016-12-02:
@cfelicio Correct!

cfelicio, 2016-12-03:
Cool, just reporting back it worked, also need to revert the stepper cable (or adjust direction in firmware), works like a charm, so quiet! Highly recommended for X and Y, I can only hear the fans on the printer now!

thinkyhead, 2016-12-03:
Thanks for suffering so the rest of us don't have to. I'll be trying these steppers in a project some time in the next month or so.

closed 2016-12-03 by thinkyhead