MarlinFirmware/Marlin #5334: Extrusion issue

RajaaKahel, 2016-11-30:
Hello, Please I'm having a strange issue. During printing everything is perfect, but if I try to lower the extrusion from 100% to 99% I get a very low extrusion like 5% or less. It becames like thin hair lines that ruin the whole thing. This happens if I lower the value from Simplify3D or the LCD panel (only during printing).
Thank you in advance. 
[image]

Erutan409, 2016-11-30:
Interesting.  I thought I was the only one having this issue.

Roxy-3D, 2016-11-30:
It might be good to save the GCode file generated by Simplify3D at 100%.  And save it at 99%.   And see what it does to the extrusion lengths.   Obviously....  The lengths should only shift by 1%.

Kaibob2, 2016-11-30:
I also use Simplify3d 3.1.0 and have a Full graphic display. Never had this issue. What Marlin version are you on?

RajaaKahel, 2016-11-30:
Roxy-3D, How should I save the Gcode when I change to 99% during printing?

Kaibob2, I'm using Marlin rc7.

psavva, 2016-12-01:
@RajaaKahel, So you are adjusting the Feed Rate to 99% whilst it's Printing is what I understood.
What version of Marlin?  Did you try RCBugFix? or are you running RC 7?

RajaaKahel, 2016-12-01:
@psavva, That's right. I'm adjusting while it's printing. If I adjust the original Gcode (while slicing before printing job) it works normally. 
It's Marlin RC7, I didn't try something else actually. I don't know what to do. Here is a screen shot for more details:
[image]

Kaibob2, 2016-12-01:
The first thing I would do is switching to RCBugfix as there were a huge amount of fixes since RC7

RajaaKahel, 2016-12-01:
@Kaibob2 I'll try and see if that fixes this issue, thanks!

thinkyhead, 2016-12-01:
I have heard of this before, but not in a long time. We may have fixed the maths in `RCBugFix`. I look forward to hearing whether or not we fixed it.

Kaibob2, 2016-12-10:
@RajaaKahel Did switching to RCBugfix fix this issue?

RajaaKahel, 2016-12-10:
Actually I was designing new carriages & rods & effector, so I'll try once I finish them. But I guess it's better to try the RC8 instead.

Kaibob2, 2016-12-10:
As RC8 is quite new you can use it too, but I would always suggest using the Bugfix version as this is always the most recent version.

RajaaKahel, 2016-12-10:
Thank you guys for the suggestions. I don't know if here is the right place, but what do you think about Marlin Kimbra?

closed 2017-10-31 by thinkyhead