MarlinFirmware/Marlin #5335: Displaying of SD Card section looks like broken

esenapaj, 2016-12-01:
With [latest RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/415294801f613e88779c0b2bb57e83767bb595cb), I realized that displaying of SD Card section looks like broken.

folder tree:
```
root
├ System Volume Information\ (hidden)
├ test\
│    └ UltimakerRobot_support.gcode
└ Leveling.gcode
```

result:
`REPRAP_DISCOUNT_SMART_CONTROLLER`

| Enter SD menu | Enter `test` folder | After moving the cursor up and down | Return to root |
| :-: | :-: | :-: | :-: |
| [image] | [image] | [image] | [image] |

<details><summary>Configuration.h</summary>

```cpp
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
[... 4686 lines cut ...]
// Some additional options are available for graphical displays:
#if ENABLED(DOGLCD)
  // A bigger font is available for edit items. Costs 3120 bytes of PROGMEM.
  // Western only. Not available for Cyrillic, Kana, Turkish, Greek, or Chinese.
  #define USE_BIG_EDIT_FONT

  // A smaller font may be used on the Info Screen. Costs 2300 bytes of PROGMEM.
  // Western only. Not available for Cyrillic, Kana, Turkish, Greek, or Chinese.
  #define USE_SMALL_INFOFONT

  // Enable this option and reduce the value to optimize screen updates.
  // The normal delay is 10µs. Use the lowest value that still gives a reliable display.
  #define DOGM_SPI_DELAY_US 5
#endif // DOGLCD

// @section safety

// The hardware watchdog should reset the microcontroller disabling all outputs,
// in case the firmware gets stuck and doesn't do temperature regulation.
#define USE_WATCHDOG

#if ENABLED(USE_WATCHDOG)
  // If you have a watchdog reboot in an ArduinoMega2560 then the device will hang forever, as a watchdog reset will leave the watchdog on.
  // The "WATCHDOG_RESET_MANUAL" goes around this by not using the hardware reset.
  //  However, THIS

Kaibob2, 2016-12-01:
I can confirm this. Exactly the same here with recent RCBugfix.

thinkyhead, 2016-12-01:
Did we not test this when doing our display optimizations? I could have sworn that was part of what we were supposed to go over.

thinkyhead, 2016-12-01:
It appears the screen is not being cleared and completely re-drawn, but only the second half is being rendered.

thinkyhead, 2016-12-01:
Try this. Set the `screen_changed` flag inside the function `menu_action_sddirectory`.

thinkyhead, 2016-12-01:
Try this. Add to the function `menu_action_sddirectory`:
```cpp
      lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT;
      screen_changed = true;
```

Blue-Marlin, 2016-12-01:
At least this:
```
@@ -2011,10 +2011,12 @@ void kill_screen(const char* lcd_msg) {
           MENU_ITEM(function, LCD_STR_REFRESH MSG_REFRESH, lcd_sd_refresh);
         #endif
       }
       else {
         MENU_ITEM(function, LCD_STR_FOLDER "..", lcd_sd_updir);
+        lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT;
+        screen_changed = true;
       }
 
       for (uint16_t i = 0; i < fileCnt; i++) {
         if (_menuLineNr == _thisItemNr) {
           card.getfilename(
@@ -2462,10 +2464,12 @@ void kill_screen(const char* lcd_msg) {
 
     void menu_action_sddirectory(const char* filename, char* longFilename) {
       UNUSED(longFilename);
       card.chdir(filename);
       encoderPosition = 0;
+      lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT;
+      screen_changed = true;
     }
 
   #endif //SDSUPPORT
 ```
is needed.

But likely we have a misconception here.
It's not the one or the other screen what changes to an other what should cause a screen update.
Likely 'consuming' an input event is the true reason. Where input event is a encoder change, any button click or the timeout.

thinkyhead, 2016-12-01:
Now that I think about it, graphical displays continuously re-draw, so I'm not sure how we ever end up with only a partial update.

thinkyhead, 2016-12-01:
Nevermind — I see this is the character display.

esenapaj, 2016-12-01:
@Blue-Marlin and @thinkyhead 
Thanks for the preparing the solutions.
I've tested and confirmed just now that displaying looks normal.

https://github.com/esenapaj/Marlin/commit/ea1086f1352444809b6a9a6b58d21f2102273654

Blue-Marlin, 2016-12-01:
> graphical displays continuously re-draw,

No. They don't!
It's only the status screen what is refreshed once per second. The menus and everything else is drawn only once or twice or not enough. :-) - regardless if graphical or char displays.
```cpp
    static int8_t lcd_status_update_delay = 1; // first update one loop delayed
    if (STATUS_UPDATE_CONDITION &&
      #if ENABLED(ULTIPANEL)
        currentScreen == lcd_status_screen &&    // !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
      #endif
        !lcd_status_update_delay--
    ) {
      lcd_status_update_delay = 9;
      lcdDrawUpdate = LCDVIEW_REDRAW_NOW; // !
    }
```
And __producing__ an input event is currently causing a redraw.

This is needed because the consuments are called during the first screen draw.
To make the changes visible it needs the second one.
To not react twice on an event, it has to be actively consumed. What we do with `encoderPosition = 0;` for example is active consumption and could be source for the second redraw.

thinkyhead, 2016-12-01:
Another problem occurs to me, @esenapaj.

The `lcd_clicked` flag is currently cleared after drawing the whole screen on character displays, but on graphical displays it's cleared after drawing the first segment of a graphical screen. So my guess is that when you click on items in the lower half of the graphical screen, nothing will happen.

Therefore we should clear `lcd_clicked` here:
```diff
  #define _MENU_ITEM_PART_1(TYPE, LABEL, ...) \
    if (_menuLineNr == _thisItemNr) { \
      if (lcdDrawUpdate) \
        lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \
      if (lcd_clicked && encoderLine == _thisItemNr) { \
+       lcd_clicked = false;
```

And…
```diff
-         #define CURRENTSCREEN() (*currentScreen)(), lcd_clicked = false
+         #define CURRENTSCREEN() (*currentScreen)()
```

> It's not the one or the other screen what changes to an other what should cause a screen update.

True. The only thing the `screen_changed` flag does is to cause the remainder of the menu to skip being drawn (saving only a few cycles in the case of the character display). And `screen_changed` itself is reset at the start of each screen.

thinkyhead, 2016-12-01:
> No. They don't!

Ach. You beat me to the correction.

thinkyhead, 2016-12-01:
> my guess is that when you click on items in the lower half of the graphical screen, nothing will happen.

On a second look, it appears that we're only filtering drawing and not event handling. So I'm wrong here. All items will be handled in the first display stripe, but only items in the current stripe will be drawn.

Blue-Marlin, 2016-12-02:
An other nice trick to omit the execution of the second (3,4) stripe could be to fire a `u8g.firstPage();` when we know the screen content has changed. Still experimenting.

thinkyhead, 2016-12-02:
True. That would obviate the need for the redraw flag.

For the character display, I guess it's important to use `LCDVIEW_CLEAR_CALL_REDRAW` when changing screens, so that shorter menus won't leave the previous menu items on the screen.

esenapaj, 2016-12-02:
Thanks for the fixing, probably this issue can be closed.

closed 2016-12-02 by esenapaj