MarlinFirmware/Marlin #5336: UBL Automatic Bed Probing with nozzle outside printbed

LeChuck71, 2016-12-01:
Hello,

I got a problem with the setup to probe bed locations where the nozzle is "outside" the bed but the probe is above the print bed.

Example:
**Bedsize 200 x 200**
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 175

**Probe definition**
#define X_PROBE_OFFSET_FROM_EXTRUDER -50  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0   	// Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1	// Z offset: -below +above  [the nozzle]

During M29 P1 probing the the nozzle never leaves the bed although soft-endstops are disabled i.e the probe can reach the x=200 position. Positions probed 200 x 150

Is there a way to "configure" the UBL system to probe the full "reachable" positions by the inductive probe?

Cheers
Axel

Roxy-3D, 2016-12-01:
Well...   The bed dimensions are set by what the nozzle can reach.   And if the probe is offset from the nozzle, that means some of the bed is not reachable by the probe.   If the nozzle really can move past the printable part of the bed, you can claim the bed is bigger, and then just tell the slicer program that you use to stay within the area that the bed exists.

It isn't that big of a deal that not all points can be probed.   Just do a G29 P3 R C0.0 to fill them with 0.000's    

That will allow you to do a Mesh Validation Pattern and you can go edit those points to what ever they need to be.

MoonshineSG, 2016-12-05:
My setup is similar, but my probe is on the right side. My probe never got to check the left most side of the bed (although there was enough space) until i configured it like this:

```
[...]
// the distance between the where the nozzle is at X-min endstop and the actual edge of the bed - additional variable.
#define X_GAP 40
[...]
#define X_MIN_POS -X_GAP
[...]
// the distance between the probe and the nozzle when the head is at x-min endstop
#define X_PROBE_OFFSET_FROM_EXTRUDER 54 
[...]
// make sure the nozzle is on the bed when X = 0
#define MANUAL_X_HOME_POS -X_GAP
[...]
```

Like this, X0 is at the edge of the bed, and the head is allowed to move into X negative space and when running G29, the probe can get to the left most edge of the bed. 

For the probe on the left side, it might be slightly different, but hopefully this help.

LeChuck71, 2016-12-05:
Thanks for both tips. I'll try to check them out - tricking out the slicer is a nice start, but I would like to prefer a configuration approach ;)

closed 2017-10-31 by thinkyhead