MarlinFirmware/Marlin #5339: Piston Paste Extruder - Question

BillStruve, 2016-12-01:
Printing by extruding a paste with a stepper motor driven piston does not require temperature control of the extrude head or the bed. It does require control of a fan to dry the paste after extrusion. It also requires a means of adjusting the extrude rate during printing.

Additions have been made to the RC7 release that pass all of the tests for my printer. I was about to download the latest RCBugFix so that the latest Marlin code would serve as a base for the added functionality, but I though I should ask a question first.

Should I make a pull request for RC7 with the changes I have, or go ahead and download the latest RCBugFix, make my changes to it, and then make a pull request for RCBugFix?

thinkyhead, 2016-12-01:
Since RC7 is unchanging, all PRs at this time should be made to `RCBugFix`. I look forward to seeing your work.

BillStruve, 2016-12-01:
you will as soon as analogRead() is replaced.

thinkyhead, 2016-12-01:
You need not worry about skipping over temperature checking. It can still continue. As long as no commands are sent to turn on heaters, they won't throw any errors.

BillStruve, 2016-12-01:
p.s. what rational is used to differentiate the compiler directives #ifdef XXX and #if ENABLED(XXX)?

thinkyhead, 2016-12-01:
By using the `ENABLED` macro, it allows options that are normally enabled (just by being `#define`d) to be overridden as -off- from a `Makefile` by assigning a value of `0` or `false` to the define.

BillStruve, 2016-12-04:
Done and tested on RC7, so just need to download RCBugFix, mod and test it.
For a pull you just need the changed files (6 + Configuration.h), right?
Very brief description is:
/* 
 * Changes made for the piston paste extruder in addition to Configuration.h
 * 
 * Hardware:
 *    A manually adjustable variable resistor, 1K to 100K pot, mounted to chassis
 *    AUX-1 pin 1 +5 volts - connect to high end of pot
 *    AUX-1 pin 2 ground - connect to low end of pot
 *    AUX-1 pin 3 input - connect to wiper of pot
 * 
 * Six files that were modified: 
 *  Marlin.h to:
 *    set up variables to be used in Marlin_main.cpp, planner.cpp, & temperature.cpp
 *  Marlin_main.cpp to:
 *    define the DRYER_FAN_PIN as an output, and initialize to off
 *    define and initialize variables
 *    add M35 turn to dryer fan on
 *    add M36 turn to dryer fan off
 *    add M37 turn to enable extrusion multiplier, enable showing it, and set range value
 *    provide user feedback of the multiplier (once per about 3 seconds)
 *  pins_RAMPS.h to:
 *    #define DRYER_FAN_PIN to RAMPS_D10_PIN
 *    re-#define HEATER_0_PIN to RAMPS_D9_PIN so the temperatreure routines won't bother pin 10
 *    #define MANUAL_EXTRUDER_PIN for analog input from user,  AUX-1 pin 3
 *  planner.cpp to:
 *    change the amount of paste being extruded
 *  enum.h to:
 *    include TempState's for analog read of the MANUAL_EXTRUDER_PIN
 *  temperatures.cpp to:
 *    read the analog voltage at the MANUAL_EXTRUDER_PIN
 */

thinkyhead, 2016-12-04:
Yes. Make the needed changes, commit them to your feature branch, and submit a PR against `RCBugFix`. I will review the PR and let you know what needs changing, and what else needs to be done for it to be mergeable.

stigjoergensen, 2016-12-04:
Hi @BillStruve 
Lookinf forward to this..
But why the trimpot? whats is it used for? Could it be an option to put in the LCD menu?

/Stig

BillStruve, 2016-12-04:
@stigjoergensen It's not a trimpot, it is a pot with a dial on it that is mounted so the user can adjust it. No LCD menu, just a USB plug. It is run from Simplify3D or Ponterface. When extruding paste, it is very useful to modify the extrude rate during the print by adjusting the dial. Paste tends to sag, so there is also a drying fan that needs G codes to turn it on and off.

The manufacturer of my machine has the knob and drying fan installed but is using a highly modified version of Marlin 1.0.2, which makes it very time-consuming to upgrade the firmware to a new Marlin version. Not only that, but if left to the users, different machines will be running differently modified Marlins, making troubleshooting a nightmare. Most users are artists or students.

Un-commenting a couple of #define's in Configuration.h, when getting a new version of Marlin, makes the upgrade trivial.

BillStruve, 2016-12-11:
@thinkyhead Help!
Studied git for a week and went from ignoramus to novice. Can't seem to generate a pull request or push to a fork of RCBugFix I made. The code was added and committed on my local PC, and pushed to:
https://github.com/BillStruve/PistonExtruderPrinter.git
but I don't seem to be able to generate a pull request from there. Below is the text of what was to be in the pull request...

piston extruder 5339
##Why (short version)
Printing by extruding a paste with a stepper motor driven piston does not require
temperature control of the extruder head or the bed.

It does require control of a fan to dry the paste after extrusion.
When not dried, the print will collapse when it is tall.
The manufacturer provides a warm air blower fan for this.
  M35 turns fan on, M36 turns fan off.
  
It also requires a means of adjusting the extrude rate during printing.
The manufacturer has provided a pot with a knob for the user.
  M37 A# - turns on/off the E multiplier Adjustment
  M37 S# - turns on/off Showing of the E multiplier
  M37 R##.# - adjusts the sensitivity of the knob (%)
    equation: y = mx + b  where y = multiplier,
    m = R/100/512,  x = ADC reading, b is calculated
    assuming (max x)/2 = 512 and at the midpoint of the
    adjustment, the multiplier will be 1.000
  
##What (overview)
Configuration.h
  add a section for a piston extruder printer
  add 2 defines to direct compiler
  add 2 default values
Marlin.h
  add 6 extern vars for use in Marlin_main.cpp,
    planner.cpp, and temperature.cpp
Marlin_main.cpp
  set initial values for vars
  add descriptions for M35, M36, and M37
  add code for M35, M36, and M37
  add case's for M35, M36, and M37
enum.h
  add a Prepare/Measure pair for ADC
pins_RAMPS.h
  define a pin for the dryer fan
  define a pin for the ADC input
planner.cpp
  alter extrusion with E multiplier
temperature.cpp
  setup for analog read
  use analog read to calculate E multiplier
  provide user feedback of E multiplier
example_configurations
  add MMM folder
  add a specific Configuration.h to MMM folder

closed 2017-02-11 by thinkyhead