MarlinFirmware/Marlin #5340: RC7 reliability

mjmeans, 2016-12-01:
I just wanted to report that RC7 (using ABL) has been very reliable with my custom Kossel XL. I have not tried any version of RCBugFix yet but I'm willing to do that over Xmas break if devs think that RCBugFix is stable enough for deltas. I rely on the printer for some actual R&D production and I'm not keen on any alpha level testing. But I'm willing to give beta level software a good a run through.

thinkyhead, 2016-12-01:
`RCBugFix` is about to be released as `1.1.0-RC8` shortly.
Just working through some last-minute fixes for Dual X and graphical LCD performance.

macst34, 2016-12-02:
@thinkyhead, 
Awesome! Thanks for all the hard work. Looking forward to RC-8. 
Not looking forward to putting all my customizations back in.

Michael

thinkyhead, 2016-12-03:
@macst34 How many customizations are we talking about?

thinkyhead, 2016-12-03:
@mjmeans If you like, I can add you to the @MarlinFirmware/testers-delta-team for occasional notifications about things needing specific delta testing.

macst34, 2016-12-04:
@thinkyhead 

Probably not a lot considering the scale of the things you are keeping track of.

but off the top of my head:
bootscreen change
custom thermistor table
pins for powersupply on.
180 lcd screen rotation (I still dont know where is the best place to put this...I think I added it to a few diffferent places in pins.h before it worked.
z home override.

Michael

thinkyhead, 2016-12-04:
> bootscreen change

We supply an easy way to do that.

> custom thermistor table

Easy enough to do, with many guides on the web

> pins for powersupply on

Of course, edit your pins file however you need.

> 180 lcd screen rotation

It slows down the screen, so isn't recommended. You can define it in your config.

> z home override

Already a part of the standard configuration options.

closed 2016-12-04 by thinkyhead