MarlinFirmware/Marlin #5341: Temperature drops and ThermalRunaways while Printing

Seri-, 2016-12-02:
Hey,

i recently started printing PETG on my printer and switched to the latest RCBugFix just a few days earlier. I have been using this frankenstein device for a few years: [http://wiki.2printbeta.de/doku.php/betaprusa](url) and i have modified it with stronger stepper drivers, an LCD Display, a sensor for bed leveling and i have Octoprint running on a Raspberry PI.

Since i have started printing PETG i keep getting thermal runaway errors while printing 3dBenchys (see googledrive links). The temperature sensor registeres a sudden drop in nozzle temperature by about 10 degrees and the printer  only then starts reacting to it and levels this out again.
For this to work i had to increase the thermal runawayprotection to 12 degrees for the printer to continue printing. I have printed ABS at similar temperatures without any issues with a much much older version of Marlin (maybe 3 years older) and i was not able to recreate the error by heating up the nozzle and agitating the wires to detect some looses connection that may cause this.

The Googledrive link contains screenshots of the temperature drops, a copy of my marlin settings i am using and the gcode of the benchy in question.
[https://drive.google.com/open?id=0ByWwxKeYZ9gvQXlUUjZYRG85NDQ](url)

Maybe someone can shed some light on what is going on there?

Thanks in advance

Sebastian

Blue-Marlin, 2016-12-02:
Can this be corelated to part-fan activity? Or part-fan airflow reflected from the part to the nozzle? Or sudden draft in the room?
Is the machine stuttering then?

Seri-, 2016-12-02:
I did set my part fans to fixed  80 % to test this before i startet editing the thermalprotection values and sadly it made no difference. There were also no drafts or anything else in the room that could explain this. The printer moves a lot on the upper part of the benchy but i have never had this kind of problem before i switched to the new firmware. It might be pure coincidence but i was not able to recreate the error on a hardware level (i even shaked the cables quite vigorously while printing to test this(please don't make me do it again it hurt my soul :) ).

thinkyhead, 2016-12-02:
Have you re-run `M303` to make sure everything is calibrated well to our current algorithms? You may want to try that and see if it helps. You will perhaps get a different set of P I and D values that you can plug in to test. If you do change them in your configuration, don't forget to do `M502` followed by `M500` to save them to EEPROM (resetting all other settings as well).

Seri-, 2016-12-02:
hmm i didn't think about that, i was still using the values i got from the manufacturer a few years back. I will try and report back after work.

Seri-, 2016-12-02:
Progress!
Calibrating to new PID values made the spikes sharper meaning they are compensated for more quickly but they still [exist](https://drive.google.com/open?id=0ByWwxKeYZ9gvZUpyTWpoMDRjeUE).
I just saw that the latest thermal errors i got were caused by the bed which seemed to have a range of just 2°C before shutting everything off? Is that the standard setting or was that something i changed accidentaly? If it is the standard i think it might be too tight for printers like mine with just a heated PCB and a glass plate on top.
So as of now my printer is printing without errors but those spikes still worry me.

macst34, 2016-12-02:
Hi @Seri- 

Did you do the same pid calibration for the bed using `M303 E-1 S65 C8` ?  E-1 selects the bed thermistor,  S65 sets target temp to 65,  and C8 does 8 cycles.  'M500' to save to EEProm.  

If you did PID tuning for the bed already and still have heating error issues, you could consider tweaking the bed hysteresis and maybe the temp window in configuration.h

`#define TEMP_BED_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_BED_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.`

Anyone more knowledgeable can correct me, but I believe this adjust changes the "acceptable range" of temps and while the temps may fluctuate, it might prevent you from being kicked out of your print.

There's a whole bunch of other parameters to set in the temp section of configuration.h. Take a look at them and see if they are "sane" for printing PETG.

Please report back if you get better results. I'm interested in what values work for your PCB+glass build plate.

Good luck.
Michael

Seri-, 2016-12-03:
Hey,
jeah i used S70 C10 to do the PID calibration. I did not save them to the eeprom but wrote them directly into the firmware. I hade to change the bed hysteresis to 4 to prevent thermal errors. The values claculated were 
  #define  DEFAULT_bedKp 148.66
  #define  DEFAULT_bedKi 16.26
  #define  DEFAULT_bedKd 339.78
the bed seems as stable as before when i used simple bang-bang heating and for some reason it now takes 8 minutes to heat up to 70°C when it was something like 2 minutes before. 
All in all Bed_PID was not a real success for me and i am not sure if i understand the formula well enough to manipulate it manually.
I will keep at it. 
Thank you for your help :)

macst34, 2016-12-03:
@Seri,

Just in case, you may want to check if your configuration.h changes stuck with `M503`. If you previous wrote to the eeprom, I believe it supersedes configuration.h changes since flashing the firmware does not erase the eeprom. When marlin boots up again, it retrieves the values from eeprom overriding whatever you changed in configuration.h. 

I think this is intended behavior...though this 'gotcha' tripped me up when I was doing temp calibrations.

Michael

thinkyhead, 2016-12-03:
> it now takes 8 minutes to heat up to 70°C when it was something like 2 minutes before

It may be due to the replacement of `HEATER_BED_DUTY_CYCLE_DIVIDER` with `MAX_BED_POWER`, and perhaps `PIDTEMPBED` was not enabled in your old configuration. You may try using a heated bed without `PIDTEMPBED` but I would guess that the temperature will not be as stable.

Seri-, 2016-12-03:
I just "finished" another round of PID-calibrating for my printerbed and i kind of gave up on it. Every set is giving widely differen numbers (every test was from roomtemp to 70°C). I just got back to disabling PID heating, maybe my heating element was just to unreliable.
Does anyone have an idea about those temperature spikes in the nozzle? They still appear even if i keep the printer stationary with just the heated nozzle while extruding filament at a steady rate.

thinkyhead, 2016-12-04:
You may want to try a new thermistor / thermocouple. If you are using a thermocouple, you need to make sure the current to it is even and reliable.

dembach, 2017-01-09:
got the same issue on my new printer.... MSG_THERMAL_RUNAWAY is triggered ... it happens after 3 to 5 hours of printing.... so what´s going wrong here ? why are the first 4 to 5 hours without any issues ... ?

using RC8 "Marlin 1.1.0-RCBugFix - 6 Dec 2016" with Hardware patch by Skynet3D for ANET Board.

maybe  a settings on the menu for the calibration would be nice.....

and calibration for bed does not work for me:
M303 E-1 S65 C8
PID Autotune failed! Bad extruder number
ok

but for HotEnd1:
M303 E0 S220 C8
PID Autotune start
 T:31.0 /0.0 B:22.0 /0.0 @:127 B@:0
 T:31.4 /0.0 B:21.9 /0.0 @:127 B@:0
.
 T:214.2 /0.0 B:22.4 /0.0 @:127 B@:0
 T:218.3 /0.0 B:22.5 /0.0 @:127 B@:0
 T:222.0 /0.0 B:22.5 /0.0 @:0 B@:0
 T:225.7 /0.0 B:22.5 /0.0 @:0 B@:0
.
T:220.0 /0.0 B:22.5 /0.0 @:0 B@:0
T:218.4 /0.0 B:22.5 /0.0 @:127 B@:0
.
T:220.3 /0.0 B:22.5 /0.0 @:0 B@:0
 .
 bias: 74 d: 74 min: 217.50 max: 228.00 T:218.9 /0.0 B:22.6 /0.0 @:74 B@:0
 bias: 73 d: 73 min: 216.51 max: 224.00 T:219.1 /0.0 B:22.7 /0.0 @:73 B@:0
 bias: 69 d: 69 min: 217.24 max: 223.94 Ku: 26.23 Tu: 37.39
 Classic PID 
 Kp: 15.74 Ki: 0.84 Kd: 73.57
 bias: 71 d: 71 min: 216.67 max: 223.00 Ku: 28.55 Tu: 36.93
 Classic PID 
 Kp: 17.13 Ki: 0.93 Kd: 79.06
  bias: 69 d: 69 min: 217.34 max: 223.00 Ku: 31.06 Tu: 36.70
 Classic PID 
 Kp: 18.64 Ki: 1.02 Kd: 85.50
 PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define  DEFAULT_Kp 19.08
#define  DEFAULT_Ki 1.07
#define  DEFAULT_Kd 85.36

what should i do ?

Blue-Marlin, 2017-01-09:
> M303 E-1 S65 C8
PID Autotune failed! Bad extruder number
ok

Enable PID for the bed.

dembach, 2017-01-09:
oh yeahh....... my fault.... 

works ... now go for testing, if the issue is gone....

closed 2017-04-02 by thinkyhead