MarlinFirmware/Marlin #5344: E motor: from retraction to printing speed without deceleration.

FalloutBe, 2016-12-02:
Hello,

When searching for the cause of a printing defect I narrowed it down to the extruder motor, which led me to taking high speed footage of the extruder motor doing it's thing, because it was making strange skipping noises.

This is the video: https://vid.me/zXF9
This image explains what happens in the video: http://imgur.com/a/j74Qn

All the accelerations and decelerations seem to happen fine, except for the one where the E motor goes from feeding the filament back (after a retraction). That move doesn't decelerate at the end.

In the video, clockwise turns are feeding and counter-clockwise moves are retractions.

This happens in Marlin 1.1 RC3

thinkyhead, 2016-12-03:
Can you test `RCBugFix`? We aren't patching `RC3` which is now ancient history.

Sebastianv650, 2016-12-04:
I have a little bit of a problem to find the points inside the video. Do you mean the movements where it does a rotational bouncing movements before it continues to rotate slowly?

As thinkyhead already mentioned, please test again with RCBugFix because there was a change in the way Marlin is now calculating the acceleration. Then report back if the problem is still there. If yes, the Configuration.h file would be also very useful.

If this is still valid, it might be an answer for a problem with the LIN_ADVANCE feature too..

Sebastianv650, 2016-12-05:
I made a short gcode containing of 3 lines with and without a retract between them, sending all trapezoid values over serial. I can't reproduce the issue, the numbers are fine. Maybe @FalloutBe you are using a very high e jerk or retract acceleration?

@thinkyhead I found another think during the test. Nothing realy bad, but to keep in mind: With a e_jerk of 10 and a XY jerk of 8, following happens at the end of a printing line, where a retract happens. The safe start/stop speed for the printed line is 8. But because the e_jerk speed is 10, the planner is linking the start speeds, thinking that the movement can continue at 10mm/s as if it would be a corner. So the printing move is starting from 8mm/s, accelerating to top speed of 30mm/s, decelerating to 10mm/s instead of 8 and then accelerating to retract speed..
On the other side, comming from an unretract, it's slowing down the unretract to 8mm/s which is also the start speed of the following printed line. There, it could finish the unretract at 10mm/s, continuing the start of the printing move at 8mm/s because the axis are not linked.

Maybe this can't be changed in the current planner implementation as the start speed of the next move is always the final speed of the last one, but for future coding it would be nice to do it more inteligent.

FalloutBe, 2016-12-05:
I'll test with a newer version soon to see if it's fixed but I don't have time to do it now. I'll write an update here if I do.

Something you can maybe try:
I've set the maximum E limit, and set the XY speed to max (250mm/s).
The idea behind it is that the speed will just adapt according to layer thickness to ensure the extruded flowrate is constant.
Could this be what's causing the different behaviour?

And yes, I mean the bouncing move, that's what I've encircled in red.
It's when it goes from a fast forward E rate to a slower forward E rate.

Also, what's the difference between the RC and RCBUGFIX branch?

thinkyhead, 2016-12-06:
> I found another thing during the test…

So if I understand correctly, we need to differentiate moves that only involve E, because currently the planner doesn't notice that the XY axes are actually coming to a full stop (or starting from a full stop)?

thinkyhead, 2016-12-06:
I'm not sure I understand the issue quite correctly yet. Let me see if I can sort it out.

> With an e_jerk of 10 and an XY jerk of 8, the following happens at the end of a printing line, where a retract happens: The safe start/stop speed for the printed line is 8. But because the e_jerk speed is 10, the planner is linking the start speeds, thinking that the movement can continue at 10mm/s as if it would be a corner. So the printing move is starting from 8mm/s, accelerating to top speed of 30mm/s, decelerating to 10mm/s instead of 8 and then accelerating to retract speed.

- Shouldn't the smallest jerk value apply and not the largest?
- Is the retract part of the movement? (`G1 X10 Y10 E-2`)
Or is it separate? (`G1 X10 Y10` ; `G1 E-2`)
- What about a print move followed by a retract move? (`G1 X10 Y10 E1` ; `G1 X0 Y0 E-2`)
Since the E reverses, I assume the E jerk limit will apply more strongly…?

> On the other side, coming from an unretract, it's slowing down the unretract to 8mm/s which is also the start speed of the following printed line. There, it could finish the unretract at 10mm/s, continuing the start of the printing move at 8mm/s because the axes are not linked.

Again, are you referring to an E-only move then a print move? (e.g., `G1 E2` ; `G1 X10 Y10 E3`)

oysteinkrog, 2016-12-06:
@FalloutBe: RCBugFix is latest RC + all new fixes (will become next RC/Release).
IMO we should just have master + stable/1.0 branches, but the current system works well too.
Please do test with latest code, the behavior in the video looks pretty bad.

thinkyhead, 2016-12-07:
@oysteinkrog Yeah, `master` might be a reasonable thing to have at this point. We haven't had a `master` branch in quite a long time. Although… we still wouldn't want to make it the "default" branch because Marlin is such a user-centric project (unusually so) and we're "showcasing" RC to keep interest high. Most users seem to get their 1.0.x versions of Marlin from their vendor, anyway. Maybe if `master` was just an alias to the latest release….

You might remember that we had `MarlinDev` as a point of development, but it was diverging so far from the main codebase that it was becoming unwieldy to maintain both it and the release-candidates for 1.1. And a lot of what was being done in `MarlinDev` was premature, in my view, because we really needed this whole year just to get the code cleaned up, better-organized, and actually maintainable.

The next plan, after `1.1.0` is released, is to follow two tracks:
- The `1.1.x` branch will become the new "default" branch where we showcase the latest release on the front page, while the `1.1.x-dev` branch will become the work branch for fixing up `1.1.x`. Future releases of `1.1.x` won't have `RC` tags, thank goodness. We'll just do point-releases like `1.1.1`, `1.1.2`, etc.
- The `1.2.x-dev` branch will reorganize the code (similar to the `breakup-marlin-idea` branch) and will require the newer Arduino toolchain. It will also be where we integrate 32-bit support. Eventually a `1.2.x` branch will be created to contain official releases of `1.2.0` and up.

The `devel-ubl` and `32-Bit-RcBugFix` branches are development points for more ambitious changes, which we will integrate as they mature. My hope is that branches like these will be conscientiously maintained and frequently rebased onto the more current codebase.

Sebastianv650, 2016-12-07:
@thinkyhead the retracts where done without XY movement. Here is the code:

```
G1 Z0.200 F10500.000 ; move to next layer (0)
G1 E-2.00000 F1800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X181.238 Y137.466 F10500.000 ; move to first perimeter point
G1 E2.00000 F1800.00000 ; unretract
G1 F1800
G1 X161.238 Y137.466 E2.21624 ; perimeter
G1 E0.21624 F1800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X157.283 Y137.466 F10500.000 ; move to first perimeter point
G1 E2.00000 F1800.00000 ; unretract
G1 F1800
G1 X137.283 Y137.466 E2.21624 ; perimeter
G1 X136.762 Y137.466 F10500.000 ; move to first perimeter point
G1 F1800
G1 X116.762 Y137.466 E2.43248 ; perimeter
```

> Shouldn't the smallest jerk value apply and not the largest?

Not the smallest. It depends what we want to get by the jerk speed. It's simmilar to the acceleration definition, where you mentioned in another issue that they are applied to the steppers, so on kinematic printers the acceleration of the print head may differ.
My understanding (might be wrong) of the jerk limit is that we want to limit the speed change of the print head, and also limit the start/stop speed when comming or ending with a full stop. So we are not using it to limit jerk of a stepper, at least that's what is inside the code.
If that's true, when we print a line followed by a retract (retract without XYZ movement), the print head is comming to a full stop after the line, just before the retract starts. In this case, the last speed has to be jerk speed (say 10mm/s). In fact, Marlin is actualy thinking oh, retract want's to happen at 40mm/s, and e jerk is 20mm/s (numbers are not linked to my real print from above). So I can slow down to only 20mm/s instead of 10mm/s of the end of the line which is wrong.

I had no time to inspect the code up to now, I also wanted to draw the print example in a graph.. Maybe in the next days.
If the behaviour as I described it (limit print head jerk) is the wanted one, Marlin would need to know which axis belong together. For example XY might move the print head, Z the bed and e is only extruder. And the trapezoid generator can not longer work with only a start speed of a move, saying end speed = start speed of next move. The print line move may end at jerk speed = 10, and the next move (e only) may start at e_jerk = 20.

I guess this is a relict from grbl. On a CNC, this problem isn't existing - the spindle is always moved, there is no seperate e axis. So it's true to say end speed = start speed of next move.

thinkyhead, 2016-12-09:
A picture is worth a thousand words! So here's the code that applies jerk limit:
```cpp
float safe_speed = block->nominal_speed;
bool limited = false;
LOOP_XYZE(i) {
  float jerk = fabs(current_speed[i]);                                         // (1)
  if (jerk > max_jerk[i]) {                                         // (2)
    // The actual jerk is lower if it has been limited by the XY jerk.
    if (limited) {                                                  // (3)
      // Spare one division by a following gymnastics:
      // Instead of jerk *= safe_speed / block->nominal_speed,
      // multiply max_jerk[i] by the divisor.
      jerk *= safe_speed;                                           // (4)
      float mjerk = max_jerk[i] * block->nominal_speed;
      if (jerk > mjerk) safe_speed *= mjerk / jerk;
      if (jerk * safe_speed > mjerk) safe_speed *= mjerk / (jerk * safe_speed);
    }
    else {
      safe_speed = max_jerk[i];                                     // (5)
      limited = true;                                               // (6)
    }
  }                                                                 // (7)
}
```
To limit the `safe_speed` based on `max_jerk`, loop X, Y, Z, and E axes and do:
1) Start `jerk` out as the desired current speed.
2) Over the `max_jerk` (most likely) (and so can't come to a full stop)?
3) Has it already been limited before? No? Go to [5].
4) Multiply `jerk` (i.e., `fabs(current_speed[i])`) by the previous axis's `safe_speed` (from [5])
  Get `mjerk` as `max_jerk[i]` * `block->nominal_speed` (potentially squaring or more?)
  If the new `jerk` value is > `mjerk`, reduce `safe_speed` by the ratio.
  The `jerk` value here might be squared
5) Set `safe_speed` to `max_jerk[first axis that can't full-stop]`
6) We have a new starting `safe_speed`.
7) Loop, so Y, Z, and E axes will also be checked.

Take note of the code at [4]. It would appear to reduce. First rearrange:
```cpp
      float mjerk = max_jerk[i] * block->nominal_speed;
      jerk *= safe_speed;
      if (jerk > mjerk) safe_speed *= mjerk / jerk;
```
Since the `jerk` value is not re-used, there's no need to modify it:
```cpp
      float mjerk = max_jerk[i] * block->nominal_speed;
      if (jerk * safe_speed > mjerk)
        safe_speed = (safe_speed * mjerk) / (jerk * safe_speed);
```
That last line can simply become:
```cpp
        if (jerk * safe_speed > mjerk) safe_speed = mjerk / jerk;
```
Right…?

And the code that immediately follows also makes my head hurt:
```cpp
if (moves_queued > 1 && previous_nominal_speed > 0.0001) {
  // Estimate a maximum velocity allowed at a joint of two successive segments.
  // If this maximum velocity allowed is lower than the minimum of the entry / exit safe velocities,
  // then the machine is not coasting anymore and the safe entry / exit velocities shall be used.

  // The junction velocity will be shared between successive segments. Limit the junction velocity to their minimum.
  bool prev_speed_larger = previous_nominal_speed > block->nominal_speed;
  float smaller_speed_factor = prev_speed_larger ? (block->nominal_speed / previous_nominal_speed) : (previous_nominal_speed / block->nominal_speed);
  // Pick the smaller of the nominal speeds. Higher speed shall not be achieved at the junction during coasting.
  vmax_junction = prev_speed_larger ? block->nominal_speed : previous_nominal_speed;
  // Factor to multiply the previous / current nominal velocities to get componentwise limited velocities.
  float v_factor = 1.f;
  limited = false;
  // Now limit the jerk in all axes.
  LOOP_XYZE(axis) {
    // Limit an axis. We have to differentiate: coasting, reversal of an axis, full stop.
    float v_exit = previous_speed[axis], v_entry = current_speed[axis];
    if (prev_speed_larger) v_exit *= smaller_speed_factor;
    if (limited) {
      v_exit *= v_factor;
      v_entry *= v_factor;
    }
    // Calculate jerk depending on whether the axis is coasting in the same direction or reversing.
    float jerk = 
      (v_exit > v_entry) ?
        ((v_entry > 0.f || v_exit < 0.f) ?
          // coasting
          (v_exit - v_entry) : 
          // axis reversal
          max(v_exit, -v_entry)) :
        // v_exit <= v_entry
        ((v_entry < 0.f || v_exit > 0.f) ?
          // coasting
          (v_entry - v_exit) :
          // axis reversal
          max(-v_exit, v_entry));
    if (jerk > max_jerk[axis]) {
      v_factor *= max_jerk[axis] / jerk;
      limited = true;
    }
  }
  if (limited) vmax_junction *= v_factor;
  // Now the transition velocity is known, which maximizes the shared exit / entry velocity while
  // respecting the jerk factors, it may be possible, that applying separate safe exit / entry velocities will achieve faster prints.
  float vmax_junction_threshold = vmax_junction * 0.99f;
  if (previous_safe_speed > vmax_junction_threshold && safe_speed > vmax_junction_threshold) {
    // Not coasting. The machine will stop and start the movements anyway,
    // better to start the segment from start.
    SBI(block->flag, BLOCK_BIT_START_FROM_FULL_HALT);
    vmax_junction = safe_speed;
  }
}
else {
  SBI(block->flag, BLOCK_BIT_START_FROM_FULL_HALT);
  vmax_junction = safe_speed;
}
```

thinkyhead, 2017-10-31:
Jerk code in Marlin has been modified. Jerk values from earlier versions of Marlin should be reduced by half for newer versions of Marlin.

closed 2017-10-31 by thinkyhead