MarlinFirmware/Marlin #5352: #define Z_RAISE* gone?

Ianmcmill, 2016-12-03:
I am using 1.1.0RC7BugFix and can't seem to find the #define Z_RAISE* definitions in Configuration.h and generally not in the whole branch as described in http://marlinfw.org/docs/development/configuration.html#auto-bed-leveling

Haven't found any commits which removed #define Z_RAISE*

Have those definitions been abandonded?

`#define Z_RAISE_BEFORE_PROBING 15   // How much the Z axis will be raised before traveling to the first probing point.
#define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
#define Z_RAISE_AFTER_PROBING 15    // How much the Z axis will be raised after the last probing point.`

thinkyhead, 2016-12-03:
When you try to compile with old settings `SanityCheck.h` will throw an error to alert you of the changes that are needed. Version 1.1.0-RC8 (coming soon) uses these options:
```cpp
/**
 * Z probes require clearance when deploying, stowing, and moving between
 * probe points to avoid hitting the bed and other hardware.
 * Servo-mounted probes require extra space for the arm to rotate.
 * Inductive probes need space to keep from triggering early.
 *
 * Use these settings to specify the distance (mm) to raise the probe (or
 * lower the bed). The values set here apply over and above any (negative)
 * probe Z Offset set with Z_PROBE_OFFSET_FROM_EXTRUDER, M851, or the LCD.
 * Only integer values >= 1 are valid here.
 *
 * Example: `M851 Z-5` with a CLEARANCE of 4  =>  9mm from bed to nozzle.
 *     But: `M851 Z+1` with a CLEARANCE of 2  =>  2mm from bed to nozzle.
 */
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```
Version 1.1.0-RC7 still has these, considered obsolete:
```cpp
//
// Probe Raise options provide clearance for the probe to deploy, stow, and travel.
//
#define Z_PROBE_DEPLOY_HEIGHT 15 // Raise to make room for the probe to deploy / stow
#define Z_PROBE_TRAVEL_HEIGHT 5  // Raise between probing points.
```

closed 2016-12-03 by thinkyhead

dalich, 2017-04-04:
Is there a way in RC8 to raise the Z axis before travel.  The nozzle leaves little lines on the print surface and sometimes may hit a random chunk on the print causing issues.

thinkyhead, 2017-04-04:
@dalich Your slicer should have a setting for that. That is the preferred approach, because every such concept only raises the performance requirements of the firmware.

landodragon141, 2017-04-04:
It's usually called "Z Hop" or "Raise Z with Retraction".... etc.