MarlinFirmware/Marlin #5353: Tool change question

Nocturnal42, 2016-12-03:
I've recently added a magnetic mount for my hot end and quick connect fittings to the wiring, with the intention of being able to swap out the hot end on the fly (for things like changing nozzle diameter). The issue that I am going to have, is that disconnecting the thermistor will cause marlin to die. I don't want to disable thermal protection permanently, but is there a way to disable it temporarily?

thinkyhead, 2016-12-03:
The latest code doesn't care if a thermistor is disconnected until you start the specific heater that it is attached to.

Nocturnal42, 2016-12-03:
So calling ```M104 S0``` before changing the hotend should do the trick. I'll give it a try.

Nocturnal42, 2016-12-06:
Yup, that works.

closed 2016-12-06 by Nocturnal42